coolsnowwolf/lede #4513: Cloudflare ddns 脚本更新失败

Paris0-0, 2020-05-02:
在https://github.com/openwrt/packages/pull/11989这替换了最新的脚本依然报错，显示
 082929  WARN : Could not detect 'zone id' for domain.tld: ' '
 082929 ERROR : No update send to DDNS Provider

KevinMX, 2020-05-02:
Cloudflare最近关闭了对.cf, .ga, .gq, .ml, .tk的API访问，需要手动进入控制面板才能改域名。
https://community.cloudflare.com/t/unable-to-update-ddns-using-api-for-some-tlds/167228

Paris0-0, 2020-05-02:
可是我的域名是.top啊不是免费的是付费的

opalhair, 2020-05-02:
我直接更新到 2.7.8-19 版本就没问题了

Paris0-0, 2020-05-02:
> 我直接更新到 2.7.8-19 版本就没问题了

这是openwrt固件的版本吗，我再编译个试试

Paris0-0, 2020-05-03:
感谢楼上，解决了在软件列表更新cloudflare ddns脚本就行了

closed 2020-05-03 by Paris0-0

Jefffish09, 2020-05-04:
> 感谢楼上，解决了在软件列表更新cloudflare ddns脚本就行了

@Paris0-0 小白求教怎样在软件列表更新cloudflare ddns脚本？

Paris0-0, 2020-05-04:
系统，软件包，搜ddns

Jefffish09, 2020-05-04:
> 系统，软件包，搜ddns

@Paris0-0 找到了，但似乎没见到“更新”之类的按钮？

Paris0-0, 2020-05-04:
。你是怎么装上openwrt的……
点未安装里有最新的覆盖不就好了

Jefffish09, 2020-05-05:
> 。你是怎么装上openwrt的……
> 点未安装里有最新的覆盖不就好了

@Paris0-0 哦哦，所以是直接安装新版（同一版本）的进行“覆盖”，就是所说的“更新”，对吧？

ssrv2, 2021-01-17:
> 我直接更新到2.7.8-19版本就没问题了

我尝试更新ddns-scripts_cloudflare.com-v4，发现最新版就到2.7.8-13，谷歌搜索到的结果也是这个最新版，请问您这个19版本是哪里来的？参考链接：https://openwrt.org/packages/pkgdata/ddns-scripts_cloudflare.com-v4

Paris0-0, 2021-01-17:
> > 我直接更新到2.7.8-19版本就没问题了
> 
> 我尝试更新ddns-scripts_cloudflare.com-v4，发现最新版就到2.7.8-13，谷歌搜索到的结果也是这个最新版，请问您这个19版本是哪里来的？参考链接：https://openwrt.org/packages/pkgdata/ddns-scripts_cloudflare.com-v4

直接安装最新的就好，现在的没问题

Kam323, 2021-09-24:
同样问题，软件包里更新了ddns-scripts-cloudflare 2.8.2-13，添加新的ddns依旧没有cloudflare选项，我想用v6的，但是v4 v6都没有