coolsnowwolf/lede #4514: ipsec连接后会导致路由重启

jacklei2008, 2020-05-02:
反馈bug/问题

## 1.IPSEC  vpn插件在6天前升级后出现问题，可以正常连接，但是连上之后会不定时导致路由重启.

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 可以正常连接，经过测试几次，连接后可以访问路由器ip地址，但是如果访问国内或国外fq网址，路由器会自动重启。

路由器：x86软路由。
测试手机：小米
同期同时测试同类软件：openvpn无任何问题

coolsnowwolf, 2020-05-02:
没有重现

jacklei2008, 2020-05-04:
又测试了一下：换不同的手机（同是小米手机，系统版本不同而已），问题依旧。
但问题更精确了一点。如下：
1、连接IPSEC后，通过浏览器上网，看YTB，看视频，看网站都没有问题。
2、连接IPSEC后，使用twitter APP，浏览视频，图片，大约过3-5分钟，路由器就会重启。由于瞬间重启，所以也不会截图，见凉。

jwf8280031, 2020-05-05:
我今天也遇到了，一连接马上就重启，wan接口MTU改为1492后感觉正常了，后来打开twitter网页就又重启了。

固件版本 | OpenWrt R20.4.8 / LuCI Master (git-20.117.60969-420c61a)
内核版本 | 4.19.115

jaove, 2020-05-08:
同样问题困扰我了好几天，换了Openvpn做接入正常，后来才发现有这个issus在。

主机名 | OpenWrt
型号 | Intel(R) Atom(TM) CPU D525 @ 1.80GHz : 2 Core 4 Thread (CpuMark : 13285.910016 Scores)
CPU信息 | 1795.364 MHz +20.0°C
固件版本 | OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120)
内核版本 | 4.19.115

电脑端：win10，使用VPN Access Manager接入
移动端：华为手机内置Ipsec连接方式接入
以上都出现重启同样问题

qq1316237299, 2020-05-08:
我是用ipsec听网易云 瞬间卡死重启

Blueskyhou, 2020-05-10:
出现同样问题，固件：OpenWrt R20.4.8 / LuCI Master (git-20.117.60969-420c61a)，客户端：iOs13

qq1316237299, 2020-05-10:
> 出现同样问题，固件：OpenWrt R20.4.8 / LuCI Master (git-20.117.60969-420c61a)，客户端：iOs13

ios也会这样吗

hello--world, 2020-05-13:
同样问题：
固件：OpenWrt R20.4.8 / LuCI Master (git-20.117.60969-420c61a)
内核：4.19.115
macOS连接，毕现。

RayleiLi, 2020-05-31:
我也出现了相同的问题，我是使用ios连接的

jacklei2008, 2020-06-04:
今天升级到回退(2020-5-25)后的ipsec,，还是重启。大家呢？

qq1316237299, 2020-06-04:
我更新了最新还是有重启的现象，上上个版本是没的

xiaoweb, 2020-06-09:
我也是一样, 人在公司使用mac连接家里IPSec VPN 10秒内路由器必重启

ak479822, 2020-06-15:
我也碰到同样的问题
固件：OpenWrt R20.5.20 / LuCI Master (git-20.141.34051-f750b5f)
内核：4.19.122
CPU：Intel(R) Core(TM) i7-8559U CPU @ 2.70GHz : 4 Core 8 Thread
硬盘：NVMe SSD 64G
移动端：华为手机内置Ipsec连接方式接入
现象：华为手机内置Ipsec连接方式接入后软路由重启

xiaoweb, 2020-06-15:
最终，我在pve里自己跑了个ubuntu，专门跑vpn服务😄

hello--world, 2020-06-17:
直接用docker解决：https://github.com/hwdsl2/docker-ipsec-vpn-server/blob/master/README-zh.md

xiaoqingfengATGH, 2020-07-01:
已经用docker解决。

可以试试自己给IPsec Docker封装的Luci App

https://github.com/xiaoqingfengATGH/HomeLede/wiki/%E5%A6%82%E4%BD%95%E4%BD%BF%E7%94%A8IPSecVPN%E5%AE%9E%E7%8E%B0%E5%A4%9A%E7%AB%AF%E6%8E%A5%E5%85%A5%E5%AE%B6%E5%BA%AD%E7%BD%91%E7%BB%9C

RayleiLi, 2020-07-04:
> 最终，我在pve里自己跑了个ubuntu，专门跑vpn服务😄

请问如果跑在其他机器上openwrt如何做端口转发呀？

closed 2020-07-04 by coolsnowwolf