coolsnowwolf/lede #4516: WIFI不能开启客户端模式(client)

1974294589, 2020-05-02:
## 1.关于你要提交的问题
WIFI不能开启客户端模式(client)
[X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
最新的源码编译的固件不能开户client,wpad\wpad-basic都试过了
用旧的R8.1.10.24 版本可以中继R20.4.8版本，反之不行，R20.4.8相互中继也不行
### (2) 路由器型号和固件版本
360 C301 R20.4.8
中兴 E8820 V1 R20.4.8
其它没试
### (3) 详细日志
Fri May  1 13:10:15 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Fri May  1 13:10:15 2020 daemon.notice hostapd: nl80211: Failed to remove interface wlan1 from bridge br-lan: Invalid argument
Fri May  1 13:10:16 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1-1) --> new PHY
Fri May  1 13:10:16 2020 kern.info kernel: [ 2825.714995] IPv6: ADDRCONF(NETDEV_UP): wlan1-1: link is not ready
Fri May  1 13:10:16 2020 kern.info kernel: [ 2825.789803] br-lan: port 2(wlan1-1) entered blocking state
Fri May  1 13:10:16 2020 kern.info kernel: [ 2825.795479] br-lan: port 2(wlan1-1) entered disabled state
Fri May  1 13:10:16 2020 kern.info kernel: [ 2825.801671] device wlan1-1 entered promiscuous mode
Fri May  1 13:10:16 2020 kern.info kernel: [ 2825.807071] br-lan: port 2(wlan1-1) entered blocking state
Fri May  1 13:10:16 2020 kern.info kernel: [ 2825.812746] br-lan: port 2(wlan1-1) entered forwarding state
Fri May  1 13:10:16 2020 kern.info kernel: [ 2825.936113] br-lan: port 2(wlan1-1) entered disabled state
Fri May  1 13:10:16 2020 daemon.notice hostapd: wlan1-1: interface state UNINITIALIZED->COUNTRY_UPDATE
Fri May  1 13:10:16 2020 daemon.err hostapd: Using interface wlan1-1 with hwaddr 7a:c1:a7:0a:e9:36 and ssid "OpenWrt"
Fri May  1 13:10:16 2020 daemon.notice hostapd: wlan1-1: interface state COUNTRY_UPDATE->ENABLED
Fri May  1 13:10:16 2020 daemon.notice hostapd: wlan1-1: AP-ENABLED
Fri May  1 13:10:16 2020 daemon.notice netifd: radio1 (14653): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Fri May  1 13:10:16 2020 daemon.notice netifd: radio1 (14653): Command failed: Invalid argument
Fri May  1 13:10:17 2020 user.notice root: ip link set dev wlan1-1 up
Fri May  1 13:10:17 2020 user.notice root: ip link set dev wlan1 up
Fri May  1 13:10:18 2020 daemon.notice netifd: radio1 (14653): Cannot find device "wlan1"
Fri May  1 13:10:18 2020 daemon.notice netifd: radio1 (14653): Interface 1 setup failed: IFUP_ERROR
Fri May  1 13:11:00 2020 daemon.notice hostapd: handle_probe_req: send failed
Fri May  1 13:11:01 2020 daemon.notice hostapd: handle_probe_req: send failed
Fri May  1 13:11:04 2020 daemon.notice hostapd: handle_probe_req: send failed
Fri May  1 13:11:04 2020 daemon.notice hostapd: handle_probe_req: send failed
Fri May  1 13:11:05 2020 daemon.notice netifd: radio1 (15501): WARNING: Variable 'data' does not exist or is not an array/object
Fri May  1 13:11:05 2020 kern.info kernel: [ 2875.352443] device wlan1-1 left promiscuous mode
Fri May  1 13:11:05 2020 kern.info kernel: [ 2875.357354] br-lan: port 2(wlan1-1) entered disabled state
Fri May  1 13:11:09 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1-1)
Fri May  1 13:11:09 2020 daemon.err hostapd: Interface name wlan1-1 already in use
Fri May  1 13:11:09 2020 daemon.notice netifd: radio1 (15530): Command failed: Invalid argument
Fri May  1 13:11:09 2020 daemon.notice netifd: radio1 (15530): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Fri May  1 13:11:09 2020 daemon.notice netifd: radio1 (15530): Command failed: Invalid argument
Fri May  1 13:11:10 2020 user.notice root: ip link set dev wlan1-1 up
Fri May  1 13:11:11 2020 user.notice root: ip link set dev wlan1 up
Fri May  1 13:11:11 2020 kern.info kernel: [ 2880.564539] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Fri May  1 13:11:41 2020 daemon.notice netifd: radio1 (15530): Command failed: Request timed out
Fri May  1 13:11:41 2020 daemon.notice netifd: radio1 (15530): Command failed: Not found
Fri May  1 13:11:41 2020 daemon.notice netifd: radio1 (15530): Interface 1 setup failed: WPA_SUPPLICANT_FAILED
Fri May  1 13:11:41 2020 daemon.notice netifd: radio1 (15530): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Fri May  1 13:11:41 2020 daemon.notice netifd: radio1 (15530): Command failed: Invalid argument
Fri May  1 13:11:41 2020 kern.info kernel: [ 2910.857879] br-lan: port 2(wlan1-1) entered blocking state
Fri May  1 13:11:41 2020 kern.info kernel: [ 2910.863554] br-lan: port 2(wlan1-1) entered disabled state
Fri May  1 13:11:41 2020 kern.info kernel: [ 2910.869606] device wlan1-1 entered promiscuous mode

shenzt68, 2020-05-02:
俺的也是

shiroyk, 2020-05-02:
同样的问题，Newifi D2

PR #4518 (2020-05-02): netifd: add dynamic wireless reconfiguration
Set new option 'reconf' in 'wifi-device' section to enable dynamic re-configuration on that radio. Also fix wifi relay and 'netifd: radio1 (9654): Command failed'.
Run-tested-by: @hyird 
fix #4319 , fix #4390 , fix #4474 , fix #4503 , fix #4516
closed 2020-05-02 by coolsnowwolf