coolsnowwolf/lede #4519: Dockerman没有进入界面的菜单入口

hououinkami, 2020-05-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：今天早上添加了dockerman进行编译，安装完成后进去后没有看到进入dockerman界面，只有在服务菜单里有个docker-ce容器

### (2) 路由器型号和固件版本

A：E3845 固件版本最新

### (3) 详细日志

A：

evangelion015, 2020-05-02:
docker ce容器就是dockerman，进去就有pdf教程，照着操作就行了

hououinkami, 2020-05-03:
> docker ce容器就是dockerman，进去就有pdf教程，照着操作就行了

dockerman的git页面不是展示了它有luci可视化界面吗？网上很多教程也都展示了那些页面。
lean大的直接编译就只能用命令行来安装web界面吗？

evangelion015, 2020-05-05:
> > docker ce容器就是dockerman，进去就有pdf教程，照着操作就行了
> 
> dockerman的git页面不是展示了它有luci可视化界面吗？网上很多教程也都展示了那些页面。
> lean大的直接编译就只能用命令行来安装web界面吗？

对啊，装完web界面不就是可视化界面了，没别的问题就可以关issue了

hououinkami, 2020-05-06:
> > > docker ce容器就是dockerman，进去就有pdf教程，照着操作就行了
> > 
> > 
> > dockerman的git页面不是展示了它有luci可视化界面吗？网上很多教程也都展示了那些页面。
> > lean大的直接编译就只能用命令行来安装web界面吗？
> 
> 对啊，装完web界面不就是可视化界面了，没别的问题就可以关issue了

我说的是有些人分享的固件截图，在左侧的菜单导航中，有一个docker菜单，其下有二级菜单：概况、容器、卷……等等luci界面，不是指的web可视化界面

evangelion015, 2020-05-06:
> > > > docker ce容器就是dockerman，进去就有pdf教程，照着操作就行了
> > > 
> > > 
> > > dockerman的git页面不是展示了它有luci可视化界面吗？网上很多教程也都展示了那些页面。
> > > lean大的直接编译就只能用命令行来安装web界面吗？
> > 
> > 
> > 对啊，装完web界面不就是可视化界面了，没别的问题就可以关issue了
> 
> 我说的是有些人分享的固件截图，在左侧的菜单导航中，有一个docker菜单，其下有二级菜单：概况、容器、卷……等等luci界面，不是指的web可视化界面

那你就自己换版本，清除掉L大的package，把你想要的luci版本放进去编译

doumeki, 2020-05-07:
我也没有,之前的版本我记得有的.  希望修改

rtcmay, 2020-05-07:
看别人的都是有的,也是用的L大的源码...我的也没有.怪了

hououinkami, 2020-05-08:
> 看别人的都是有的,也是用的L大的源码...我的也没有.怪了

是的，我也是看的别人用L大的源码，最后是有界面的，怎么这么迷😂

9l, 2020-05-11:
来几个对比图看看啊，没搞明白你在说什么

9l, 2020-05-11:
哦~~~知道了，你去问那个有界面的人，问问他用的是哪里的docker包，他用的不是lede自带的dockerman

slitilanber, 2020-05-11:
猜测你要找的是这个
https://github.com/lisaac/luci-app-dockerman.git

superjk2012, 2020-05-14:
L大的源码以前的确是有二级菜单：概况、容器、卷……等等luci界面的，后来不知从哪个版本开始就没有，同样也希望L大添加回去，的确太好用了

moxianiu, 2020-05-24:
> 猜测你要找的是这个
> https://github.com/lisaac/luci-app-dockerman.git

是的，怎么搞呢？

closed 2020-07-04 by coolsnowwolf