coolsnowwolf/lede #4520: X64云编译依赖报错

bin20088, 2020-05-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：X64云编译依赖报错

### (2) 路由器型号和固件版本

A：x64今天最新源码

### (3) 详细日志

A：
[image]

bin20088, 2020-05-03:
[image]

BROBIRD, 2020-05-04:
都已经很清楚告诉你了，编译的库重复了

bin20088, 2020-05-04:
> 都已经很清楚告诉你了，编译的库重复了

我是云编译，每次都是最新环境啊。同样的源码，选择同样的插件，N1顺利通过，就x64不行，正在尝试去掉-full，就得去掉ttyd，要是去-openssl，那就取消的更多了。我怀疑还是源码里有点问题吧

bin20088, 2020-05-04:
> 都已经很清楚告诉你了，编译的库重复了

取消-full、ttyd确实成功了，可是ttyd也有用啊，不想取消还有别的办法吗

leungyauming, 2020-05-19:
如果你要libwebsockets的東西，你可以試試直接用libwebsockets-full，他會把libwebsockets-openssl也裝上。

closed 2020-07-04 by coolsnowwolf