coolsnowwolf/lede #4526: Newifi D2 编译好的固件 无故断网

a1392429292, 2020-05-03:
最新编译的Newifi D2固件，过程无任何报错，刷机也一切顺利，但，就是使用过程中，无故断网，必须重启路由器方可，重启后依然过一会断网。。。无解

bluehj777, 2020-05-03:
你试试看把HWNAT关了

AmadeusGhost, 2020-05-03:
> 你试试看把HWNAT关了

也就是说那个hwnat补丁其实不大适用嘛？我看snapshot的commit message是适应最近的更改（应该是指ramips support kernel 5.4）。

a1392429292, 2020-05-03:
> 你试试看把HWNAT关了

就是说现在【Turbo ACC 网络加速设置】不可用，是吗？

bluehj777, 2020-05-03:
> > 你试试看把HWNAT关了
> 
> 就是说现在【Turbo ACC 网络加速设置】不可用，是吗？

可以开着软件加速吧
[image]

bluehj777, 2020-05-03:
> > 你试试看把HWNAT关了
> 
> 也就是说那个hwnat补丁其实不大适用嘛？我看snapshot的commit message是适应最近的更改（应该是指ramips support kernel 5.4）。

关于mt7621的HWNAT补丁我一直在关注和测试，之前2月左右出过2个补丁，打上后可以开启HWNAT生效，但是会造成路由重启，后来看到coolsnowwolf给撤了。前几天又出了这个641-netfilter-nf_flow_table-support-hw-offload-through-v.patch 我也测了，目前看在不开启FULLCONE NAT加速的前提下，路由重启的几率减少多了，还在进一步观察。
          目前我总结是如果不用HWNAT补丁的话，在19.07系列上MT7621的硬件加速是不起作用的。满速下载时候CPU占用还是挺高的，但是启用了又会造成路由重启，所以挺纠结的。但有看到讨论说在18.06系列上MT7621的HWNAT硬件加速没啥问题，也没加什么补丁，我也特意编译过一个18.06版本的固件，没有很长时间测试，但是硬件加速是启作用的，也没有路由重启现象。但只测了1天时间，没更长时间观察

a1392429292, 2020-05-04:
> > > 你试试看把HWNAT关了
> > 
> > 
> > 就是说现在【Turbo ACC 网络加速设置】不可用，是吗？
> 
> 可以开着软件加速吧
> [image]

自从不开启HWNAT，好像确实不会无故断网了。。。

closed 2020-05-04 by a1392429292