coolsnowwolf/lede #4531: Newifi3 D2 4.26后编译的固件总是自动重启

Aliceliese, 2020-05-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [×] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：从04.26开始，后面编译的固件每个都会隔一段时间就自动重启，因为重启每次都是在我打开网易云后几分钟，因此我尝试在编译时去掉此源码中的云音乐插件，改为使用cnsilvan/luci-app-unblockneteasemusic希望以此解决，但实际上路由器还是会在我打开网易云听歌后几分钟到十几分钟时间不等内重启。

### (2) 路由器型号和固件版本

A：Newifi 3 D2

### (3) 详细日志

[A：](https://lyhd.lanzous.com/ic8qi2b
)

a1392429292, 2020-05-04:
https://github.com/coolsnowwolf/lede/issues/4526

看这里

Aliceliese, 2020-05-04:
> #4526
> 
> 看这里

这个问题我是知道的，也很久没有打开过了，开启Turbo加速只会导致网络断开，而且这是在同时打开Turbo加速和广告过滤时才会出现；但我这个的问题是，当打开连接网易云后，整个路由器都会自动重启，而且无论重启多少次必定重现。

coolsnowwolf, 2020-05-04:
HWNAT 导致的内核崩溃

Aliceliese, 2020-05-04:
> HWNAT 导致的内核崩溃
我也想到过这个，于是昨天在有一次编译时就直接去掉了整个turbo加速，但是状况依旧没有消除，还是会在打开网易云（或者说其他触发条件下）直接整个路由器重启。
根据编译了好几个固件的测试：
1.在更新好固件后不打开网易云软件的条件下是基本不会重启的。
2.在04.26及之前的编译是没有出现不断重启这种情况的。
3.我在编译时选择的插件基本没有变过，应该不会出现冲突的情况。
4.turbo加速导致无法联网我这边是只会在打开广告过滤时才会出现，并且也不会导致路由器重启，只会导致网络断开。
5.也使用过lienol的源码进行编译，但还是会出现相同情况；但是当使用以前编译好的固件就不会出现这种情况，所以应该可以排除路由器本身问题。

vitalpig, 2020-05-05:
我也是会自动重启，同Newifi D2

vitalpig, 2020-05-06:
我发现我编译的时候只要不加入 应用过滤appfilter，编译的固件就不会出现重启的问题。
只要加入了应用过滤，最多坚持一个多钟就会重启，之后都是几分钟内就开始重启了。。。

Aliceliese, 2020-05-06:
> 我发现我编译的时候只要不加入 应用过滤appfilter，编译的固件就不会出现重启的问题。
> 只要加入了应用过滤，最多坚持一个多钟就会重启，之后都是几分钟内就开始重启了。。。

今天尝试在编译时去掉了oaf过滤，发现问题的确得到解决，感谢你的建议。

closed 2020-05-06 by Aliceliese