coolsnowwolf/lede #4532: 在VPS上面编译，不会出错，用虚拟机或云容易各种出错？实体机安装Ubuntu没有试过。

hlz781105, 2020-05-04:
能逛到这里的基本上都会有一个小鸡
在VPS上面编译，feeds.conf.default最后一行去掉注释，不会出错。
首先VPS面板重置一个Ubuntu 18 x86_x64
一般VPS面板重置好后只有一个root用户需要
sudo adduser ***
chmod +w /etc/sudoers
nano /etc/sudoers或者你喜欢用vi
*** ALL=(ALL:ALL) ALL
chmod -w /etc/sudoers
sudo apt-get update
apt-get upgrade
apt-get clean
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf
换你新建的用户
git clone https://github.com/coolsnowwolf/lede
cd lede
vi feeds.conf.default去掉最后一行注释
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
make -j8 download V=s
make -j1 V=s
小鸡的硬盘最好有40G及以上，我这个小鸡是一个UP主视频介绍的年付14USD 2vcpu 2G 40G 3T的，居然也是走的CN2 GIA 59.43，但是不知道超售了多少，各种卡。在这以前觉得是上当了，现在拿来编译固件还是不错。这种小鸡最好是白天编译，速度很快，晚上就算了，下载不动。。。试过 -j6，2个多小时完成。取回固件时XFTP最好设置代理，不然速度很伤心的。

自己全局环境用虚拟机总是出错需要2次编译才行。
云也试过也是出错，还是因为自已小白，技术不行。
实体机安装Ubuntu没有试过。

hlz781105, 2020-05-04:
开始没有加apt-get upgrade，X86和P 4B正常，但是friendlywrt+lede不行。

chenhao6938, 2020-05-04:
我也遇到了，在云主机上编译x86总是过不了，但是本地电脑上就没问题。

whlsxl, 2020-05-04:
docker 编译？
https://github.com/whlsxl/lede_docker

lydasia, 2020-05-14:
各种自己的虚拟机，各种云，wsl，从没成功过，物理机一般没问题

tpunix, 2020-06-23:
各种云主机内存太小, 很容易内存不够而被kill. 2G都不保险.

lydasia, 2020-06-23:
> 各种云主机内存太小, 很容易内存不够而被kill. 2G都不保险.

我特意开过一个64线192GB机器试，依然失败。。我自己的破赛扬物理机一般都没问题，但是不管是vps还是本地的wsl，hyper-v从来没成功过。

tpunix, 2020-06-23:
如果不是因为内存小而失败, 只能看看具体错误, 才知道怎么回事.

closed 2020-07-04 by coolsnowwolf