coolsnowwolf/lede #4533: 软路由需要 Turbo ACC 网络加速 吗？

a1392429292, 2020-05-04:
纯好奇，问问大家，软路由需要 Turbo ACC 网络加速 吗？
主要考虑的点是，软路由没有nat加速，HWNAT也没什么用；
所以，你们在编译的时候，软路由加【Turbo ACC 网络加速】吗？

cnclg, 2020-05-04:
基于X86的，选了，估计也没什么用处。

swxk521, 2020-05-04:
我这x86选不选感觉没啥区别，就没选

hxlls, 2020-05-04:
Flow Offload 和HWNAT是两个不同的概念，还有T ACC 也不单只只有这个

hcym, 2020-05-04:
x86都选有好处没问题，armv8不要用什么full，编译了也要在防火墙关闭，不然的话有副作用

bluehj777, 2020-05-04:
我的理解是**Turbo ACC 网络加速**是一个集成的网络加速**控制界面**ipk应用，只是为了**方便开关**包含以下几种加速：
1.Flow Offloading 转发加速 ,来自高通开源的 Shortcut FE 转发加速引擎 (Fast Path) ，一种纯软件算法上的优化。可以针对openwrt所有型号的路由器。
2.HWNAT加速。一般只针对MT762x芯片型号的路由器可以开启。比如常见的NEWIFI D2,K2P等。但目前针对19.07系列不是很完善，会造成路由器崩溃，不是MT762x芯片的路由器你开启也没有作用.
3. BBR加速。是谷歌弄出来的一个内核级别的tcp协议的优化，可以提高网络连接的能力(争抢带宽能力)，这是linux内核4.9以后都可以用的一个东西，内核大于4.9的所有型号路由器都能用
4.DNS 加速。主要是利用 pdnsd 做本地缓存，提高DNS解析速度，所有型号路由器都能用

在这个控制界面里还可以看FULLCONE NAT是否启用的，它的启用和关闭不在Turbo ACC 网络加速 里，而在“网络”->“防火墙”里可以来控制启用。它主要是针对iptables的一个改进吧，不过原作者的github里也说了，只针对几个linux内核版本做过测试，有些linux内核版本上使用是有问题的

a1392429292, 2020-05-04:
> 我的理解是**Turbo ACC 网络加速**是一个集成的网络加速**控制界面**ipk应用，只是为了**方便开关**包含以下几种加速：
> 1.Flow Offloading 转发加速 ,来自高通开源的 Shortcut FE 转发加速引擎 (Fast Path) ，一种纯软件算法上的优化。可以针对openwrt所有型号的路由器。 [...]

解说的很到位，我描述一下我目前的看法哈；

Flow Offloading和多wan拨号有冲突，比如2根宽带的建议不要开；

HWNAT；软路由用不上；

启用 BBR；看各位食客爱好；

DNS 加速，不知道和【网络】-【接口】-【LAN】里的 【使用自定义的 DNS 服务器】不知道是不是一个概念，如果是一个意义，那么也可以不需要。

bluehj777, 2020-05-04:
> > 我的理解是**Turbo ACC 网络加速**是一个集成的网络加速**控制界面**ipk应用，只是为了**方便开关**包含以下几种加速：
> > 1.Flow Offloading 转发加速 ,来自高通开源的 Shortcut FE 转发加速引擎 (Fast Path) ，一种纯软件算法上的优化。可以针对openwrt所有型号的路由器。
> > 2.HWNAT加速。一般只针对MT762x芯片型号的路由器可以开启。比如常见的NEWIFI D2,K2P等。但目前针对19.07系列不是很完善，会造成路由器崩溃，不是MT762x芯片的路由器你开启也没有作用.
> > 3. BBR加速。是谷歌弄出来的一个内核级别的tcp协议的优化，可以提高网络连接的能力(争抢带宽能力)，这是linux内核4.9以后都可以用的一个东西，内核大于4.9的所有型号路由器都能用
> > 4.DNS 加速。主要是利用 pdnsd 做本地缓存，提高DNS解析速度，所有型号路由器都能用
> > 在这个控制界面里还可以看FULLCONE NAT是否启用的，它的启用和关闭不在Turbo ACC 网络加速 里，而在“网络”->“防火墙”里可以来控制启用。它主要是针对iptables的一个改进吧，不过原作者的github里也说了，只针对几个linux内核版本做过测试，有些linux内核版本上使用是有问题的
> 
> 解说的很到位，我描述一下我目前的看法哈；
> 
> Flow Offloading和多wan拨号有冲突，比如2根宽带的建议不要开；
> 
> HWNAT；软路由用不上；
> 
> 启用BBR；看各位食客爱好；
> 
> DNS加速，和【网络】-【接口】-【LAN】里的 【使用自定义的 DNS 服务器】不知道是不是一个概念，如果是一个意义，那么也可以不需要。

这个是对dnsmasq的
pdnsd做缓存支持长缓，重启不丢失，还有pdns支持TCP/UDP查询，dnsmasq只UDP.假设你的SS服务端不支持udp那么用pdnsd可以用来TCP查询解决dns污染

a1392429292, 2020-05-04:
> 这个是对dnsmasq的
> pdnsd做缓存支持长缓，重启不丢失，还有pdns支持TCP/UDP查询，dnsmasq只UDP.假设你的SS服务端不支持udp那么用pdnsd可以用来TCP查询解决dns污染

听君一席话，感觉【Turbo ACC 网络加速】还是有必要添加食用的哈

dongjuanyong, 2020-05-05:
Flow Offloading和多wan拨号有冲突？sfe与多wan的冲突是有过，不过现在x86都用4.19内核了，早就不用sfe了，我双线用Flow Offloading正常

badcrazy, 2020-05-05:
Flow Offloading对软路由也有效的，据说开了以后n270也能跑满千兆，不过个人表示怀疑，n270在win和黑裙下网速都只有80m上下，这个Flow Offloading不可能那么神吧

a1392429292, 2020-05-05:
关闭Issues了哈，软路由还是有必要食用Turbo ACC 网络加速的哈，谢谢各位。

closed 2020-05-05 by a1392429292

xhqpp, 2020-07-03:
> x86都选有好处没问题，armv8不要用什么full，编译了也要在防火墙关闭，不然的话有副作用

fullcone?

chancat87, 2020-12-06:
还有桥接加速呢

liawt, 2022-04-16:
cpu i7 7300u, 不开软件流量分载千兆宽带测速300+,开启之后测速900+,无论开启与否cpu利用率都5%左右.

vision57, 2022-07-20:
> 我的理解是**Turbo ACC 网络加速**是一个集成的网络加速**控制界面**ipk应用，只是为了**方便开关**包含以下几种加速： 1.Flow Offloading 转发加速 ,来自高通开源的 Shortcut FE 转发加速引擎 (Fast Path) ，一种纯软件算法上的优化。可以针对openwrt所有型号的路由器。 [...]

op防火墙自带的software off load，跟第一条是一个东西吗

refined-fish, 2022-11-09:
> op防火墙自带的software off load，跟第一条是一个东西吗

是一个东西，两个地方都能开关

loydcheung, 2024-09-21:
请问现在flow offloading和多wan的冲突还是没有修复吗？

michaelngsh, 2024-09-23:
修复了

Michael NG
Mobile:  13851514952