coolsnowwolf/lede #4534: 4.19內核部分機種無法2.4G無線無法正常運作

aaronchin, 2020-05-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
A：4.19內核無法2.4G無線無法正常運作

### (2) 路由器型号和固件版本

A：Newifi D2 無法正確抓取到2.4G網卡 5G網卡正常
      竟斗雲2.0 R619AC 2.4G網卡無法當作客戶端client

### (3) 详细日志

A：以上兩機種改刷R20.4.8版固件都沒這些問題

shenzt68, 2020-05-05:
反正竟斗雲2.0 R619AC 2.4G網卡無法當作客戶端client，一直反复重启

AmadeusGhost, 2020-05-05:
日志呢？光这样说有用吗？
[image]
同竞斗云，正常使用

shenzt68, 2020-05-05:
> 日志呢？光这样说有用吗？
> [image]
> 同竞斗云，正常使用

你试试用2.4G无线中继，就出现问题了

aaronchin, 2020-05-08:
很凶的老哥 您要的日誌來了 BTW你要不要看下你的內核版本呢
[image]
[image]
[image]
有圖有真相 上面的是4.19版內核的狀況還有日誌
[log.txt](https://github.com/coolsnowwolf/lede/files/4596709/log.txt)
[log1.txt](https://github.com/coolsnowwolf/lede/files/4596711/log1.txt)
下面這個日誌是把2.4G下所有設定都刪除直接客戶端client連接時的日誌
[log3.txt](https://github.com/coolsnowwolf/lede/files/4596718/log3.txt)
下面的是4.14版內核的狀態與日誌
[image]
[image]
[log4.txt](https://github.com/coolsnowwolf/lede/files/4596723/log4.txt)
以上兩個版本皆為自己編譯的除了內核以外其他設定都是R619AC缺省直設定
大概看了一下 跟我這邊不可以的差異在於他當做client的時候是沒加密的....

AmadeusGhost, 2020-05-09:
凶啥啊，没有日志就几句话怎么分析问题
我今天全新编译的竞斗云固件中继也是正常的（kernel version 4.19.115）
[image]

AmadeusGhost, 2020-05-09:
> 有圖有真相 上面的是4.19版內核的狀況還有日誌

这个日志还是之前的问题，理论上前几天的commit已修复，要不老哥你全新编译试下？

closed 2020-05-11 by coolsnowwolf