coolsnowwolf/lede #4535: commits2505源码编译报错，请大佬帮忙瞅瞅咋回事

smallprogram, 2020-05-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
commits2505源码编译报错

### (2) 路由器型号和固件版本
x86软路由

### (3) 详细日志
···shell
rstrip.sh: /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-x86_64/libffmpeg-full/usr/lib/libswscale.so.4.8.100: shared object
rstrip.sh: /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-x86_64/libffmpeg-full/usr/lib/libavcodec.so.57.107.100: shared object
rstrip.sh: /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-x86_64/libffmpeg-full/usr/lib/libavformat.so.57.83.100: shared object
rstrip.sh: /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-x86_64/libffmpeg-full/usr/lib/libavresample.so.3.7.0: shared object
rstrip.sh: /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-x86_64/libffmpeg-full/usr/lib/libavutil.so.55.78.100: shared object
rstrip.sh: /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-x86_64/libffmpeg-full/usr/lib/libavdevice.so.57.10.100: shared object
rstrip.sh: /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-x86_64/libffmpeg-full/usr/lib/libavfilter.so.6.107.100: shared object
rstrip.sh: /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-x86_64/libffmpeg-full/usr/lib/libswresample.so.2.9.100: shared object
(cd /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-x86_64/libffmpeg-full/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -x "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -x "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/zhusir/lede/bin/packages/x86_64/packages
/home/zhusir/lede/scripts/ipkg-build -c -o 0 -g 0 /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-x86_64/libffmpeg-full /home/zhusir/lede/bin/packages/x86_64/packages
Packaged contents of /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-x86_64/libffmpeg-full into /home/zhusir/lede/bin/packages/x86_64/packages/libffmpeg-full_3.4.7-5_x86_64.ipk
rm -rf /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/.pkgdir/libffmpeg-full.installed /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/.pkgdir/libffmpeg-full
mkdir -p /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/.pkgdir/libffmpeg-full
install -d -m0755 /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/.pkgdir/libffmpeg-full/usr/lib
cp -fpR /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/ipkg-install/usr/lib/lib{avcodec,avdevice,avfilter,avformat,avresample,avutil,swresample,swscale}.so.* /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/.pkgdir/libffmpeg-full/usr/lib/
touch /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/.pkgdir/libffmpeg-full.installed
mkdir -p /home/zhusir/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/zhusir/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zhusir/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-3.4.7/.pkgdir/libffmpeg-full/. /home/zhusir/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zhusir/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libffmpeg-full_installed
echo "libffmpeg-full" >> /home/zhusir/lede/staging_dir/target-x86_64_musl/pkginfo/ffmpeg.full.install
make[3]: Leaving directory '/home/zhusir/lede/feeds/packages/multimedia/ffmpeg'
time: package/feeds/packages/ffmpeg/full/compile#398.43#24.69#430.91
make[2]: Leaving directory '/home/zhusir/lede'
package/Makefile:107: recipe for target '/home/zhusir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/zhusir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhusir/lede'
Build failed - please re-run with -j1 to see the real error message
/home/zhusir/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1
```

bin20088, 2020-05-04:
please re-run with -j1 to see the real error message

smallprogram, 2020-05-04:
Real error message :
```shell
src/gpsdclient.c: In function 'readFromGpsd':
src/gpsdclient.c:581:19: error: too few arguments to function 'gps_read'
     gpsReadCode = gps_read(gpsdata);
                   ^~~~~~~~
In file included from src/gpsdclient.h:49:0,
                 from src/gpsdclient.c:46:
/home/zhusir/lede/staging_dir/target-x86_64_musl/usr/include/gps.h:2307:12: note: declared here
 extern int gps_read(struct gps_data_t *, char *message, int message_len);
            ^~~~~~~~
../../Makefile.inc:351: recipe for target 'src/gpsdclient.o' failed
make[5]: *** [src/gpsdclient.o] Error 1
make[5]: Leaving directory '/home/zhusir/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/pud'
Makefile:241: recipe for target 'libs' failed
make[4]: *** [libs] Error 2
make[4]: Leaving directory '/home/zhusir/lede/build_dir/target-x86_64_musl/olsrd-0.9.7'
Makefile:293: recipe for target '/home/zhusir/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/.built' failed
make[3]: *** [/home/zhusir/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/.built] Error 2
make[3]: Leaving directory '/home/zhusir/lede/feeds/routing/olsrd'
time: package/feeds/routing/olsrd/compile#0.23#0.14#0.59
package/Makefile:111: recipe for target 'package/feeds/routing/olsrd/compile' failed
make[2]: *** [package/feeds/routing/olsrd/compile] Error 2
make[2]: Leaving directory '/home/zhusir/lede'
package/Makefile:107: recipe for target '/home/zhusir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/zhusir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhusir/lede'
/home/zhusir/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

smallprogram, 2020-05-04:
删了lede文件夹重头来过之后编译成功了

closed 2020-05-04 by smallprogram