coolsnowwolf/lede #4538: Compile error

orangezyx, 2020-05-04:
( cd /home/zyx/lede/build_dir/hostpkg/go-1.14/src ; GOROOT_FINAL=/home/zyx/lede/staging_dir/hostpkg/lib/go-cross  GOROOT_BOOTSTRAP=/home/zyx/lede/build_dir/hostpkg/go-1.14/.go_bootstrap GOCACHE=/home/zyx/lede/build_dir/hostpkg/go-1.14/.go_work/cache CC=gcc CXX=g++  bash make.bash --no-banner ; )
Building Go cmd/dist using /home/zyx/lede/build_dir/hostpkg/go-1.14/.go_bootstrap. (go1.4-bootstrap-20170531 linux/amd64)
Building Go toolchain1 using /home/zyx/lede/build_dir/hostpkg/go-1.14/.go_bootstrap.
# bootstrap/cmd/compile/internal/ssa
/home/zyx/lede/build_dir/hostpkg/go-1.14/src/cmd/compile/internal/ssa/rewritedec64.go:326[/home/zyx/lede/build_dir/hostpkg/go-1.14/pkg/bootstrap/src/bootstrap/cmd/compile/internal/ssa/rewritedec64.go:330]: out of memory
go tool dist: FAILED: /home/zyx/lede/build_dir/hostpkg/go-1.14/.go_bootstrap/bin/go install -gcflags=-l -tags=math_big_pure_go compiler_bootstrap bootstrap/cmd/...: exit status 2
Makefile:327: recipe for target '/home/zyx/lede/build_dir/hostpkg/go-1.14/.built' failed
make[3]: *** [/home/zyx/lede/build_dir/hostpkg/go-1.14/.built] Error 2
make[3]: Leaving directory '/home/zyx/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#73.88#13.42#89.81
package/Makefile:111: recipe for target 'package/feeds/packages/golang/host/compile' failed
make[2]: *** [package/feeds/packages/golang/host/compile] Error 2
make[2]: Leaving directory '/home/zyx/lede'
package/Makefile:107: recipe for target '/home/zyx/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/zyx/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zyx/lede'
/home/zyx/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

It seems that the error was happened while compile the golang

ElectronicElephant, 2020-05-05:
```
/home/zyx/lede/build_dir/hostpkg/go-1.14/src/cmd/compile/internal/ssa/rewritedec64.go:326[/home/zyx/lede/build_dir/hostpkg/go-1.14/pkg/bootstrap/src/bootstrap/cmd/compile/internal/ssa/rewritedec64.go:330]: out of memory
```

OOM

AmadeusGhost, 2020-05-05:
编译机内存不够

chuckyang, 2020-05-05:
> 编译机内存不够

多大内存才够，我16G都搞不定

closed 2020-05-07 by orangezyx