coolsnowwolf/lede #4543: JMicron千兆网卡不能识别和驱动的问题

carlyang1986, 2020-05-05:
我的笔记本电脑自带了一个JMicron PCI Express Gigabit Ethernet Adapter 的千兆网卡，但是我没能在make menu_config 中找到这个网卡的驱动，编译后的镜像也无法识别该网卡。我在kernel.org的4.14以及4.19版本的driver/net/ethernet中都查到了该网卡的驱动jme.c和jme.h。想请问一下如何编译才能够识别出该网卡。

dongjuanyong, 2020-05-05:
把下面这一段加在 package/kernel/linux/modules/netdevices.mk 里面：
```
define KernelPackage/jme
  SUBMENU:=$(NETWORK_DEVICES_MENU)
  TITLE:=JMicron(R) PCI-Express Gigabit Ethernet support
  DEPENDS:=@PCI_SUPPORT
  KCONFIG:=CONFIG_JME
  FILES:=$(LINUX_DIR)/drivers/net/ethernet/jme.ko
  AUTOLOAD:=$(call AutoProbe,jme)
endef

define KernelPackage/jme/description
 Kernel modules for JMicron(R) PCI-Express Gigabit Ethernet adapters
endef

$(eval $(call KernelPackage,jme))
```
make menuconfig里面Kernel modules->Network Devices->kmod-jme按y打上*

coolsnowwolf, 2020-05-05:
很少见这个型号

carlyang1986, 2020-05-07:
> 把下面这一段加在 package/kernel/linux/modules/netdevices.mk 里面：
> 
> ```
> define KernelPackage/jme
>   SUBMENU:=$(NETWORK_DEVICES_MENU)
>   TITLE:=JMicron(R) PCI-Express Gigabit Ethernet support
>   DEPENDS:=@PCI_SUPPORT +kmod-mii
>   KCONFIG:=CONFIG_JME
>   FILES:=$(LINUX_DIR)/drivers/net/ethernet/jme.ko
>   AUTOLOAD:=$(call AutoProbe,jme)
> endef
> 
> define KernelPackage/jme/description
>  Kernel modules for JMicron(R) PCI-Express Gigabit Ethernet adapters
> endef
> 
> $(eval $(call KernelPackage,jme))
> ```
> 
> make menuconfig里面Kernel modules->Network Devices->kmod-jme按y打上*

感谢指导，现在驱动正常打进去了。

closed 2020-05-07 by carlyang1986