coolsnowwolf/lede #4545: 配置CDN隐藏IP的模式下，无法运行ssr-plus

yylaji2, 2020-05-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：搭建的V2Ray+ws+tls+web,还配置CDN隐藏IP,在ssr-plus里面设置无法成功运行，服务器地址填写的是web域名。

### (2) 路由器型号和固件版本

A：固件版本最新，刚编译。

### (3) 详细日志

A：x

coolsnowwolf, 2020-05-05:
有订阅或者 导入的URL 吗

whlsxl, 2020-05-05:
本地可以运行吗？

yylaji, 2020-05-06:
1）估计家里的上面的路由有问题，我今天测试了其他上级无线（同样作为AP），倒可以运行IPV4的V2Ray + ws + tls + web+CDN隐藏IP，
2）IPV6的V2Ray + ws + tls + web+CDN隐藏IP，可以在windows的v2rayn运行，不可以在ANDROID的V2RAYNG运行，可以在android的KIT（设置启用fake dns模式下可以用，但如果启用域名嗅探不可以用）下用。

whlsxl, 2020-05-06:
看看是不是tls(https)配置的问题，访问下网页，看下https 的证书是不是正确生成。

yylaji2, 2020-05-06:
> 看看是不是tls(https)配置的问题，访问下网页，看下https 的证书是不是正确生成。

果然，测试了，https：//www. Google.com ，发现证书无法建立，现在我在v2ray服务器地址，websocket host和tls host三个地方 都写的是域名 比如www.bbb.com，请教如何修改？

whlsxl, 2020-05-06:
应该是访问`www.bbb.com` 看证书获取是否正常。
如果是证书问题，CDN一般要求托管DNS解析，因为证书要CDN申请。

yylaji2, 2020-05-06:
> 应该是访问`www.bbb.com`看证书获取是否正常。
> 如果是证书问题，CDN一般要求托管DNS解析，因为证书要CDN申请。

路由的ssr+plus设置了全局模式，然后测试www.bbb.com，连接正常，测试www.baidu.com，正常，但测试www.google.com，熄火，怪了。

whlsxl, 2020-05-06:
访问`https://www.bbb.com:[端口号]`

yylaji, 2020-05-06:
> 访问`https://www.bbb.com:[端口号]`

在用ssrplus-vps6纯小鸡全局模式下，访问https://www.bbb.com:443，可以访问，看到那个锁的标志，访问国内也行，通过BAIDU里查询IP是VPS6的IPV6地址，国外网站无法访问，不知道有没有纯IPV6小鸡用了CND隐藏+ws+tls的在SSR-PLUS实验成功的。

whlsxl, 2020-05-06:
还没测试过ipv6，有空测试下。

keivenliao, 2020-05-11:
用了CDN 至少速度慢了6、7成，不用也罢，多搞几个小机场、小vps，以备不时之需

closed 2020-07-04 by coolsnowwolf