coolsnowwolf/lede #4546: SoftEther VPN 外网无法连接有没能指导一下?

sillycat3, 2020-05-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
SoftEther VPN局域网Ip能连接，外网ddns无法连接。自己简单排查端口已打开，ddns地址映射有效，内网Ip能成功连接l2tp,openvpn均如此。网卡桥接，没有开启SoftEther内的nat，ipv6没使用。

### (2) 路由器型号和固件版本
A:X64  R20.4.8 

### (3) 详细日志

A：

shenyz, 2020-05-06:
编译的时候是带luci的还是不带luci的，带luci的话会自动端口映射，不带luci的话要手动映射 50,500,4500 端口

sillycat3, 2020-05-06:
> 
> 
> 编译的时候是带luci的还是不带luci的，带luci的话会自动端口映射，不带luci的话要手动映射 50,500,4500 端口

带LUCI ，端口是映射了

caicaicai21, 2020-05-06:
公网IP？不是公网IP的话，无法连接。

sillycat3, 2020-05-06:
> 
> 
> 公网IP？不是公网IP的话，无法连接。

100%假一罚十公网Ip.....

caicaicai21, 2020-05-06:
> > 公网IP？不是公网IP的话，无法连接。
> 
> 100%假一罚十公网Ip.....

是OpenWrt拨号？我本地是光猫拨号+OpenWrt做主路由，端口是在光猫手动映射好的，我这边是正常的。

caicaicai21, 2020-05-06:
> > 公网IP？不是公网IP的话，无法连接。
> 
> 100%假一罚十公网Ip.....

哦，对了，Flow Offload对WAN口传入连接会有影响，如果你开了，尝试关闭再试试连接。

sillycat3, 2020-05-06:
> 
> 
> > > 公网IP？不是公网IP的话，无法连接。
> > 
> > 
> > 100%假一罚十公网Ip.....
> 
> 哦，对了，Flow Offload对WAN口传入连接会有影响，如果你开了，尝试关闭再试试连接。

谢谢，关了还是不行，是openwrt拨号

king050203, 2020-05-06:
不要装luci-app-softethervpn，它会把softethervpn5-bridge、softethervpn5-client、softethervpn5-server都装上，实际只装softethervpn5-server就可以了，我目前这么用的

caicaicai21, 2020-05-06:
> > > > 公网IP？不是公网IP的话，无法连接。
> > > 
> > > 
> > > 100%假一罚十公网Ip.....
> > 
> > 
> > 哦，对了，Flow Offload对WAN口传入连接会有影响，如果你开了，尝试关闭再试试连接。
> 
> 谢谢，关了还是不行，是openwrt拨号

不排除是运营商屏蔽了端口，有尝试换端口看看？

caicaicai21, 2020-05-06:
> 不要装luci-app-softethervpn，它会把softethervpn5-bridge、softethervpn5-client、softethervpn5-server都装上，实际只装softethervpn5-server就可以了，我目前这么用的

装哪个都没有影响。

sillycat3, 2020-05-06:
> 
> 
> > > > > 公网IP？不是公网IP的话，无法连接。
> > > > 
> > > > 
> > > > 100%假一罚十公网Ip.....
> > > 
> > > 
> > > 哦，对了，Flow Offload对WAN口传入连接会有影响，如果你开了，尝试关闭再试试连接。
> > 
> > 
> > 谢谢，关了还是不行，是openwrt拨号
> 
> 不排除是运营商屏蔽了端口，有尝试换端口看看？
openvpn的端口自己换过也是不行，其实L大固件之前我在MTK7621平台编译也用过SoftEther VPN没问题，现在换到X64平台就不行。设置都一样......

xinxijishuwyq, 2020-05-07:
> > 不要装luci-app-softethervpn，它会把softethervpn5-bridge、softethervpn5-client、softethervpn5-server都装上，实际只装softethervpn5-server就可以了，我目前这么用的
> 
> 装哪个都没有影响。

不知道现在好了吗，之前bridge和server如果同时装上会出现没法用server的功能的问题

caicaicai21, 2020-05-07:
> > > 不要装luci-app-softethervpn，它会把softethervpn5-bridge、softethervpn5-client、softethervpn5-server都装上，实际只装softethervpn5-server就可以了，我目前这么用的
> > 
> > 
> > 装哪个都没有影响。
> 
> 不知道现在好了吗，之前bridge和server如果同时装上会出现没法用server的功能的问题

我一直都是用配置 luci-app-softethervpn，一直都是正常
[image]
[image]
[image]

caicaicai21, 2020-05-07:
> > > > > > 公网IP？不是公网IP的话，无法连接。
> > > > > 
> > > > > 
> > > > > 100%假一罚十公网Ip.....
> > > > 
> > > > 
> > > > 哦，对了，Flow Offload对WAN口传入连接会有影响，如果你开了，尝试关闭再试试连接。
> > > 
> > > 
> > > 谢谢，关了还是不行，是openwrt拨号
> > 
> > 
> > 不排除是运营商屏蔽了端口，有尝试换端口看看？
> > openvpn的端口自己换过也是不行，其实L大固件之前我在MTK7621平台编译也用过SoftEther VPN没问题，现在换到X64平台就不行。设置都一样......

我也是x86_64平台，自编译的固件，不过是安装在PVE虚拟机里面。
如果OpenVPN不能连接，那你试试Softether Client是否可以连接，一般来说是可以连接的，即使是内网，它也能穿透。如果还是不行，那就只能是运营商的原因或者是端口映射的原因了。

host2318, 2020-05-09:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [x ] 没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> SoftEther VPN局域网Ip能连接，外网ddns无法连接。自己简单排查防火墙端口已打开，ddns地址映射有效，内网Ip能成功连接l2tp,openvpn均如此。虚拟网卡桥接lan，没有开启SoftEther内的nat，ipv6没使用。
> 
> ### (2) 路由器型号和固件版本
> A:X64 R20.4.8
> 
> ### (3) 详细日志
> A：
有没有在接口 lan里面的-物理设置-接口，勾选桥接的接口啊
[image]

sillycat3, 2020-05-17:
> 
> 
> > > 公网IP？不是公网IP的话，无法连接。
> > 
> > 
> > 100%假一罚十公网Ip.....
> 
> 哦，对了，Flow Offload对WAN口传入连接会有影响，如果你开了，尝试关闭再试试连接。

sillycat3, 2020-05-17:
问题解决了，谢谢楼上的答案，重置了系统不开启Flow Offload 成功

closed 2020-05-17 by sillycat3

mkcczh, 2020-09-09:
楼主你好，我遇到了一样的情况，内网能连，外网不能，请问怎样关闭 FLow Offload，我的版本防火墙里面没有直接关闭的选项？谢谢！

sillycat3, 2020-09-09:
> 楼主你好，我遇到了一样的情况，内网能连，外网不能，请问怎样关闭 FLow Offload，我的版本防火墙里面没有直接关闭的选项？谢谢！

不在防火墙，在Turbo ACC 里面，大概记得是第一个转发选项（和多WAN或者QOS功能可能会冲突）这里。

mkcczh, 2020-09-09:
SFE Fast Path 加速 选项， 我已经关闭了，还是不行。但是我没有重置整个系统，可能要试下重置了。

sillycat3, 2020-09-09:
> SFE Fast Path 加速 选项， 我已经关闭了，还是不行。但是我没有重置整个系统，可能要试下重置了。

我之前不重置也不行的，如果重置还不行尝试找其他原因吧

mkcczh, 2020-09-11:
重置后，不开启SFE Fast Path 加速选项，行了，谢谢！

cncqcyc, 2021-03-29:
我的是有时行有时又不行，softether有时自己会改配置

cncqcyc, 2021-03-29:
> 我的是有时行有时又不行，softether有时自己会改配置

有时没有生成日志文件，刚才刷出来了日志文件，查看是没有分配到有效的ip地址，将softether自带的dhcp启用了，现在正常了

lixh1223, 2022-10-29:
这里有视频教程 ，可以看一下。https://www.youtube.com/watch?v=fWGw6Qdh1pY

iKira, 2022-11-20:
> > > 公网IP？不是公网IP的话，无法连接。
> > 
> > 
> > 100%假一罚十公网Ip.....
> 
> 哦，对了，Flow Offload对WAN口传入连接会有影响，如果你开了，尝试关闭再试试连接。

这个答案才是正解。

myfalcom, 2022-11-20:
> > > > 公网IP？不是公网IP的话，无法连接。
> > > 
> > > 
> > > 100%假一罚十公网Ip.....
> > 
> > 
> > 哦，对了，Flow Offload对WAN口传入连接会有影响，如果你开了，尝试关闭再试试连接。
> 
> 这个答案才是正解。

刚试了下，还是不行，aac那里关了。连上后就断开。