coolsnowwolf/lede #4547: ipq40xx设备第一个有线网口故障

lydasia, 2020-05-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

Netgear EX6150v2刷OpenWrt原生固件唯一网口正常，刷自编译的本固件网口故障

爬论坛发现，类似方案的Linksys EA6350 v3等设备似乎同样存在wan口失效，第一个lan口变成wan口的故障。

怀疑与之前的某个patch有关。

### (2) 路由器型号和固件版本

A：ipq40xx方案路由器，最新版本

### (3) 详细日志

A：luci中eth0设备可见，插网线灯常亮，显示连接时间0h0m0s，仅有发出数据包，无收到数据包，无法获取ip地址，无论配置为lan或wan均无法正常使用。

AmadeusGhost, 2020-05-06:
照着这个commit改下试试？https://github.com/AmadeusGhost/lede/commit/333ada7d5f7c4c3296d9c48937ed1bc31d31526e

lydasia, 2020-05-06:
> 照着这个commit改下试试？[AmadeusGhost@e65c4b6](https://github.com/AmadeusGhost/lede/commit/e65c4b618e24383add07191270bd9a7dcb2ab53f)

谢谢，我试试

lydasia, 2020-05-06:
> 照着这个commit改下试试？[AmadeusGhost@e65c4b6](https://github.com/AmadeusGhost/lede/commit/e65c4b618e24383add07191270bd9a7dcb2ab53f)

还没试，但我跟openwrt的board.d对照着看了一下，感觉靠谱！

lydasia, 2020-05-06:
> 照着这个commit改下试试？[AmadeusGhost@e65c4b6](https://github.com/AmadeusGhost/lede/commit/e65c4b618e24383add07191270bd9a7dcb2ab53f)

还没试，现在lean和原版op的board.d是相似的，感觉不是这个问题，但因为可能改了别的地方，这个commit或许有可能解决。

不知道跟这个patch有没有关系：
https://github.com/coolsnowwolf/lede/blob/master/target/linux/ipq40xx/patches-4.14/901-essedma-disable-default-vlan-tagging.patch

AmadeusGhost, 2020-05-06:
> 还没试，现在lean和原版op的board.d是相似的，感觉不是这个问题，但因为可能改了别的地方，这个commit或许有可能解决。

是因为这个patch改了vlan，这里和官方是不同的处理方式，别纠结这个了，先试下再说吧🙄

lydasia, 2020-05-06:
> > 还没试，现在lean和原版op的board.d是相似的，感觉不是这个问题，但因为可能改了别的地方，这个commit或许有可能解决。
> 
> 是因为这个patch改了vlan，这里和官方是不同的处理方式，别纠结这个了，先试下再说吧🙄

按这个commit编译刷了，直接变半砖状态。。不知道什么原因，刚debrick好

AmadeusGhost, 2020-05-06:
> 按这个commit编译刷了，直接变半砖状态。。不知道什么原因，刚debrick好

不应该啊，只是改了网口配置

lydasia, 2020-05-07:
> > 按这个commit编译刷了，直接变半砖状态。。不知道什么原因，刚debrick好
> 
> 不应该啊，只是改了网口配置

还是不行，直接刷原版装饮料+了，很稳。。

AmadeusGhost, 2020-05-07:
> 还是不行，直接刷原版装饮料+了，很稳。。

ex6150v2几个lan？

lydasia, 2020-05-07:
> > 还是不行，直接刷原版装饮料+了，很稳。。
> 
> ex6150v2几个lan？

这个原厂设定是当Range Extender/Access Point，仅有一个有线网口，侧面有一拨动开关。第三方固件的话应该设定为：刷完以后网口默认是wan，默认打开无线；或者默认关闭无线，网口默认是lan。以便于配置。我看您的commit里switch0设定了5个lan？

AmadeusGhost, 2020-05-07:
> 这个原厂设定是当Range Extender/Access Point，仅有一个有线网口，侧面有一拨动开关。第三方固件的话应该设定为：刷完以后网口默认是wan，默认打开无线；或者默认关闭无线，网口默认是lan。以便于配置。我看你的commit里switch0设定了5个lan？

跟ex6200弄混了😂一个口应该原来的写法就行，所以还是有点迷（

AmadeusGhost, 2020-05-07:
> 还是不行，直接刷原版装饮料+了，很稳。。

请问官方op固件`swconfig list`和`swconfig dev switch0 show`的结果是什么？

lydasia, 2020-05-07:
> > 还是不行，直接刷原版装饮料+了，很稳。。
> 
> 请问官方op固件`swconfig list`和`swconfig dev switch0 show`的结果是什么？

root@OpenWrt:~# swconfig list

Found: switch0 - 90000.mdio

root@OpenWrt:~# swconfig dev switch0 show

Global attributes:
        enable_vlan: 1
        enable_mirror_rx: 0
        enable_mirror_tx: 0
        mirror_monitor_port: 0
        mirror_source_port: 0
        linkdown: ???

Port 0:
        mib: Port 0 MIB counters
统计信息略，有数据

        pvid: 0
        link: port:0 link:up speed:1000baseT full-duplex txflow rxflow
Port 1:
        mib: Port 1 MIB counters
统计信息略，均为0

        pvid: 1
        link: port:1 link:down
Port 2:
        mib: Port 2 MIB counters
统计信息略，均为0

        pvid: 1
        link: port:2 link:down
Port 3:
        mib: Port 3 MIB counters
统计信息略，均为0

        pvid: 1
        link: port:3 link:down
Port 4:
        mib: Port 4 MIB counters
统计信息略，有数据

        pvid: 1
        link: port:4 link:up speed:100baseT full-duplex auto
Port 5:
        mib: Port 5 MIB counters
统计信息略，均为0

        pvid: 2
        link: port:5 link:down
VLAN 1:
        vid: 1
        ports: 0t 1 2 3 4
VLAN 2:
        vid: 2
        ports: 0t 5

lydasia, 2020-05-07:
[MobaXterm_192.168.1.1root_20200507_205602.txt](https://github.com/coolsnowwolf/lede/files/4593027/MobaXterm_192.168.1.1root_20200507_205602.txt)

lydasia, 2020-05-07:
网口是千兆的，目前协商速率是百兆

AmadeusGhost, 2020-05-07:
> 网口是千兆的，目前协商速率是百兆

```
netgear,ex6100v2|\
netgear,ex6150v2)
	ucidef_add_switch "switch0" \
		"0t@eth0" "4:lan"
	;;
```
可能是这样定义

lydasia, 2020-05-07:
> > 网口是千兆的，目前协商速率是百兆
> 
> ```
> netgear,ex6100v2|\
> netgear,ex6150v2)
> 	ucidef_add_switch "switch0" \
> 		"0t@eth0" "4:lan"
> 	;;
> ```
> 
> 可能是这样定义

嗯谢谢，我有空再试试，原版貌似就是ucidef_set_interface_lan "eth0"。
我这个型号其实比较小众吧，关键是不是在于ipq40xx设备最好有个通用的解决方案？

lydasia, 2020-05-07:
> > 网口是千兆的，目前协商速率是百兆
> 
> ```
> netgear,ex6100v2|\
> netgear,ex6150v2)
> 	ucidef_add_switch "switch0" \
> 		"0t@eth0" "4:lan"
> 	;;
> ```
> 
> 可能是这样定义

啊我看了下，也就这两组没有定义了：

`compex,wpj428)
		ucidef_set_interface_lan "eth0 eth1"
		;;
	engenius,eap1300|\
	meraki,mr33|\
	netgear,ex6100v2|\
	netgear,ex6150v2|\
	zyxel,wre6606)
		ucidef_set_interface_lan "eth0"`

上面那个是个单板机可以忽略，下面这几个，由于布线不同，可能得分别写。您写的应该没问题，我再冒险试试，反正网件很好debrick。

AmadeusGhost, 2020-05-08:
>原版貌似就是ucidef_set_interface_lan "eth0"。

我上面的写法可能还是不行，只是少定义了几个lan，还有一种写法：
```
netgear,ex6100v2|\
netgear,ex6150v2)
	ucidef_add_switch "switch0" \
		"0u@eth0" "4:lan"
	;;
```

lydasia, 2020-05-08:
> > 原版貌似就是ucidef_set_interface_lan "eth0"。
> 
> 我上面的写法可能还是不行，只是少定义了几个lan，还有一种写法：
> 
> ```
> netgear,ex6100v2|\
> netgear,ex6150v2)
> 	ucidef_add_switch "switch0" \
> 		"0u@eth0" "4:lan"
> 	;;
> ```

反馈：已经编译刷写试过了，第一种写法（0t@eth0" "4:lan）是可以使设备正常工作的。看上去有点奇怪，因为只有一个网口，显示有一个交换机eth0，一个vlan，eth0.1。但是测试性能等都没有问题。第二种写法目前还没试。

lydasia, 2020-05-08:
> > 原版貌似就是ucidef_set_interface_lan "eth0"。
> 
> 我上面的写法可能还是不行，只是少定义了几个lan，还有一种写法：
> 
> ```
> netgear,ex6100v2|\
> netgear,ex6150v2)
> 	ucidef_add_switch "switch0" \
> 		"0u@eth0" "4:lan"
> 	;;
> ```

swconfig输出和原版无任何区别。

AmadeusGhost, 2020-05-08:
> 第二种写法目前还没试。

可以用就不用试了，那我就pr了
> swconfig输出和原版无任何区别。

指第一种写法？

lydasia, 2020-05-08:
> > 第二种写法目前还没试。
> 
> 可以用就不用试了，那我就pr了
> 
> > swconfig输出和原版无任何区别。
> 
> 指第一种写法？

是的，指第一种写法。

lydasia, 2020-05-08:
> > 第二种写法目前还没试。
> 
> 可以用就不用试了，那我就pr了
> 
> > swconfig输出和原版无任何区别。
> 
> 指第一种写法？

或者其实可以写成wan，毕竟这两个型号都只有一个以太网口，lean的固件默认开启wlan

lydasia, 2020-05-08:
> > 第二种写法目前还没试。
> 
> 可以用就不用试了，那我就pr了
> 
> > swconfig输出和原版无任何区别。
> 
> 指第一种写法？

经过反复测试，"0u@eth0" "4:lan"是最完美的。

closed 2020-05-08 by lydasia

AmadeusGhost, 2020-05-08:
> 经过反复测试，"0u@eth0" "4:lan"是最完美的。

怎么个完美法？

lydasia, 2020-05-08:
> > 经过反复测试，"0u@eth0" "4:lan"是最完美的。
> 
> 怎么个完美法？

"0t@eth0" "4:lan"     显示有一个交换机eth0，一个vlan，eth0.1对应唯一网口
"0t@eth0" "4:wan"     无线不会自动加入lan，初始配置困难
"0u@eth0" "4:lan"     仅显示eth0，对应唯一网口，菜单中出现交换机，点开可进行配置
"0u@eth0" "4:wan"     无线不会自动加入lan，初始配置困难

PR #4550 (2020-05-09): ipq40xx: adjust switch configuration for netgear ex61x0v2
Fix #4547