coolsnowwolf/lede #4551: 想问用一下firewall.user里的 --dport 53 -j REDIRECT --to-ports 53是给谁用的

bluehj777, 2020-05-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
lean 的固件编译后会在firewall.user里添加2条iptables语句
#iptables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 53
#iptables -t nat -A PREROUTING -p tcp --dport 53 -j REDIRECT --to-ports 53
语句本身明白是什么意思，但不太清楚为啥要加上它呢，似乎不加也没什么问题

Bnosora, 2020-05-06:
有的时候，劫持dns比修改每个设备上的dns更简单

bluehj777, 2020-05-06:
> 有的时候，劫持dns比修改每个设备上的dns更简单

可是在这里的应用场景，并不需要劫持客户端的dns啊，本来客户端都是分配用路由的ip:53来当dns啊

bluehj777, 2020-05-06:
> 有的时候，劫持dns比修改每个设备上的dns更简单

是不是可以这样理解，假设网络里所有客户端都是按规矩地动态从路由里过去ip,网关，dns
那么，这2个语句是没必要加的。假设有一个客户端它自己指定了dns,在这里如果有这两个语句的话就可以劫持这个客户端的设置了。那么在这里这2条语句全是个强制限定的设置

Bnosora, 2020-05-06:
> > 有的时候，劫持dns比修改每个设备上的dns更简单
> 
> 可是在这里的应用场景，并不需要劫持客户端的dns啊，本来客户端都是分配用路由的ip:53来当dns啊

有些设备不会自动获取的，比如chromecast，写死了8.8.8.8。

Bnosora, 2020-05-06:
> > 有的时候，劫持dns比修改每个设备上的dns更简单
> 
> 是不是可以这样理解，假设网络里所有客户端都是按规矩地动态从路由里获取ip,网关，dns
> 那么，这2个语句是没必要加的。假设有一个客户端它自己指定了dns,在这里如果有这两个语句的话就可以劫持这个客户端的设置了。那么在这里这2条语句算是个强制限定的设置

是

closed 2020-05-06 by bluehj777