coolsnowwolf/lede #4552: ARP劫持问题

icecna, 2020-05-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：** ARP劫持问题**
编译出的固件作为路由部署后，会劫持内网其他机器ip（包括其他网段）劫持为该路由MAC地址，造成网关非该路由的内网设备无法上网。

### (2) 路由器型号和固件版本

A：OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120)

### (3) 详细日志

A：
`PS C:\Windows\system32> arp -a

接口: 192.168.1.160 --- 0x10
  Internet 地址         物理地址              类型
  192.168.1.1           00-e0-70-52-e4-83     动态
  192.168.8.1           d4-5f-25-fd-fd-98     动态
  224.0.0.2             01-00-5e-00-00-02     静态
  224.0.0.22            01-00-5e-00-00-16     静态
  224.0.0.251           01-00-5e-00-00-fb     静态
  224.0.0.252           01-00-5e-00-00-fc     静态
  239.255.255.250       01-00-5e-7f-ff-fa     静态
接口: 192.168.12.33 --- 0x11
  Internet 地址         物理地址              类型
  224.0.0.22            01-00-5e-00-00-16     静态
  224.0.0.251           01-00-5e-00-00-fb     静态
  239.255.255.250       01-00-5e-7f-ff-fa     静态
PS C:\Windows\system32> arp -d
PS C:\Windows\system32> arp -a

接口: 192.168.1.160 --- 0x10
  Internet 地址         物理地址              类型
  192.168.8.1           00-e0-70-52-e4-83     动态
  224.0.0.2             01-00-5e-00-00-02     静态
  224.0.0.22            01-00-5e-00-00-16     静态

接口: 192.168.12.33 --- 0x11
  Internet 地址         物理地址              类型
  224.0.0.22            01-00-5e-00-00-16     静态
  224.0.0.251           01-00-5e-00-00-fb     静态
  239.255.255.250       01-00-5e-7f-ff-fa     静态`

yanest, 2020-05-07:
同样的问题，之前提issue被关闭

programmejava, 2020-05-09:
关于楼主你这个沙雕问题 来说两句  前一次应该也是你发的说是这个固件劫持网络之类的  首先你在同一网段中放两个路由器 不知道做什么用的，xp之后的系统已经免疫arp病毒，再者这个固件有这么多用户使用 可就单单你一个沙雕有这个问题。,提issue没有人有相同的问题，你是不是故意来捣乱的？扣这么大一个帽子？  确定是不是哪个UP主派来的？

yanest, 2020-05-09:
卧槽，同一个网段不能放两台路由器？我看你倒像傻屌。我就不能再接一个路由器的lan口当ap，我也有同样的问题，人家楼主把事实列出来提问，你特么验证一下不就知道了。

icecna, 2020-05-09:
> 关于楼主你这个沙雕问题 来说两句 前一次应该也是你发的说是这个固件劫持网络之类的 首先你在同一网段中放两个路由器 不知道做什么用的，xp之后的系统已经免疫arp病毒，再者这个固件有这么多用户使用 可就单单你一个沙雕有这个问题。,提issue没有人有相同的问题，你是不是故意来捣乱的？扣这么大一个帽子？ 确定是不是哪个UP主派来的？

你这个沙雕上来就骂人什么意思？我只是反应真实存在的问题到你这就成了扣帽子了。 选择用ARP劫持的初衷可能是为了旁路由免配置科学上网，但是实际上效果并不好，在内网多台服务设备的情况下还对用户造成了困扰。

programmejava, 2020-05-10:
> 卧槽，同一个网段不能放两台路由器？我看你倒像傻屌。我就不能再接一个路由器的lan口当ap？我也有同样的问题，默认编译接上就劫持。人家楼主把事实列出来提问，你特么验证一下不就知道了。

人才  ！！！你确定不是猴子搬来的救兵？默认编译接上就劫持！！！哈哈哈，笑死个人。openwrt会配置吗？网络基础了解了吗？步子迈太大容易扯到蛋！！你们两个沙雕难兄难弟可以一起交流交流！！！

icecna, 2020-05-10:
这tm谁家的狗？ 某些问题不准讨论，你TM直接公告啊，CNM，不用了还不行吗？

yanest, 2020-05-10:
> > 卧槽，同一个网段不能放两台路由器？我看你倒像傻屌。我就不能再接一个路由器的lan口当ap？我也有同样的问题，默认编译接上就劫持。人家楼主把事实列出来提问，你特么验证一下不就知道了。
> 
> 人才 ！！！你确定不是猴子搬来的救兵？默认编译接上就劫持！！！哈哈哈，笑死个人。openwrt会配置吗？网络基础了解了吗？步子迈太大容易扯到蛋！！你们两个沙雕难兄难弟可以一起交流交流！！！

你特么也配谈网络基础，就jb会咬人，你告诉我哪些代码是你提交的。

9l, 2020-05-11:
没搞明白你怎么设置，把所有设置贴上来看看

icecna, 2020-05-11:
> 没搞明白你怎么设置，把所有设置贴上来看看
> 
> PS：你同一个网络里开了不止一个DHCP服务器？没有做vlan隔离么，那会一片混乱哦

已经找到问题所在，proxy_arp 默认打开的
```
root@OpenWrt:~# cat /proc/sys/net/ipv4/conf/br-lan/proxy_arp
1
```

9l, 2020-05-12:
cat /etc/sysctl.conf
cat /etc/sysctl.d/10-default.conf

可以看看 sysctl 里面有关 proxy_arp 的设置吧，我这里是没有这个设置的

icecna, 2020-05-12:
```
root@OpenWrt:~# cat /etc/sysctl.d/10-default.conf
# Do not edit, changes to this file will be lost on upgrades
# /etc/sysctl.conf can be used to customize sysctl settings

kernel.panic=3
kernel.core_pattern=/tmp/%e.%t.%p.%s.core
fs.suid_dumpable=2

fs.protected_hardlinks=1
fs.protected_symlinks=1

net.ipv4.conf.default.arp_ignore=1
net.ipv4.conf.all.arp_ignore=1
net.ipv4.ip_forward=1
net.ipv4.icmp_echo_ignore_broadcasts=1
net.ipv4.icmp_ignore_bogus_error_responses=1
net.ipv4.igmp_max_memberships=100
net.ipv4.tcp_fin_timeout=30
net.ipv4.tcp_keepalive_time=120
net.ipv4.tcp_syncookies=1
net.ipv4.tcp_timestamps=1
net.ipv4.tcp_sack=1
net.ipv4.tcp_dsack=1

net.ipv6.conf.default.forwarding=1
net.ipv6.conf.all.forwarding=1
```

```
root@OpenWrt:~# cat /etc/sysctl.conf
# Defaults are configured in /etc/sysctl.d/* and can be customized in this file
```

> cat /etc/sysctl.conf
> cat /etc/sysctl.d/10-default.conf
> 
> 可以看看 sysctl 里面有关 proxy_arp 的设置吧，我这里是没有这个设置的

9l, 2020-05-12:
在配置文件里改设置要重启才能生效，或者是有别的软件或服务在系统重启的时候动态修改了配置

我是只装了一个酸酸乳，别的软件和服务都没有装，这个系统只拿来做梯子。路由 DHCP 啥的都在别的系统上

icecna, 2020-05-12:
是重启后不生效
我也怀疑是被某个脚本修改了配置

proomoteshen, 2020-05-13:
package/lean/luci-app-ipsec-vpnd/root/etc/ipsec.include
第十三行，搜索一下echo 1 > /proc/sys/net/ipv4/conf/br-lan/proxy_arp就可以

closed 2020-07-04 by coolsnowwolf