coolsnowwolf/lede #4554: openwrt本身无法上网

miroda, 2020-05-06:
1、电脑通过openwrt可以正常上网，ap链接openwrt后也能上网
2、openwrt设备本身能上网

#===================== 系统信息 =====================#
主机型号: Intel(R) Celeron(R) CPU J1900 @ 1.99GHz : 4 Core 4 Thread
固件版本: OpenWrt SNAPSHOT r0-6b142ba
LuCI版本: git-20.088.29068-cebe120-1
内核版本: 4.19.108
处理器架构: x86_64

[image]
无法ping通百度

miroda, 2020-05-06:
[image]

miroda, 2020-05-06:
[image]

miroda, 2020-05-06:
[image]

miroda, 2020-05-06:
root@Openwrt:~# cat /etc/config/network

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config interface 'lan'
        option type 'bridge'
        option ifname 'eth0 eht2 eth3'
        option proto 'static'
        option ipaddr '192.168.100.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option mtu '1492'

config interface 'wan'
        option ifname 'eth1'
        option proto 'dhcp'

config interface 'wan6'
        option ifname 'eth3'
        option proto 'dhcpv6'

config route

config interface 'vpn0'
        option ifname 'tun0'
        option proto 'none'

Mattraks, 2020-05-07:
关掉翻车插件（clash），那个玩意劫持dns

miroda, 2020-05-07:
正解，关闭clash之后就可以了 ，哎
PING openwrt.org (139.59.209.225): 56 data bytes
64 bytes from 139.59.209.225: seq=0 ttl=52 time=223.262 ms
64 bytes from 139.59.209.225: seq=1 ttl=52 time=223.322 ms
64 bytes from 139.59.209.225: seq=2 ttl=52 time=223.072 ms
64 bytes from 139.59.209.225: seq=4 ttl=52 time=223.106 ms

--- openwrt.org ping statistics ---
5 packets transmitted, 4 packets received, 20% packet loss
round-trip min/avg/max = 223.072/223.190/223.322 ms

closed 2020-05-07 by miroda