coolsnowwolf/lede #4561: 最新代码首次编译报错 /home/luy/lede/include/toplevel.mk:225: recipe for target 'world' failed

codersluy, 2020-05-07:
##  1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新代码首次 Ubuntu18 下编译报错 /home/luy/lede/include/toplevel.mk:225: recipe for target 'world' failed

### (2) 系统版本

A：ubuntu-18.04.4-live-server-amd64

### (3) 详细日志

A：
```shell
collect2: error: ld returned 1 exit status
/home/luy/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/c/Make-lang.in:85: recipe for target 'cc1' failed
make[6]: *** [cc1] Error 1
make[6]: Leaving directory '/home/luy/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/gcc'
Makefile:4230: recipe for target 'all-gcc' failed
make[5]: *** [all-gcc] Error 2
make[5]: Leaving directory '/home/luy/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final'
Makefile:896: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/luy/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final'
Makefile:86: recipe for target '/home/luy/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/.built' failed
make[3]: *** [/home/luy/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/.built] Error 2
make[3]: Leaving directory '/home/luy/lede/toolchain/gcc/final'
time: toolchain/gcc/final/compile#1196.22#109.93#1309.97
toolchain/Makefile:98: recipe for target 'toolchain/gcc/final/compile' failed
make[2]: *** [toolchain/gcc/final/compile] Error 2
make[2]: Leaving directory '/home/luy/lede'
toolchain/Makefile:96: recipe for target '/home/luy/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/luy/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/luy/lede'
/home/luy/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

lydasia, 2020-05-14:
同错误，而且发现sdk里softether消失了

codersluy, 2020-05-18:
@lydasia 解决了吗

leungyauming, 2020-05-19:
你安裝了所需的依賴了嗎？如果可以的話，你在make指令後加V=s再跑一次，然後把結果放上來。

qjiang888, 2020-05-24:
同错误,怎么都编译不了

qjiang888, 2020-05-24:
sed -i "s/Installed-Time: .*/Installed-Time: 1590296757/" /home/qjiang/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/status
rm -rf /home/qjiang/lede/build_dir/target-mipsel_24kc_musl/root-ramips/tmp/*
rm -f /home/qjiang/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/lists/*
rm -f /home/qjiang/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/info/*.postinst*
rm -f /home/qjiang/lede/build_dir/target-mipsel_24kc_musl/root-ramips/var/lock/*.lock
rm -rf /home/qjiang/lede/build_dir/target-mipsel_24kc_musl/root-ramips/boot
find /home/qjiang/lede/build_dir/target-mipsel_24kc_musl/root-ramips/ -mindepth 1 -execdir touch -hcd "@1590296757" "{}" +
find: The relative path 'Files/WindowsApps/CanonicalGroupLimited.Ubuntu18.04onWindows_1804.2020.423.0_x64__79rhkp1fndgsc' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/qjiang/lede'
package/Makefile:108: recipe for target '/home/qjiang/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/qjiang/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/qjiang/lede'
Build failed - please re-run with -j1 to see the real error message
/home/qjiang/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1

codersluy, 2020-05-25:
@leungyauming 就是`make -j1 V=s`出错了

lydasia, 2020-05-26:
似乎是有个包没更新，已解决。

qjiang888, 2020-05-26:
大佬 是什么包啊

lydasia, 2020-05-26:
> 大佬 是什么包啊

我很菜啊，你重新按照readme来一遍吧，我是原来没问题的环境，重新clone最新代码就失败了，执行apt-get发现有更新，然后就没问题了。

codersluy, 2020-05-29:
1. `sudo apt-get update`使用ubuntu官方源
2. 保证磁盘空间充足，`df -h` 查看
3. 然后重试
楼主已成功!!!

closed 2020-05-29 by codersluy

txdywy, 2020-07-23:
最近老是这个错误❌...

zhangguanzhang, 2021-07-31:
```yaml
jobs:
  Build-OpenWrt:
    runs-on: ubuntu-20.04
```

TURIING, 2023-03-08:
我也遇到了这个错误，解决方法是往`/etc/wsl.conf`追加
[interop]
appendWindowsPath = false
原因是wsl的环境变量塞入了过多的win环境变量