coolsnowwolf/lede #4564: 修改配置后，点击应用，一直卡住

Guxing0321, 2020-05-07:
Hi Team,

最近发现一个奇怪现象
就是修改配置，点击应用过后，一直卡住，重启机器也没有用

请问，有解决的办法吗

[image]

路由型号： Newifi3 D2
固件： OpenWrt R20.3.5 / LuCI Master (git-20.062.50551-a6189c5)

BR
Robin

hcym, 2020-05-08:
尽量精简插件尽量避免用v2，

Guxing0321, 2020-05-08:
Hi hcym，

这个固件是我自己编译的
v2 是哪个插件，还是什么包，我下次把这个去掉

BR
Robin

hcym, 2020-05-08:
自己编译的就简单了，大雕默认设置很多插件，不用的都去掉，ssr不要选v2ray，就那个trojian和ss，我是很久不用新三，用ac58u和r7800跑v2ray都吃力，现在编译x86和s905，s912也就一个ssr和一个win激活

zhangguanzhang, 2020-05-09:
我也是，现在路由器卡住，无法解析dns，重启也没用，无法修改任何配置

zhangguanzhang, 2020-05-09:
我是订阅了地址后造成配置文件内容过多撑爆了空间
ssh上去后
```
cat  /etc/config/shadowsocksr
rm -f  /etc/config/shadowsocksr
vi  /etc/config/shadowsocksr
```
把多余server忽略了，其他配置粘贴上去就行了

closed 2020-07-04 by coolsnowwolf