coolsnowwolf/lede #4565: 无法解析多玩的域名(www.duowan.com)

r0achnnan9p, 2020-05-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
 [*] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
默认编译安装，没开任何其他玩意，只是配置了下PPPOE拨号，然后依然打不开多玩的网站。
查看解析记录，www.duowan.com返回的IP是127.0.0.1。
在软路由本身查询dns也是返回的这个本机地址。
补充一下：
电信分配给我的DNS是222.246.129.80，59.51.78.210。默认情况下，我解析www.duowan.com返回127.0.0.1。如下图：
[image]
[image]

但是我在openwrt-网络-Turbo ACC 网络加速设置-DNS加速中，设置这两个DNS，就能返回正常值。如下图：
image
image
[image]
[image]

这应该是一个的BUG？？？
### (2) 路由器型号和固件版本

A：
主机型号 | Intel(R) Atom(TM) CPU D2500 @ 1.86GHz : 2 Core 2 Thread
固件版本 | OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120)
内核版本 | 4.19.115

### (3) 详细日志

A：
无

SouHong, 2020-05-08:
主机型号 | HiWiFi HC5962
-- | --
架构 | MediaTek MT7621 ver:1 eco:3 x 4
固件版本 | OpenWrt  R20.4.8 / LuCI Master (git-20.117.60969-420c61a)
内核版本 | 4.14.176

一样的结果。

r0achnnan9p, 2020-05-08:
> 主机型号	HiWiFi HC5962
> 架构	MediaTek MT7621 ver:1 eco:3 x 4
> 固件版本	OpenWrt R20.4.8 / LuCI Master (git-20.117.60969-420c61a)
> 内核版本	4.14.176
> DNS加速中，设置这两个DNS也不能打开这个网。

多换几个DNS试试，114.114.114.114，223.5.5.5，1.1.1.1，8.8.8.8这些都试试，说不定就好了。

r0achnnan9p, 2020-05-23:
破案了，是湖南电信下发的2个DNS的问题，换了华硕路由器原厂固件，用这2个DNS解析出来，依然经常获得127.0.0.1的地址。

[image]

closed 2020-05-23 by r0achnnan9p