coolsnowwolf/lede #4568: 你好L大，本次修复的ipsec紧接着又出了一个bug，用手机连接ipsec听网易云瞬间卡死，导致路由重启

qq1316237299, 2020-05-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述你好L大，本次修复的ipsec紧接着又出了一个bug，用手机连接ipsec听网易云瞬间卡死，导致路由重启

### (1) 具体问题

A：你好L大，本次修复的ipsec紧接着又出了一个bug，用手机连接ipsec听网易云瞬间卡死，导致路由重启

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：Fri May  8 19:02:13 2020 kern.info kernel: [    9.756532] e1000e 0000:01:00.0: Interrupt Mode set to 1
Fri May  8 19:02:13 2020 kern.info kernel: [    9.830574] e1000e 0000:01:00.0 0000:01:00.0 (uninitialized): registered PHC clock
Fri May  8 19:02:13 2020 kern.info kernel: [    9.914324] e1000e 0000:01:00.0 eth0: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:3c
Fri May  8 19:02:13 2020 kern.info kernel: [    9.922849] e1000e 0000:01:00.0 eth0: Intel(R) PRO/1000 Network Connection
Fri May  8 19:02:13 2020 kern.info kernel: [    9.930090] e1000e 0000:01:00.0 eth0: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Fri May  8 19:02:13 2020 kern.info kernel: [    9.937334] e1000e 0000:02:00.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
Fri May  8 19:02:13 2020 kern.info kernel: [   10.010521] e1000e 0000:02:00.0 0000:02:00.0 (uninitialized): registered PHC clock
Fri May  8 19:02:13 2020 kern.info kernel: [   10.094284] e1000e 0000:02:00.0 eth1: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:3d
Fri May  8 19:02:13 2020 kern.info kernel: [   10.102654] e1000e 0000:02:00.0 eth1: Intel(R) PRO/1000 Network Connection
Fri May  8 19:02:13 2020 kern.info kernel: [   10.109897] e1000e 0000:02:00.0 eth1: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Fri May  8 19:02:13 2020 kern.info kernel: [   10.117102] e1000e 0000:03:00.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
Fri May  8 19:02:13 2020 kern.info kernel: [   10.190521] e1000e 0000:03:00.0 0000:03:00.0 (uninitialized): registered PHC clock
Fri May  8 19:02:13 2020 kern.info kernel: [   10.274324] e1000e 0000:03:00.0 eth2: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:3e
Fri May  8 19:02:13 2020 kern.info kernel: [   10.282965] e1000e 0000:03:00.0 eth2: Intel(R) PRO/1000 Network Connection
Fri May  8 19:02:13 2020 kern.info kernel: [   10.290769] e1000e 0000:03:00.0 eth2: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Fri May  8 19:02:13 2020 kern.info kernel: [   10.298469] e1000e 0000:04:00.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
Fri May  8 19:02:13 2020 kern.info kernel: [   10.383549] e1000e 0000:04:00.0 0000:04:00.0 (uninitialized): registered PHC clock
Fri May  8 19:02:13 2020 kern.info kernel: [   10.465409] e1000e 0000:04:00.0 eth3: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:3f
Fri May  8 19:02:13 2020 kern.info kernel: [   10.473820] e1000e 0000:04:00.0 eth3: Intel(R) PRO/1000 Network Connection
Fri May  8 19:02:13 2020 kern.info kernel: [   10.481064] e1000e 0000:04:00.0 eth3: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Fri May  8 19:02:13 2020 kern.info kernel: [   10.488279] e1000e 0000:05:00.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
Fri May  8 19:02:13 2020 kern.info kernel: [   10.573554] e1000e 0000:05:00.0 0000:05:00.0 (uninitialized): registered PHC clock
Fri May  8 19:02:13 2020 kern.info kernel: [   10.655404] e1000e 0000:05:00.0 eth4: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:40
Fri May  8 19:02:13 2020 kern.info kernel: [   10.663777] e1000e 0000:05:00.0 eth4: Intel(R) PRO/1000 Network Connection
Fri May  8 19:02:13 2020 kern.info kernel: [   10.671038] e1000e 0000:05:00.0 eth4: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Fri May  8 19:02:13 2020 kern.info kernel: [   10.678259] e1000e 0000:06:00.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
Fri May  8 19:02:13 2020 kern.info kernel: [   10.763508] e1000e 0000:06:00.0 0000:06:00.0 (uninitialized): registered PHC clock
Fri May  8 19:02:13 2020 kern.info kernel: [   10.845424] e1000e 0000:06:00.0 eth5: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:41
Fri May  8 19:02:13 2020 kern.info kernel: [   10.853777] e1000e 0000:06:00.0 eth5: Intel(R) PRO/1000 Network Connection
Fri May  8 19:02:13 2020 kern.info kernel: [   10.861010] e1000e 0000:06:00.0 eth5: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Fri May  8 19:02:13 2020 kern.info kernel: [   10.869467] i40e: Intel(R) Ethernet Connection XL710 Network Driver - version 2.3.2-k
Fri May  8 19:02:13 2020 kern.info kernel: [   10.877712] i40e: Copyright (c) 2013 - 2014 Intel Corporation.
Fri May  8 19:02:13 2020 kern.info kernel: [   10.884981] i40evf: Intel(R) 40-10 Gigabit Virtual Function Network Driver - version 3.2.2-k
Fri May  8 19:02:13 2020 kern.info kernel: [   10.894514] Copyright (c) 2013 - 2015 Intel Corporation.
Fri May  8 19:02:13 2020 kern.warn kernel: [   10.904935] init oaf sysctl...ok
Fri May  8 19:02:13 2020 kern.warn kernel: [   10.908722] register char dev....ok
Fri May  8 19:02:13 2020 kern.info kernel: [   10.926884] pcnet32: pcnet32.c:v1.35 21.Apr.2008 tsbogend@alpha.franken.de
Fri May  8 19:02:13 2020 kern.info kernel: [   10.935391] usbcore: registered new interface driver r8152
Fri May  8 19:02:13 2020 kern.info kernel: [   10.942821] usbcore: registered new interface driver rtl8150
Fri May  8 19:02:13 2020 kern.info kernel: [   10.953219] tulip: Linux Tulip driver version 1.1.15-NAPI (Feb 27, 2007)
Fri May  8 19:02:13 2020 kern.info kernel: [   10.961045] usbcore: registered new interface driver ums-alauda
Fri May  8 19:02:13 2020 kern.info kernel: [   10.967952] usbcore: registered new interface driver ums-cypress
Fri May  8 19:02:13 2020 kern.info kernel: [   10.974937] usbcore: registered new interface driver ums-datafab
Fri May  8 19:02:13 2020 kern.info kernel: [   10.981964] usbcore: registered new interface driver ums-freecom
Fri May  8 19:02:13 2020 kern.info kernel: [   10.989016] usbcore: registered new interface driver ums-isd200
Fri May  8 19:02:13 2020 kern.info kernel: [   10.995931] usbcore: registered new interface driver ums-jumpshot
Fri May  8 19:02:13 2020 kern.info kernel: [   11.003069] usbcore: registered new interface driver ums-karma
Fri May  8 19:02:13 2020 kern.info kernel: [   11.010003] usbcore: registered new interface driver ums-sddr09
Fri May  8 19:02:13 2020 kern.info kernel: [   11.016929] usbcore: registered new interface driver ums-sddr55
Fri May  8 19:02:13 2020 kern.info kernel: [   11.023874] usbcore: registered new interface driver ums-usbat
Fri May  8 19:02:13 2020 kern.info kernel: [   11.035773] xt_time: kernel timezone is -0000
[... 606 lines cut ...]
Fri May  8 19:04:41 2020 cron.info crond[12335]: crond (busybox 1.30.0) started, log level 5
Fri May  8 19:04:41 2020 cron.info crond[12438]: crond (busybox 1.30.0) started, log level 5
Fri May  8 19:04:42 2020 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Fri May  8 19:04:42 2020 user.notice ddns-scripts[12598]: myddns_ipv4: PID '12598' started at 2020-05-08 19:04
Fri May  8 19:04:42 2020 cron.info crond[13288]: crond (busybox 1.30.0) started, log level 5
Fri May  8 19:04:42 2020 user.warn ddns-scripts[12598]: myddns_ipv4: Service section disabled! - TERMINATE
Fri May  8 19:04:42 2020 user.warn ddns-scripts[12598]: myddns_ipv4: PID '12598' exit WITH ERROR '1' at 2020-05-08 19:04
Fri May  8 19:04:42 2020 cron.info crond[14127]: crond (busybox 1.30.0) started, log level 5
Fri May  8 19:04:43 2020 cron.info crond[14231]: crond (busybox 1.30.0) started, log level 5
Fri May  8 19:04:45 2020 user.info mwan3track[11750]: Lost 2 ping(s) on interface wan (pppoe-wan)
Fri May  8 19:05:31 2020 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Fri May  8 19:05:31 2020 daemon.notice netifd: Interface 'wan_6' is enabled
Fri May  8 19:05:31 2020 daemon.notice netifd: Interface 'wan_6' has link connectivity
Fri May  8 19:05:31 2020 daemon.notice netifd: Interface 'wan_6' is setting up now
Fri May  8 19:05:31 2020 daemon.notice netifd: Interface 'wan_6' is now up
Fri May  8 19:07:09 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri May  8 19:07:09 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri May  8 19:07:12 2020 cron.info crond[17600]: crond (busybox 1.30.0) started, log level 5
Fri May  8 19:07:13 2020 cron.info crond[18000]: crond (busybox 1.30.0) started, log level 5
Fri May  8 19:07:13 2020 cron.info crond[18109]: crond (busybox 1.30.0) started, log level 5
Fri May  8 19:07:13 2020 daemon.err dnscache[18219]: * 05/08 19:07:13| pdnsd: info: pdnsd-1.2.9b-par starting.
Fri May  8 19:07:13 2020 daemon.err dnscache[18219]: * 05/08 19:07:13| pdnsd: warning: EOF encountered while reading entry count from disk cache file.
Fri May  8 19:19:01 2020 daemon.info : 00[DMN] signal of type SIGINT received. Shutting down
Fri May  8 19:19:01 2020 authpriv.info ipsec_starter[2728]: charon stopped after 200 ms
Fri May  8 19:19:01 2020 authpriv.info ipsec_starter[2728]: ipsec starter stopped

qq1316237299, 2020-05-08:
补充 手机为1+ 6T h2os 10.0.4版本

qq1316237299, 2020-05-08:
在补充一点  就是看油管现在也是一看就自动重启

coolsnowwolf, 2020-05-09:
关闭 flow 加速试试

qq1316237299, 2020-05-09:
> 关闭 flow 加速试试

你好l大，关掉还是同样的问题。。。请知悉

qq1316237299, 2020-05-09:
Sat May  9 15:47:15 2020 kern.info kernel: [   10.008990] e1000e 0000:01:00.0 0000:01:00.0 (uninitialized): registered PHC clock
Sat May  9 15:47:15 2020 kern.info kernel: [   10.092770] e1000e 0000:01:00.0 eth0: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:3c
Sat May  9 15:47:15 2020 kern.info kernel: [   10.101559] e1000e 0000:01:00.0 eth0: Intel(R) PRO/1000 Network Connection
Sat May  9 15:47:15 2020 kern.info kernel: [   10.109271] e1000e 0000:01:00.0 eth0: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Sat May  9 15:47:15 2020 kern.info kernel: [   10.116971] e1000e 0000:02:00.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
Sat May  9 15:47:15 2020 kern.info kernel: [   10.199000] e1000e 0000:02:00.0 0000:02:00.0 (uninitialized): registered PHC clock
Sat May  9 15:47:15 2020 kern.info kernel: [   10.282729] e1000e 0000:02:00.0 eth1: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:3d
Sat May  9 15:47:15 2020 kern.info kernel: [   10.291195] e1000e 0000:02:00.0 eth1: Intel(R) PRO/1000 Network Connection
Sat May  9 15:47:15 2020 kern.info kernel: [   10.298518] e1000e 0000:02:00.0 eth1: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Sat May  9 15:47:15 2020 kern.info kernel: [   10.305755] e1000e 0000:03:00.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
Sat May  9 15:47:15 2020 kern.info kernel: [   10.378956] e1000e 0000:03:00.0 0000:03:00.0 (uninitialized): registered PHC clock
Sat May  9 15:47:15 2020 kern.info kernel: [   10.462734] e1000e 0000:03:00.0 eth2: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:3e
Sat May  9 15:47:15 2020 kern.info kernel: [   10.471135] e1000e 0000:03:00.0 eth2: Intel(R) PRO/1000 Network Connection
Sat May  9 15:47:15 2020 kern.info kernel: [   10.478378] e1000e 0000:03:00.0 eth2: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Sat May  9 15:47:15 2020 kern.info kernel: [   10.485590] e1000e 0000:04:00.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
Sat May  9 15:47:15 2020 kern.info kernel: [   10.571957] e1000e 0000:04:00.0 0000:04:00.0 (uninitialized): registered PHC clock
Sat May  9 15:47:15 2020 kern.info kernel: [   10.653849] e1000e 0000:04:00.0 eth3: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:3f
Sat May  9 15:47:15 2020 kern.info kernel: [   10.662221] e1000e 0000:04:00.0 eth3: Intel(R) PRO/1000 Network Connection
Sat May  9 15:47:15 2020 kern.info kernel: [   10.669469] e1000e 0000:04:00.0 eth3: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Sat May  9 15:47:15 2020 kern.info kernel: [   10.676674] e1000e 0000:05:00.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
Sat May  9 15:47:15 2020 kern.info kernel: [   10.761977] e1000e 0000:05:00.0 0000:05:00.0 (uninitialized): registered PHC clock
Sat May  9 15:47:15 2020 kern.info kernel: [   10.843870] e1000e 0000:05:00.0 eth4: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:40
Sat May  9 15:47:15 2020 kern.info kernel: [   10.852573] e1000e 0000:05:00.0 eth4: Intel(R) PRO/1000 Network Connection
Sat May  9 15:47:15 2020 kern.info kernel: [   10.860343] e1000e 0000:05:00.0 eth4: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Sat May  9 15:47:15 2020 kern.info kernel: [   10.868062] e1000e 0000:06:00.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
Sat May  9 15:47:15 2020 kern.info kernel: [   10.961883] e1000e 0000:06:00.0 0000:06:00.0 (uninitialized): registered PHC clock
Sat May  9 15:47:15 2020 kern.info kernel: [   11.043863] e1000e 0000:06:00.0 eth5: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:41
Sat May  9 15:47:15 2020 kern.info kernel: [   11.052229] e1000e 0000:06:00.0 eth5: Intel(R) PRO/1000 Network Connection
Sat May  9 15:47:15 2020 kern.info kernel: [   11.059487] e1000e 0000:06:00.0 eth5: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Sat May  9 15:47:15 2020 kern.info kernel: [   11.067940] i40e: Intel(R) Ethernet Connection XL710 Network Driver - version 2.3.2-k
Sat May  9 15:47:15 2020 kern.info kernel: [   11.076224] i40e: Copyright (c) 2013 - 2014 Intel Corporation.
Sat May  9 15:47:15 2020 kern.info kernel: [   11.083508] i40evf: Intel(R) 40-10 Gigabit Virtual Function Network Driver - version 3.2.2-k
Sat May  9 15:47:15 2020 kern.info kernel: [   11.093079] Copyright (c) 2013 - 2015 Intel Corporation.
Sat May  9 15:47:15 2020 kern.warn kernel: [   11.103505] init oaf sysctl...ok
Sat May  9 15:47:15 2020 kern.warn kernel: [   11.107268] register char dev....ok
Sat May  9 15:47:15 2020 kern.info kernel: [   11.125429] pcnet32: pcnet32.c:v1.35 21.Apr.2008 tsbogend@alpha.franken.de
Sat May  9 15:47:15 2020 kern.info kernel: [   11.134066] usbcore: registered new interface driver r8152
Sat May  9 15:47:15 2020 kern.info kernel: [   11.141471] usbcore: registered new interface driver rtl8150
Sat May  9 15:47:15 2020 kern.info kernel: [   11.151779] tulip: Linux Tulip driver version 1.1.15-NAPI (Feb 27, 2007)
Sat May  9 15:47:15 2020 kern.info kernel: [   11.159544] usbcore: registered new interface driver ums-alauda
Sat May  9 15:47:15 2020 kern.info kernel: [   11.166468] usbcore: registered new interface driver ums-cypress
Sat May  9 15:47:15 2020 kern.info kernel: [   11.173564] usbcore: registered new interface driver ums-datafab
Sat May  9 15:47:15 2020 kern.info kernel: [   11.180541] usbcore: registered new interface driver ums-freecom
Sat May  9 15:47:15 2020 kern.info kernel: [   11.187585] usbcore: registered new interface driver ums-isd200
[... cut ...]
Sat May  9 15:50:04 2020 cron.info crond[17694]: crond (busybox 1.30.0) started, log level 5
Sat May  9 15:50:04 2020 cron.info crond[17799]: crond (busybox 1.30.0) started, log level 5
Sat May  9 15:50:08 2020 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sat May  9 15:50:08 2020 daemon.notice netifd: Interface 'wan_6' is enabled
Sat May  9 15:50:08 2020 daemon.notice netifd: Interface 'wan_6' has link connectivity
Sat May  9 15:50:08 2020 daemon.notice netifd: Interface 'wan_6' is setting up now
Sat May  9 15:50:08 2020 daemon.notice netifd: Interface 'wan_6' is now up

qq1316237299, 2020-05-09:
Sat May  9 16:11:54 2020 kern.info kernel: [   10.913915] e1000e 0000:06:00.0 eth5: (PCI Express:2.5GT/s:Width x1) 00:e0:67:15:bd:41
Sat May  9 16:11:54 2020 kern.info kernel: [   10.922605] e1000e 0000:06:00.0 eth5: Intel(R) PRO/1000 Network Connection
Sat May  9 16:11:54 2020 kern.info kernel: [   10.930139] e1000e 0000:06:00.0 eth5: MAC: 4, PHY: 8, PBA No: FFFFFF-0FF
Sat May  9 16:11:54 2020 kern.info kernel: [   10.938960] i40e: Intel(R) Ethernet Connection XL710 Network Driver - version 2.3.2-k
Sat May  9 16:11:54 2020 kern.info kernel: [   10.947556] i40e: Copyright (c) 2013 - 2014 Intel Corporation.
Sat May  9 16:11:54 2020 kern.info kernel: [   10.954637] i40evf: Intel(R) 40-10 Gigabit Virtual Function Network Driver - version 3.2.2-k
Sat May  9 16:11:54 2020 kern.info kernel: [   10.964166] Copyright (c) 2013 - 2015 Intel Corporation.
Sat May  9 16:11:54 2020 kern.warn kernel: [   10.974361] init oaf sysctl...ok
Sat May  9 16:11:54 2020 kern.warn kernel: [   10.978139] register char dev....ok
Sat May  9 16:11:54 2020 kern.info kernel: [   10.996327] pcnet32: pcnet32.c:v1.35 21.Apr.2008 tsbogend@alpha.franken.de
Sat May  9 16:11:54 2020 kern.info kernel: [   11.004928] usbcore: registered new interface driver r8152
Sat May  9 16:11:54 2020 kern.info kernel: [   11.012424] usbcore: registered new interface driver rtl8150
Sat May  9 16:11:54 2020 kern.info kernel: [   11.022943] tulip: Linux Tulip driver version 1.1.15-NAPI (Feb 27, 2007)
Sat May  9 16:11:54 2020 kern.info kernel: [   11.030738] usbcore: registered new interface driver ums-alauda
Sat May  9 16:11:54 2020 kern.info kernel: [   11.037694] usbcore: registered new interface driver ums-cypress
Sat May  9 16:11:54 2020 kern.info kernel: [   11.044696] usbcore: registered new interface driver ums-datafab
Sat May  9 16:11:54 2020 kern.info kernel: [   11.051725] usbcore: registered new interface driver ums-freecom
Sat May  9 16:11:54 2020 kern.info kernel: [   11.058811] usbcore: registered new interface driver ums-isd200
Sat May  9 16:11:54 2020 kern.info kernel: [   11.065703] usbcore: registered new interface driver ums-jumpshot
Sat May  9 16:11:54 2020 kern.info kernel: [   11.072842] usbcore: registered new interface driver ums-karma
Sat May  9 16:11:54 2020 kern.info kernel: [   11.079662] usbcore: registered new interface driver ums-sddr09
Sat May  9 16:11:54 2020 kern.info kernel: [   11.086490] usbcore: registered new interface driver ums-sddr55
Sat May  9 16:11:54 2020 kern.info kernel: [   11.093422] usbcore: registered new interface driver ums-usbat
Sat May  9 16:11:54 2020 kern.info kernel: [   11.105280] xt_time: kernel timezone is -0000
Sat May  9 16:11:54 2020 kern.info kernel: [   11.111247] usbcore: registered new interface driver asix
Sat May  9 16:11:54 2020 kern.info kernel: [   11.117914] usbcore: registered new interface driver ax88179_178a
Sat May  9 16:11:54 2020 kern.info kernel: [   11.156980] PPP generic driver version 2.4.2
Sat May  9 16:11:54 2020 kern.info kernel: [   11.162079] PPP MPPE Compression module registered
Sat May  9 16:11:54 2020 kern.info kernel: [   11.167901] NET: Registered protocol family 24
Sat May  9 16:11:54 2020 kern.info kernel: [   11.207966] usbcore: registered new interface driver snd-usb-audio
Sat May  9 16:11:54 2020 kern.info kernel: [   11.232507] usbcore: registered new interface driver rt2800usb
Sat May  9 16:11:54 2020 kern.info kernel: [   11.241500] usbcore: registered new interface driver ath9k_htc
Sat May  9 16:11:54 2020 user.info kernel: [   11.249578] kmodloader: done loading kernel modules from /etc/modules.d/*
Sat May  9 16:11:54 2020 daemon.crit mdadm[2002]: DeviceDisappeared event detected on md device /dev/md0
Sat May  9 16:11:54 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat May  9 16:11:54 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat May  9 16:11:54 2020 user.notice : Added device handler type: 8021ad
Sat May  9 16:11:54 2020 user.notice : Added device handler type: 8021q
Sat May  9 16:11:54 2020 user.notice : Added device handler type: macvlan
Sat May  9 16:11:54 2020 user.notice : Added device handler type: veth
Sat May  9 16:11:54 2020 user.notice : Added device handler type: bridge
Sat May  9 16:11:54 2020 user.notice : Added device handler type: Network device
Sat May  9 16:11:54 2020 user.notice : Added device handler type: tunnel
Sat May  9 16:11:54 2020 daemon.notice wpa_supplicant[2207]: Successfully initialized wpa_supplicant
Sat May  9 16:11:54 2020 daemon.err block: /dev/loop0 is already mounted on /rom/overlay
Sat May  9 16:11:54 2020 kern.info kernel: [   13.129939] EXT4-fs (sda1): recovery complete
Sat May  9 16:11:54 2020 kern.info kernel: [   13.135489] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts:
Sat May  9 16:11:54 2020 daemon.err block: /dev/sdb1 is already mounted on /boot
Sat May  9 16:11:54 2020 daemon.err block: /dev/sdb2 is already mounted on /rom
Sat May  9 16:11:54 2020 kern.info kernel: [   13.157987] EXT4-fs (sdb3): recovery complete
Sat May  9 16:11:54 2020 kern.info kernel: [   13.163687] EXT4-fs (sdb3): mounted filesystem with ordered data mode. Opts:
Sat May  9 16:11:54 2020 daemon.notice procd: /etc/rc.d/S50autoreboot: Command failed: Not found
Sat May  9 16:11:54 2020 daemon.notice procd: /etc/rc.d/S50autoreboot: Auto REBOOT has started.
Sat May  9 16:11:54 2020 cron.info crond[2301]: crond (busybox 1.30.0) started, log level 5
[... cut ...]
Sat May  9 16:23:48 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat May  9 16:23:48 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat May  9 16:23:51 2020 cron.info crond[8653]: crond (busybox 1.30.0) started, log level 5
Sat May  9 16:23:51 2020 cron.info crond[9063]: crond (busybox 1.30.0) started, log level 5
Sat May  9 16:23:52 2020 cron.info crond[9167]: crond (busybox 1.30.0) started, log level 5
Sat May  9 16:24:46 2020 user.warn ddns-scripts[16965]: www: Updating IP at DDNS provider failed - starting retry 1/0

keivenliao, 2020-05-11:
我个人建议用 softether   来跑  ipsec  用了3、4年了， 很稳定。

qq1316237299, 2020-05-12:
qaq

evilchris, 2020-05-13:
连上后打开youtube
路由器就会重启

lydasia, 2020-05-14:
有个朋友遇到同样问题，自己一直用的softether，但是最新的sdk里消失了

qq1316237299, 2020-05-22:
你好，l大，，最新的还是有这个自动重启的问题，今天刚更新的。
Fri May 22 19:05:42 2020 kern.info kernel: [   11.073168] usbcore: registered new interface driver r8152
Fri May 22 19:05:42 2020 kern.info kernel: [   11.080656] usbcore: registered new interface driver rtl8150
Fri May 22 19:05:42 2020 kern.info kernel: [   11.091056] tulip: Linux Tulip driver version 1.1.15-NAPI (Feb 27, 2007)
Fri May 22 19:05:42 2020 kern.info kernel: [   11.098856] usbcore: registered new interface driver ums-alauda
Fri May 22 19:05:42 2020 kern.info kernel: [   11.105816] usbcore: registered new interface driver ums-cypress
Fri May 22 19:05:42 2020 kern.info kernel: [   11.112851] usbcore: registered new interface driver ums-datafab
Fri May 22 19:05:42 2020 kern.info kernel: [   11.119875] usbcore: registered new interface driver ums-freecom
Fri May 22 19:05:42 2020 kern.info kernel: [   11.126962] usbcore: registered new interface driver ums-isd200
Fri May 22 19:05:42 2020 kern.info kernel: [   11.133894] usbcore: registered new interface driver ums-jumpshot
Fri May 22 19:05:42 2020 kern.info kernel: [   11.141046] usbcore: registered new interface driver ums-karma
Fri May 22 19:05:42 2020 kern.info kernel: [   11.147934] usbcore: registered new interface driver ums-sddr09
Fri May 22 19:05:42 2020 kern.info kernel: [   11.154837] usbcore: registered new interface driver ums-sddr55
Fri May 22 19:05:42 2020 kern.info kernel: [   11.161753] usbcore: registered new interface driver ums-usbat
Fri May 22 19:05:42 2020 kern.info kernel: [   11.173637] xt_time: kernel timezone is -0000
Fri May 22 19:05:42 2020 kern.info kernel: [   11.179647] usbcore: registered new interface driver asix
Fri May 22 19:05:42 2020 kern.info kernel: [   11.186244] usbcore: registered new interface driver ax88179_178a
Fri May 22 19:05:42 2020 kern.info kernel: [   11.224993] PPP generic driver version 2.4.2
Fri May 22 19:05:42 2020 kern.info kernel: [   11.230110] PPP MPPE Compression module registered
Fri May 22 19:05:42 2020 kern.info kernel: [   11.235931] NET: Registered protocol family 24
Fri May 22 19:05:42 2020 kern.info kernel: [   11.275823] usbcore: registered new interface driver snd-usb-audio
Fri May 22 19:05:42 2020 kern.info kernel: [   11.301027] usbcore: registered new interface driver rt2800usb
Fri May 22 19:05:42 2020 kern.info kernel: [   11.309995] usbcore: registered new interface driver ath9k_htc
Fri May 22 19:05:42 2020 user.info kernel: [   11.318049] kmodloader: done loading kernel modules from /etc/modules.d/*
Fri May 22 19:05:42 2020 daemon.crit mdadm[2003]: DeviceDisappeared event detected on md device /dev/md0
Fri May 22 19:05:42 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri May 22 19:05:42 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri May 22 19:05:42 2020 daemon.notice wpa_supplicant[2208]: Successfully initialized wpa_supplicant
Fri May 22 19:05:42 2020 user.notice : Added device handler type: 8021ad
Fri May 22 19:05:42 2020 user.notice : Added device handler type: 8021q
Fri May 22 19:05:42 2020 user.notice : Added device handler type: macvlan
Fri May 22 19:05:42 2020 user.notice : Added device handler type: veth
Fri May 22 19:05:42 2020 user.notice : Added device handler type: bridge
Fri May 22 19:05:42 2020 user.notice : Added device handler type: Network device
Fri May 22 19:05:42 2020 user.notice : Added device handler type: tunnel
Fri May 22 19:05:42 2020 daemon.err block: /dev/loop0 is already mounted on /rom/overlay
Fri May 22 19:05:42 2020 kern.info kernel: [   13.164235] EXT4-fs (sda1): recovery complete
Fri May 22 19:05:42 2020 kern.info kernel: [   13.169821] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts:
Fri May 22 19:05:42 2020 daemon.err block: /dev/sdb1 is already mounted on /boot
Fri May 22 19:05:42 2020 daemon.err block: /dev/sdb2 is already mounted on /rom
Fri May 22 19:05:42 2020 kern.info kernel: [   13.184673] EXT4-fs (sdb3): recovery complete
Fri May 22 19:05:42 2020 kern.info kernel: [   13.189652] EXT4-fs (sdb3): mounted filesystem with ordered data mode. Opts:
Fri May 22 19:05:42 2020 daemon.notice procd: /etc/rc.d/S50autoreboot: Command failed: Not found
Fri May 22 19:05:42 2020 daemon.notice procd: /etc/rc.d/S50autoreboot: Auto REBOOT has started.
Fri May 22 19:05:42 2020 cron.info crond[2302]: crond (busybox 1.30.0) started, log level 5
Fri May 22 19:05:43 2020 kern.info kernel: [   13.465269] 8021q: adding VLAN 0 to HW filter on device eth0
Fri May 22 19:05:43 2020 kern.info kernel: [   13.471739] br-lan: port 1(eth0) entered blocking state
Fri May 22 19:05:43 2020 kern.info kernel: [   13.477651] br-lan: port 1(eth0) entered disabled state
Fri May 22 19:05:43 2020 kern.info kernel: [   13.483670] device eth0 entered promiscuous mode
Fri May 22 19:05:43 2020 kern.info kernel: [   13.489184] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
Fri May 22 19:05:43 2020 daemon.notice netifd: Interface 'lan' is enabled
Fri May 22 19:05:43 2020 daemon.notice netifd: Interface 'lan' is setting up now
Fri May 22 19:05:43 2020 daemon.notice netifd: Interface 'lan' is now up
Fri May 22 19:05:43 2020 daemon.notice procd: /etc/rc.d/S60dbus: Unknown group "netdev" in message bus configuration file
Fri May 22 19:05:43 2020 daemon.notice procd: /etc/rc.d/S60flowoffload: Start DNS Cache
Fri May 22 19:05:43 2020 daemon.notice procd: /etc/rc.d/S60flowoffload: net.ipv4.tcp_congestion_control = bbr
[... cut ...]
Fri May 22 19:24:08 2020 daemon.err uhttpd[2368]:
Fri May 22 19:24:08 2020 daemon.err uhttpd[2368]:      0K .......... .......... .......... .......... ..........  864K
Fri May 22 19:24:08 2020 daemon.err uhttpd[2368]:     50K .                                                      21.1M=0.06s
Fri May 22 19:24:08 2020 daemon.err uhttpd[2368]:
Fri May 22 19:24:08 2020 daemon.err uhttpd[2368]: 2020-05-22 19:24:08 (893 KB/s) - written to stdout [52992]
Fri May 22 19:24:08 2020 daemon.err uhttpd[2368]:

qq1316237299, 2020-05-24:
11

RayleiLi, 2020-05-31:
我也有这样对问题

closed 2020-07-04 by coolsnowwolf