coolsnowwolf/lede #4573: 编译lxml.etree出错

gdw1986, 2020-05-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

打开了一些python3的模块，编译报错。

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：running build_ext
building 'lxml.etree' extension
x86_64-openwrt-linux-musl-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/david/lede/build_dir/target-x86_64_musl/Python-3.8.2:Python-3.8.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/david/lede/build_dir/target-x86_64_musl/Python-3.8.2:Python-3.8.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/david/lede/build_dir/target-x86_64_musl/Python-3.8.2:Python-3.8.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -I/home/david/lede/staging_dir/target-x86_64_musl/usr/include -I/home/david/lede/staging_dir/target-x86_64_musl/include -I/home/david/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/david/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/david/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/david/lede/build_dir/target-x86_64_musl/pypi/python-lxml-python3/lxml-4.4.2:lxml-4.4.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/david/lede/staging_dir/target-x86_64_musl/usr/include -I/home/david/lede/staging_dir/target-x86_64_musl/include -I/home/david/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/david/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/david/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -I/home/david/lede/staging_dir/target-x86_64_musl/usr/include/python3.8 -fPIC -DCYTHON_CLINE_IN_TRACEBACK=0 -I/home/david/lede/staging_dir/target-x86_64_musl/usr/include/ -I/home/david/lede/staging_dir/target-x86_64_musl/usr/include/libxml2 -Isrc -Isrc/lxml/includes -I/home/david/lede/staging_dir/target-x86_64_musl/usr/include/python3.8 -c src/lxml/etree.c -o build/temp.linux2-3.8/src/lxml/etree.o -w -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/david/lede/build_dir/target-x86_64_musl/pypi/python-lxml-python3/lxml-4.4.2:lxml-4.4.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro
cc1: note: someone does not honour COPTS correctly, passed 5 times
src/lxml/etree.c:732:10: fatal error: libexslt/exslt.h: No such file or directory
 #include "libexslt/exslt.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
Compile failed: command 'x86_64-openwrt-linux-musl-gcc' failed with exit status 1
cc -I/home/david/lede/staging_dir/target-x86_64_musl/usr/include/ -I/home/david/lede/staging_dir/target-x86_64_musl/usr/include/libxml2 -I/usr/include/libxml2 -c /home/david/lede/tmp/xmlXPathInituj0yue6t.c -o home/david/lede/tmp/xmlXPathInituj0yue6t.o
cc home/david/lede/tmp/xmlXPathInituj0yue6t.o -L/home/david/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/david/lede/staging_dir/target-x86_64_musl/lib -lxml2 -o a.out
/usr/bin/ld: warning: libc.so, needed by /home/david/lede/staging_dir/target-x86_64_musl/usr/lib/libxml2.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: a.out: hidden symbol `stat' in /usr/lib/x86_64-linux-gnu/libc_nonshared.a(stat.oS) is referenced by DSO
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
*********************************************************************************
Could not find function xmlCheckVersion in library libxml2. Is libxml2 installed?
*********************************************************************************
error: command 'x86_64-openwrt-linux-musl-gcc' failed with exit status 1
Makefile:89: recipe for target '/home/david/lede/build_dir/target-x86_64_musl/pypi/python-lxml-python3/lxml-4.4.2/.built' failed
make[3]: *** [/home/david/lede/build_dir/target-x86_64_musl/pypi/python-lxml-python3/lxml-4.4.2/.built] Error 1
make[3]: Leaving directory '/home/david/lede/feeds/packages/lang/python/python-lxml'
time: package/feeds/packages/python-lxml/python3/compile#1.36#0.42#2.22
package/Makefile:111: recipe for target 'package/feeds/packages/python-lxml/compile' failed
make[2]: *** [package/feeds/packages/python-lxml/compile] Error 2
make[2]: Leaving directory '/home/david/lede'
package/Makefile:107: recipe for target '/home/david/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/david/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/david/lede'
/home/david/lede/include/toplevel.mk:225: recipe for target 'world' failed

gdw1986, 2020-06-11:
xxx

closed 2020-06-11 by gdw1986

hzjnet, 2022-07-15:
看来这个鬼一直存在