coolsnowwolf/lede #4577: R20.5.9 安装luci-compat后luci组件异常

zyc199847, 2020-05-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R20.5.9 安装luci-compat后luci组件异常，提示insufficient permissions to read uci configuration。卸载luci-compat后luci组件崩溃。

虽然R20.5.9下依赖于luci-compat的app可以在未安装luci-compat时正常使用，但是通过opkg安装部分软件包时会自动安装此依赖，导致luci异常。R20.4.8版会由于文件冲突无法安装luci-compat。

### (2) 路由器型号和固件版本

A：x86_64，OpenWrt R20.5.9 / LuCI Master (git-20.117.60969-420c61a)

### (3) 详细日志

A：NA

lphgor, 2020-05-10:
这里用的是旧版的luci，不需要luci-compat，也不兼容官方源的软件。
luci-compat加上旧版的软件可以用在新版的luci上面，你反过来就不行。

lphgor, 2020-05-10:
这里用的是旧版的luci，不需要luci-compat，也不兼容官方源的软件，官方源都已经升级到新版luci了。
luci-compat加上旧版的软件可以用在新版的luci上面，你反过来就不行。

zyc199847, 2020-05-11:
> 这里用的是旧版的luci，不需要luci-compat，也不兼容官方源的软件，官方源都已经升级到新版luci了。
> luci-compat加上旧版的软件可以用在新版的luci上面，你反过来就不行。

确实是这样，只能说要避免安装官方源软件，反正装了luci-compat后luci就炸了，还是比较坑的

closed 2020-05-20 by coolsnowwolf