coolsnowwolf/lede #4581: 重启卡住在Synchronizing SCSI cache

Jefffish09, 2020-05-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
_当插上u盘，并挂在成功后，如果软重启（比如web端选择重启）则会卡住在Synchronizing SCSI cache，一直无法重启。当拔掉u盘，又可以正常重启了。_

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
_当插上u盘，并挂在成功后，如果软重启则会卡住在Synchronizing SCSI cache，一直无法重启。当拔掉u盘，又可以正常重启了。_

### (2) 路由器型号和固件版本

_固件是最新版，CPU是j1900的软路由_

### (3) 详细日志

_如图：_
[image]

Jefffish09, 2020-05-12:
在网上搜索了一下，不是我一个人的个例，也有人遇到过:
https://koolshare.cn/thread-154078-1-1.html
https://koolshare.cn/thread-152054-1-1.html
https://koolshare.cn/thread-134176-1-1.html
https://www.right.com.cn/forum/thread-3441794-1-1.html

closed 2020-07-04 by coolsnowwolf

kenvix, 2023-11-15:
同样遇到了此问题，如果是近期构建的（默认内核版本 6.1），降级内核版本到 5.15 可以解决。