coolsnowwolf/lede #4584: 竞斗云使用 ath10k-firmware-qca4019驱动的wifi问题

zheshifandian, 2020-05-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：2.4gwifi和5gwifi只能打开其中一个
开2.4没5g，开5g没2.4

### (2) 路由器型号和固件版本

A：竞斗云2.0-r619ac+最新源码

### (3) 详细日志

A：
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.176 (zheshifandian@HP-Pavilion-15) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r2534-d9704c2b)) #0 SMP Sun May 10 02:19:36 2020
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: P&W R619AC 128M
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 130560
[    0.000000] free_area_init_node: node 0, pgdat c0a279c0, node_mem_map dfb79000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x3e8 with crng_init=0
[    0.000000] percpu: Embedded 15 pages/cpu s29196 r8192 d24052 u61440
[    0.000000] pcpu-alloc: s29196 r8192 d24052 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 rootfstype=squashfs
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 509312K/522240K available (4590K kernel code, 160K rwdata, 1240K rodata, 1024K init, 225K bss, 12928K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0208000 - 0xc077b868   (5583 kB)
[    0.000000]       .init : 0xc0900000 - 0xc0a00000   (1024 kB)
[    0.000000]       .data : 0xc0a00000 - 0xc0a282c0   ( 161 kB)
[    0.000000]        .bss : 0xc0a2a000 - 0xc0a62598   ( 226 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000024] Switching to timer-based delay loop, resolution 20ns
[    0.000245] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000265] pid_max: default: 32768 minimum: 301
[    0.000434] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000451] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001216] CPU: Testing write buffer coherency: ok
[    0.001981] Setting up static identity map for 0x80300000 - 0x80300060
[    0.002132] Hierarchical SRCU implementation.
[    0.002818] smp: Bringing up secondary CPUs ...
[    0.005666] smp: Brought up 1 node, 4 CPUs
[... 246 lines cut ...]
[   32.252508] ath: Country alpha2 being used: AU
[   32.252511] ath: Regpair used: 0x3b
[   32.252516] ath: regdomain 0x8024 dynamically updated by user
[   32.393257] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   32.394184] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   35.035725] ath10k_ahb a000000.wifi: unsupported HTC service id: 1536
[   35.198103] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   35.203652] br-lan: port 2(wlan0) entered blocking state
[   35.203710] br-lan: port 2(wlan0) entered disabled state
[   35.208801] device wlan0 entered promiscuous mode
[   35.590156] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   35.590373] br-lan: port 2(wlan0) entered blocking state
[   35.595639] br-lan: port 2(wlan0) entered forwarding state
[  237.722425] device wlan0 left promiscuous mode
[  237.722587] br-lan: port 2(wlan0) entered disabled state
[  237.776025] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 1
[  237.776115] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 1
[  239.974112] ath10k_ahb a800000.wifi: unsupported HTC service id: 1536
[  240.099459] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  240.102954] br-lan: port 2(wlan1) entered blocking state
[  240.104196] br-lan: port 2(wlan1) entered disabled state
[  240.110069] device wlan1 entered promiscuous mode
[  240.612431] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  240.612945] br-lan: port 2(wlan1) entered blocking state
[  240.617909] br-lan: port 2(wlan1) entered forwarding state

garygary1, 2020-05-11:
遇到了同样问题。

Hey-Leo, 2020-05-12:
极路由HC5962也有问题，会出现无线未关联

PR #4590 (2020-05-12): Revert "ipq40xx: update ath10k firmware"
Revert this commit due to broken wireless, fix #4584
closed 2020-05-12 by coolsnowwolf