coolsnowwolf/lede #4588: 20200511最新代码编译不成功

jovishow, 2020-05-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述
20200511下午拉取最新的代码首次编译，0510已尝试编译过，但未能成功，之前的报错是依赖的问题，看了isses后我在menuconfig中把报错的项给关了...make clean，之前编译的报错信息与本次的一样，直到昨天下午发现更新了代码，重新拉取编译

编译执行过程
①sudo apt-get update
②sudo apt-get -y install...
③./scripts/feeds update -a && ./scripts/feeds install -a
④nohup make -j8 download V=s >> install.log 2>&1 &
⑤nohup make -j1 V=s >> install.log 2>&1 &

### (1) 具体问题

A：编译不成功

### (2) 路由器型号和固件版本

A：K2 Phicomm PSG1218 rev.A

### (3) 详细日志
Configuring strongswan-mod-xauth-generic.
Configuring luci-app-ipsec-vpnd.
Configuring luci-i18n-aria2-zh-cn.
Configuring urandom-seed.
Configuring luci-app-cshark.
Configuring luci-i18n-qos-zh-cn.
Configuring kmod-mppe.
Configuring ppp.
Configuring kmoCollected errors:
 * check_data_file_clashes: Package libustream-openssl20150806 wants to install file /home/jovishow/detmp/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls20150806
 * opkg_install_cmd: Cannot install package libustream-openssl20150806.
d-nft-netdev.
Configuring socat.
Configuring kmod-leds-gpio.
Configuring luci-i18n-ssr-plus-zh-cn.
Configuring kmod-gpio-button-hotplug.
Configuring logd.
Configuring luci-app-accesscontrol.
Configuring luci-i18n-accesscontrol-zh-cn.
Configuring libtirpc.
Configuring libpopt.
Configuring samba4-libs.
Configuring nftables.
Configuring nft-qos.
Configuring luci-app-nft-qos.
Configuring luci-app-p910nd.
Configuring luci-i18n-p910nd-zh-cn.
Configuring luci-i18n-nft-qos-zh-cn.
Configuring luci-i18n-ipsec-vpnd-zh-cn.
Configuring mac-telnet-client.
Configuring rpcd-mod-rpcsys.
Configuring attendedsysupgrade-common.
Configuring luci-app-attendedsysupgrade.
Configuring luci-i18n-ddns-zh-cn.
Configuring kmod-nf-nathelper-extra.
Configuring luci-app-xlnetacc.
Configuring luci-i18n-adbyby-plus-zh-cn.
Configuring luci-app-ramfree.
Configuring luci-i18n-ramfree-zh-cn.
Configuring luci-app-wol.
Configuring luci-i18n-wol-zh-cn.
Configuring luci-app-upnp.
Configuring luci-i18n-mwan3-zh-cn.
Configuring default-settings.
Configuring luci-i18n-ttyd-zh-cn.
Configuring luci-theme-netgear.
Configuring luci-i18n-base-zh-cn.
Configuring kmod-rt2800-soc.
Configuring kmod-mt76x2.
Configuring luci-i18n-upnp-zh-cn.
Configuring luci-i18n-ntpc-zh-cn.
Configuring acme.
Configuring luci-app-acme.
Configuring urngd.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/jovishow/detmp/lede'
package/Makefile:108: recipe for target '/home/jovishow/detmp/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/jovishow/detmp/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/jovishow/detmp/lede'
/home/jovishow/detmp/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

A:
Configuring kmoCollected errors:
 * check_data_file_clashes: Package libustream-openssl20150806 wants to install file /home/jovishow/detmp/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/libustream-ssl.so
        But that file is already provided by package  * libustream-mbedtls20150806
 * opkg_install_cmd: Cannot install package libustream-openssl20150806.

B：
Configuring luci-i18n-nlbwmon-zh-cn.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/jovishow/detmp/lede'
package/Makefile:108: recipe for target '/home/jovishow/detmp/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/jovishow/detmp/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/jovishow/detmp/lede'
/home/jovishow/detmp/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

jovishow, 2020-05-12:
关于A问题，libustream-openssl和libustream-mbedtls中间去掉一个

bin20088, 2020-05-12:
> 关于A问题，libustream-openssl和libustream-mbedtls中间去掉一个

https://github.com/coolsnowwolf/lede/issues/4520  和这个基本差不多

jovishow, 2020-05-12:
在编译选项目录Libraries里面，我选择了libwebsockets-openss，
libwebsockets-mbedtls. libwebsockets (mbedTLS) 
 libwebsockets-openssl. libwebsockets (OpenSSL)  

> > 关于A问题，libustream-openssl和libustream-mbedtls中间去掉一个
> 
> #4520 和这个基本差不多
在编译选项目录Libraries里面，我选择了libwebsockets-openssl，看了#4520后，还要去正在尝试去掉-full，就得去掉ttyd？
libwebsockets-mbedtls. libwebsockets (mbedTLS) 
libwebsockets-openssl. libwebsockets (OpenSSL)

jovishow, 2020-05-12:
> > 关于A问题，libustream-openssl和libustream-mbedtls中间去掉一个
> 
> #4520 和这个基本差不多

在编译选项目录Libraries里面，我选择了libwebsockets-openssl，看了#4520后，还要去正在尝试去掉-full，就得去掉ttyd？
libwebsockets-mbedtls. libwebsockets (mbedTLS)
libwebsockets-openssl. libwebsockets (OpenSSL)

9l, 2020-05-13:
只改target，默认设置，都不能编译成功么，别改其他的设置，也不要增加或删除luci app

jovishow, 2020-05-13:
> 只改target，默认设置，都不能编译成功么，别改其他的设置，也不要增加或删除luci app
> 
> 在哪里编译自己的电脑还是远程服务器，自己的电脑编译要挂梯子的，不然有些包下载不了会出错

在gcp上编译的，重新make clean后，重新编译问题A和B的报错依然存在，我今天试下只改target，看是否能编译成功

jovishow, 2020-05-13:
> 只改target，默认设置，都不能编译成功么，别改其他的设置，也不要增加或删除luci app
> 
> 在哪里编译自己的电脑还是远程服务器，自己的电脑编译要挂梯子的，不然有些包下载不了会出错

只改了如下编译成功了
①target system:mediatek ralink mips
②subtarget:mt7620
③target profile:phicomm psg1218 rev.ax
④luci->applications->ssr_p、v2ray、trojan、Ss V2ray Plugin
其它未配置

jovishow, 2020-05-13:
@coolsnowwolf 

> > 只改target，默认设置，都不能编译成功么，别改其他的设置，也不要增加或删除luci app
> > 在哪里编译自己的电脑还是远程服务器，自己的电脑编译要挂梯子的，不然有些包下载不了会出错
> 
> 只改了如下编译成功了
> ①target system:mediatek ralink mips
> ②subtarget:mt7620
> ③target profile:phicomm psg1218 rev.ax
> ④luci->applications->ssr_p、v2ray、trojan、Ss V2ray Plugin
> 其它未配置

@coolsnowwolf 14.8M的固件安装不能安装，是v2ray大吗？

jovishow, 2020-05-13:
求问 怎么将包体控制在7.68M以内，还是有改内存的方式

jovishow, 2020-05-14:
仅勾选trojan 空间16G 包体8.1M 花时1小时x分钟
LuCI ---> Applications ---> Include Trojan	#trojan协议  
去除
luci-app-ddns. LuCI Support for Dynamic DNS Client (ddns-scripts 
luci-app-flowoffload. LuCI support for Flow Offloa
luci-app-vlmcsd. LuCI page for KM 
luci-app-wol. LuCI Support for Wake-on-LA
<M> luci-app-vsftpd. LuCI support for VSFTP 
<M> luci-app-wrtbwmon. LuCI support for Wrtbwmo  
<M> luci-app-filetransfer. LuCI page for IPK uploa
<M> luci-app-adbyby-plus. LuCI support for Adbyb 

如以上瘦身，固件整理大小为7.61M，求问<M>被编译但不进固件这些软件包的输入路径是；lede/bin/packages/mipsel_24kc吗？

jovishow, 2020-05-15:
固件可能太大了，保存配置重启后会清空，我试了多次配置上瘦身，编译的最少固件是7.6m

closed 2020-07-04 by coolsnowwolf