coolsnowwolf/lede #4592: 端口转发，发现个有意思的bug

82kg, 2020-05-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
有类似的issue，但是没下文。

## 2. 详细叙述

假如我有个http代理，开在 lan:666 端口上。

端口转发里，wan:666 --> lan:666  转发的端口设置成一样的，转发速度就正常。达到3.5MB/S

如果  wan:777-->lan:666，外网端口如果设置的不一样。转发速度就只有125kb/s

如果关闭 Flow Offloading ，访问wan:777的速度 就正常了。

请教下，这是bug么？ 好奇怪的操作。

### (2) 路由器型号和固件版本

A：X86软路由

hyyz17200, 2020-05-12:
woc 是真的，终于明白为什么FlowOffloading会让我的远程桌面卡死，因为我转发的时候也改了端口号。

hxlls, 2020-05-12:
早就有了，FlowOffloading和端口转发还有一堆东西有冲突，我早就直接关闭完事，那东西没用出什么提升，再X86下，资源也不需要考虑太多

eallion, 2020-05-12:
难道说我找到了游戏被迫挂机的原因

ricksuzade-maker, 2020-05-12:
确实是呀，我似乎也找到了我远程桌面卡死的原因了，万万没想到。

kkxkx, 2020-05-13:
没有大佬维护FlowOffloading了吗？

caicaicai21, 2020-05-13:
很早之前就有这种情况，只是一直也没有修复，所以我直接就关闭了。

coolsnowwolf, 2020-05-13:
老问题了，等上游修复吧

hxlls, 2020-05-13:
好像这个问题还会影响docker 的一些玩意，但是我很少用docker 没得测试

82kg, 2020-05-14:
> 早就有了，FlowOffloading和端口转发还有一堆东西有冲突，我早就直接关闭完事，那东西没用出什么提升，再X86下，资源也不需要考虑太多

FlowOffloading 真的没啥提升吗，好像也没找到评测。如果没啥明显帮助，我也准备关了

82kg, 2020-05-14:
> 很早之前就有这种情况，只是一直也没有修复，所以我直接就关闭了。

这FlowOffloading  是可以让cpu占用变低么？

a1392429292, 2020-05-16:
看了这个issue后，我觉得我之前提问的  https://github.com/coolsnowwolf/lede/issues/4533  可以放弃食用了。。。

jzhou404, 2020-05-22:
谢谢你的提醒， 我终于知道我以前群晖 ds file速度慢的原因了

果断关闭了

hyyz17200, 2020-05-24:
但是内网转发端口号一致的话，并不会降速。 我测试了一周多了，包括RDP的各种服务都没有问题。

LGA1150, 2020-05-26:
> 老问题了，等上游修复吧

升级到 5.4 内核吧 :doge:

82kg, 2020-06-28:
> > 老问题了，等上游修复吧
> 
> 升级到 5.4 内核吧 :doge:

5.4修复了此bug了？

82kg, 2020-06-28:
waiting for core 5.4

closed 2020-06-28 by 82kg

chancat87, 2021-01-22:
还有后续吗?

yyysuo, 2023-04-14:
> 还有后续吗?

到2023年4月，还是没解决。