coolsnowwolf/lede #4595: mt7621+mt7603+mt76x2配置，2.4G wifi和5G wifi无法共存。

shikiroot, 2020-05-12:
小米R3Gv1，mt7621+mt7603+mt76x2的配置。今天编译完之后进入wifi设置，初始状态显示两个wifi都正常。修改了一下之后就变成只能打开一个，如果在这种状态下去再启用被关闭的那个，两个都会被关闭。猜测是[839522a](https://github.com/coolsnowwolf/lede/commit/839522a0c7e0209a828019750c7e9d511c7ed7cc)或者[b65f1eb](https://github.com/coolsnowwolf/lede/commit/b65f1ebf0089a2fa0d3e1ad30efc21385a013139)更新导致的。

## system log
```
Tue May 12 17:13:22 2020 daemon.notice netifd: Network device 'wlan0' link is up
Tue May 12 17:14:16 2020 kern.info kernel: [  481.945871] device wlan0 left promiscuous mode
Tue May 12 17:14:16 2020 kern.info kernel: [  481.950707] br-lan: port 2(wlan0) entered disabled state
Tue May 12 17:14:16 2020 daemon.notice hostapd: Remove interface 'wlan0'
Tue May 12 17:14:16 2020 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Tue May 12 17:14:16 2020 daemon.notice hostapd: wlan0: AP-DISABLED
Tue May 12 17:14:16 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Tue May 12 17:14:16 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Tue May 12 17:14:16 2020 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument
Tue May 12 17:14:17 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Tue May 12 17:14:17 2020 kern.info kernel: [  483.109437] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Tue May 12 17:14:17 2020 kern.info kernel: [  483.123258] br-lan: port 2(wlan0) entered blocking state
Tue May 12 17:14:17 2020 kern.info kernel: [  483.128725] br-lan: port 2(wlan0) entered disabled state
Tue May 12 17:14:17 2020 kern.info kernel: [  483.134623] device wlan0 entered promiscuous mode
Tue May 12 17:14:17 2020 kern.info kernel: [  483.139657] br-lan: port 2(wlan0) entered blocking state
Tue May 12 17:14:17 2020 kern.info kernel: [  483.145040] br-lan: port 2(wlan0) entered forwarding state
Tue May 12 17:14:17 2020 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Tue May 12 17:14:17 2020 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Tue May 12 17:14:17 2020 kern.info kernel: [  483.156492] br-lan: port 2(wlan0) entered disabled state
Tue May 12 17:14:17 2020 daemon.notice netifd: radio0 (8426): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Tue May 12 17:14:17 2020 daemon.notice netifd: radio0 (8426): Command failed: Invalid argument
Tue May 12 17:14:18 2020 daemon.notice hostapd: 20/40 MHz operation not permitted on channel pri=1 sec=5 based on overlapping BSSes
Tue May 12 17:14:18 2020 daemon.err hostapd: Using interface wlan0 with hwaddr 78:11:dc:42:59:d1 and ssid "XM"
Tue May 12 17:14:18 2020 kern.info kernel: [  484.549364] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Tue May 12 17:14:18 2020 kern.info kernel: [  484.556067] br-lan: port 2(wlan0) entered blocking state
Tue May 12 17:14:18 2020 kern.info kernel: [  484.561368] br-lan: port 2(wlan0) entered forwarding state
Tue May 12 17:14:18 2020 daemon.notice netifd: Network device 'wlan0' link is up
Tue May 12 17:14:18 2020 daemon.notice hostapd: wlan0: interface state HT_SCAN->ENABLED
Tue May 12 17:14:18 2020 daemon.notice hostapd: wlan0: AP-ENABLED
Tue May 12 17:14:24 2020 daemon.notice hostapd: Remove interface 'wlan0'
Tue May 12 17:14:24 2020 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Tue May 12 17:14:24 2020 daemon.notice hostapd: wlan0: AP-DISABLED
Tue May 12 17:14:24 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Tue May 12 17:14:24 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Tue May 12 17:14:24 2020 kern.info kernel: [  489.955851] device wlan0 left promiscuous mode
Tue May 12 17:14:24 2020 kern.info kernel: [  489.960471] br-lan: port 2(wlan0) entered disabled state
Tue May 12 17:14:24 2020 daemon.notice netifd: Network device 'wlan0' link is down
Tue May 12 17:14:25 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Tue May 12 17:14:25 2020 kern.info kernel: [  491.219264] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Tue May 12 17:14:25 2020 kern.info kernel: [  491.229139] br-lan: port 2(wlan1) entered blocking state
Tue May 12 17:14:25 2020 kern.info kernel: [  491.234449] br-lan: port 2(wlan1) entered disabled state
Tue May 12 17:14:25 2020 kern.info kernel: [  491.240224] device wlan1 entered promiscuous mode
Tue May 12 17:14:25 2020 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Tue May 12 17:14:25 2020 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
Tue May 12 17:14:25 2020 daemon.notice netifd: radio1 (8767): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Tue May 12 17:14:25 2020 daemon.notice netifd: radio1 (8767): Command failed: Invalid argument
Tue May 12 17:14:26 2020 daemon.err hostapd: Using interface wlan1 with hwaddr 78:11:dc:42:59:d2 and ssid "TP-LINK"
Tue May 12 17:14:26 2020 kern.info kernel: [  492.288454] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Tue May 12 17:14:26 2020 kern.info kernel: [  492.295111] br-lan: port 2(wlan1) entered blocking state
Tue May 12 17:14:26 2020 kern.info kernel: [  492.300563] br-lan: port 2(wlan1) entered forwarding state
Tue May 12 17:14:26 2020 daemon.notice hostapd: wlan1: interface state HT_SCAN->ENABLED
Tue May 12 17:14:26 2020 daemon.notice hostapd: wlan1: AP-ENABLED
Tue May 12 17:14:26 2020 daemon.notice netifd: Network device 'wlan1' link is up
Tue May 12 17:14:40 2020 kern.info kernel: [  506.248256] device wlan1 left promiscuous mode
Tue May 12 17:14:40 2020 kern.info kernel: [  506.252899] br-lan: port 2(wlan1) entered disabled state
Tue May 12 17:14:40 2020 daemon.notice hostapd: Remove interface 'wlan1'
Tue May 12 17:14:40 2020 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Tue May 12 17:14:40 2020 daemon.notice hostapd: wlan1: AP-DISABLED
Tue May 12 17:14:40 2020 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Tue May 12 17:14:40 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Tue May 12 17:14:40 2020 daemon.notice hostapd: nl80211: Failed to remove interface wlan1 from bridge br-lan: Invalid argument
Tue May 12 17:18:06 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Tue May 12 17:18:06 2020 kern.info kernel: [  712.008620] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Tue May 12 17:18:06 2020 kern.info kernel: [  712.018645] br-lan: port 2(wlan1) entered blocking state
Tue May 12 17:18:06 2020 kern.info kernel: [  712.023953] br-lan: port 2(wlan1) entered disabled state
Tue May 12 17:18:06 2020 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
[... 37 lines cut ...]
Tue May 12 17:18:16 2020 daemon.notice hostapd: wlan0: interface state HT_SCAN->ENABLED
Tue May 12 17:18:16 2020 daemon.notice hostapd: wlan0: AP-ENABLED
```
## kernel log
```
[  712.008620] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  712.018645] br-lan: port 2(wlan1) entered blocking state
[  712.023953] br-lan: port 2(wlan1) entered disabled state
[  712.029802] device wlan1 entered promiscuous mode
[  713.068582] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  713.075325] br-lan: port 2(wlan1) entered blocking state
[  713.080705] br-lan: port 2(wlan1) entered forwarding state
[  719.745792] device wlan1 left promiscuous mode
[  719.750454] br-lan: port 2(wlan1) entered disabled state
[  721.025218] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  721.038347] br-lan: port 2(wlan0) entered blocking state
[  721.043712] br-lan: port 2(wlan0) entered disabled state
[  721.049722] device wlan0 entered promiscuous mode
[  721.054703] br-lan: port 2(wlan0) entered blocking state
[  721.060119] br-lan: port 2(wlan0) entered forwarding state
[  721.175658] br-lan: port 2(wlan0) entered disabled state
[  722.461121] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  722.467787] br-lan: port 2(wlan0) entered blocking state
[  722.473088] br-lan: port 2(wlan0) entered forwarding state
```

AmadeusGhost, 2020-05-12:
不是因为那个commit，问题commit我已经在 #4590 revert了

shikiroot, 2020-05-12:
> 不是因为那个commit，问题commit我已经在 #4590 revert了

thanks

closed 2020-05-12 by shikiroot