coolsnowwolf/lede #4596: 哪位吊大的老板知道这是什么鬼情况

ferre-han, 2020-05-12:
mv -f /home/parallels/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.23.1/.go_work/build/bin/linux_arm/main /home/parallels/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.23.1/.go_work/build/bin/linux_arm/v2ray
/home/parallels/openwrt/staging_dir/host/bin/upx --lzma --best /home/parallels/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.23.1/.go_work/build/bin/linux_arm/v2ray
bash: /home/parallels/openwrt/staging_dir/host/bin/upx: No such file or directory
Makefile:308: recipe for target '/home/parallels/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.23.1/.built' failed
make[3]: *** [/home/parallels/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.23.1/.built] Error 127
make[3]: Leaving directory '/home/parallels/openwrt/package/lean/v2ray'
time: package/lean/v2ray/compile#49.93#10.36#124.06
package/Makefile:111: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/parallels/openwrt'
package/Makefile:107: recipe for target '/home/parallels/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/parallels/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/parallels/openwrt'
/home/parallels/openwrt/include/toplevel.mk:218: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2020-05-12:
bash: /home/parallels/openwrt/staging_dir/host/bin/upx: No such file or directory

ferre-han, 2020-05-14:
> bash: /home/parallels/openwrt/staging_dir/host/bin/upx: No such file or directory

谢谢💋 老板 🙏

closed 2020-05-14 by ferre-han