coolsnowwolf/lede #4611: 竟斗云 最新源码自编译的固件，随机几分钟重启

lyin888, 2020-05-15:
从5月的源码开始自编译的固件随机性的几分钟重启，最新源码的内核日志如下

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.180 (runner@fv-az121) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r0-fd1fbe5)) #0 SMP Thu May 14 03:17:01 2020
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: P&W R619AC 128M
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 130560
[    0.000000] free_area_init_node: node 0, pgdat c0a27a00, node_mem_map dfb79000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x3ec with crng_init=0
[    0.000000] percpu: Embedded 15 pages/cpu s29196 r8192 d24052 u61440
[    0.000000] pcpu-alloc: s29196 r8192 d24052 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 rootfstype=squashfs
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 509312K/522240K available (4592K kernel code, 160K rwdata, 1240K rodata, 1024K init, 225K bss, 12928K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0208000 - 0xc077c370   (5585 kB)
[    0.000000]       .init : 0xc0900000 - 0xc0a00000   (1024 kB)
[    0.000000]       .data : 0xc0a00000 - 0xc0a28300   ( 161 kB)
[    0.000000]        .bss : 0xc0a2a000 - 0xc0a62598   ( 226 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000009] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000023] Switching to timer-based delay loop, resolution 20ns
[    0.000244] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000261] pid_max: default: 32768 minimum: 301
[    0.000425] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000441] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001201] CPU: Testing write buffer coherency: ok
[    0.001976] Setting up static identity map for 0x80300000 - 0x80300060
[    0.002131] Hierarchical SRCU implementation.
[    0.002823] smp: Bringing up secondary CPUs ...
[    0.005678] smp: Brought up 1 node, 4 CPUs
[    0.005698] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.005706] CPU: All CPU(s) started in SVC mode.
[    0.010285] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.010442] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.010467] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.010724] pinctrl core: initialized pinctrl subsystem
[    0.011650] NET: Registered protocol family 16
[    0.011986] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.013027] cpuidle: using governor ladder
[    0.013073] cpuidle: using governor menu
[    0.031417] usbcore: registered new interface driver usbfs
[    0.031482] usbcore: registered new interface driver hub
[    0.031568] usbcore: registered new device driver usb
[    0.031635] pps_core: LinuxPPS API ver. 1 registered
[    0.031645] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.031672] PTP clock support registered
[    0.032966] clocksource: Switched to clocksource arch_sys_counter
[    0.033870] NET: Registered protocol family 2
[    0.034472] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
[... 235 lines cut ...]
[   35.353998] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   37.496033] ath10k_ahb a000000.wifi: unsupported HTC service id: 1536
[   37.634306] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   37.655986] br-lan: port 2(wlan0) entered blocking state
[   37.656047] br-lan: port 2(wlan0) entered disabled state
[   37.661009] device wlan0 entered promiscuous mode
[   38.126406] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   38.126937] br-lan: port 2(wlan0) entered blocking state
[   38.131892] br-lan: port 2(wlan0) entered forwarding state
[   39.668167] ath10k_ahb a800000.wifi: unsupported HTC service id: 1536
[   39.795267] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   39.808055] br-lan: port 3(wlan1) entered blocking state
[   39.808140] br-lan: port 3(wlan1) entered disabled state
[   39.813485] device wlan1 entered promiscuous mode
[   40.002390] pppoe-wan: renamed from ppp0
[   40.398740] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   40.399018] br-lan: port 3(wlan1) entered blocking state
[   40.404221] br-lan: port 3(wlan1) entered forwarding state
[   49.913775] config size: 25,data = { "op": 3, "data": { } }
[   49.913775] 
[   49.914528] #########show app status##########
[   49.920926] 
[   49.920926] 
[   49.920926] 
[   51.919040] ksmbd: kill_server_store:480: kill command received

lyin888, 2020-05-15:
系统日志

Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000] Linux version 4.14.180 (runner@fv-az121) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r0-fd1fbe5)) #0 SMP Thu May 14 03:17:01 2020
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] OF: fdt: Machine model: P&W R619AC 128M
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Fri May 15 14:16:11 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 130560
Fri May 15 14:16:11 2020 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat c0a27a00, node_mem_map dfb79000
Fri May 15 14:16:11 2020 kern.debug kernel: [    0.000000]   Normal zone: 1152 pages used for memmap
Fri May 15 14:16:11 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Fri May 15 14:16:11 2020 kern.debug kernel: [    0.000000]   Normal zone: 130560 pages, LIFO batch:31
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x3ec with crng_init=0
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s29196 r8192 d24052 u61440
Fri May 15 14:16:11 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s29196 r8192 d24052 u61440 alloc=15*4096
Fri May 15 14:16:11 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 rootfstype=squashfs
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] Memory: 509312K/522240K available (4592K kernel code, 160K rwdata, 1240K rodata, 1024K init, 225K bss, 12928K reserved, 0K cma-reserved, 0K highmem)
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000] Virtual kernel memory layout:
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000]       .text : 0xc0208000 - 0xc077c370   (5585 kB)
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000]       .init : 0xc0900000 - 0xc0a00000   (1024 kB)
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000]       .data : 0xc0a00000 - 0xc0a28300   ( 161 kB)
Fri May 15 14:16:11 2020 kern.notice kernel: [    0.000000]        .bss : 0xc0a2a000 - 0xc0a62598   ( 226 kB)
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] Hierarchical RCU implementation.
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000009] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000023] Switching to timer-based delay loop, resolution 20ns
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000244] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000261] pid_max: default: 32768 minimum: 301
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000425] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Fri May 15 14:16:11 2020 kern.info kernel: [    0.000441] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Fri May 15 14:16:11 2020 kern.info kernel: [    0.001201] CPU: Testing write buffer coherency: ok
Fri May 15 14:16:11 2020 kern.info kernel: [    0.001976] Setting up static identity map for 0x80300000 - 0x80300060
[... cut ...]
Fri May 15 14:37:08 2020 daemon.info procd: Instance nginx::instance1 s in a crash loop 6 crashes, 3 seconds since last crash
Fri May 15 14:37:09 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Fri May 15 14:37:10 2020 cron.info crond[14384]: crond (busybox 1.30.0) started, log level 5
Fri May 15 14:37:13 2020 cron.info crond[14889]: crond (busybox 1.30.0) started, log level 5
Fri May 15 14:37:14 2020 user.err ddns-scripts[9118]: ipv6: BusyBox nslookup error: '1'
Fri May 15 14:37:14 2020 cron.info crond[15091]: crond (busybox 1.30.0) started, log level 5

lyin888, 2020-05-15:
系统日志

Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000] Linux version 4.14.180 (runner@fv-az121) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r0-fd1fbe5)) #0 SMP Thu May 14 03:17:01 2020
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] OF: fdt: Machine model: P&W R619AC 128M
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Fri May 15 15:04:29 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 130560
Fri May 15 15:04:29 2020 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat c0a27a00, node_mem_map dfb79000
Fri May 15 15:04:29 2020 kern.debug kernel: [    0.000000]   Normal zone: 1152 pages used for memmap
Fri May 15 15:04:29 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Fri May 15 15:04:29 2020 kern.debug kernel: [    0.000000]   Normal zone: 130560 pages, LIFO batch:31
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x3ec with crng_init=0
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s29196 r8192 d24052 u61440
Fri May 15 15:04:29 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s29196 r8192 d24052 u61440 alloc=15*4096
Fri May 15 15:04:29 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 rootfstype=squashfs
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] Memory: 509312K/522240K available (4592K kernel code, 160K rwdata, 1240K rodata, 1024K init, 225K bss, 12928K reserved, 0K cma-reserved, 0K highmem)
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000] Virtual kernel memory layout:
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000]       .text : 0xc0208000 - 0xc077c370   (5585 kB)
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000]       .init : 0xc0900000 - 0xc0a00000   (1024 kB)
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000]       .data : 0xc0a00000 - 0xc0a28300   ( 161 kB)
Fri May 15 15:04:29 2020 kern.notice kernel: [    0.000000]        .bss : 0xc0a2a000 - 0xc0a62598   ( 226 kB)
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] Hierarchical RCU implementation.
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000021] Switching to timer-based delay loop, resolution 20ns
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000243] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000264] pid_max: default: 32768 minimum: 301
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000433] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Fri May 15 15:04:29 2020 kern.info kernel: [    0.000448] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Fri May 15 15:04:29 2020 kern.info kernel: [    0.001211] CPU: Testing write buffer coherency: ok
Fri May 15 15:04:29 2020 kern.info kernel: [    0.001979] Setting up static identity map for 0x80300000 - 0x80300060
[... cut ...]
Fri May 15 15:04:51 2020 daemon.info aria2: Instance "main" disabled.
Fri May 15 15:04:52 2020 daemon.err uhttpd[1604]: cat: can't open '/etc/bench.log': No such file or directory
Fri May 15 15:04:52 2020 daemon.err uhttpd[1604]: cat: can't open '/etc/bench.log': No such file or directory
Fri May 15 15:10:31 2020 daemon.notice netifd: Interface 'wan_6' is now up
Fri May 15 15:10:32 2020 daemon.warn odhcpd[1500]: A default route is present but there is no public prefix on lan thus we don't announce a default route!

AmadeusGhost, 2020-05-15:
oaf这个插件去掉看看

Aes64X, 2020-05-15:
packages退到https://github.com/coolsnowwolf/packages/commit/cd354e8b1abc165617c519e24cd0e55fb3b35018

Aes64X, 2020-05-15:
我也不知道为什么。。。

coolsnowwolf, 2020-05-15:
#########show app status##########

注意下 OAF 和 CIFSD 这两个插件

lyin888, 2020-05-15:
谢谢老大的回答，去掉oaf重新编译一个试试

lyin888, 2020-05-16:
经测试是oaf 导致的重启，感谢老大以及各位的帮助。

closed 2020-05-16 by lyin888