coolsnowwolf/lede #4614: 竞斗云断流问题。

garygary1, 2020-05-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：断流时机器没重启，但wifi断了，路由模式。

### (2) 路由器型号和固件版本

A：竞斗云2.0最新编译的固件。

### (3) 详细日志

A：Sat May 16 18:34:08 2020 daemon.info hostapd: wlan1: STA d4:25:8b:51:5b:f8 WPA: pairwise key handshake completed (RSN)
Sat May 16 18:34:11 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED d4:25:8b:51:5b:f8
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.300844] ath10k_ahb a800000.wifi: Invalid peer id 19 peer stats buffer
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.300968] ------------[ cut here ]------------
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.306950] WARNING: CPU: 3 PID: 0 at target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/backports-5.4.27-1/net/mac80211/sta_info.c:1937 ieee80211_sta_update_pending_airtime+0x104/0x1ec [mac80211]
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.311473] STA d4:25:8b:51:5b:f8 AC 2 txq pending airtime underflow: 4294967236, 60
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.311478] Modules linked in: xt_FULLCONENAT pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_socket xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_esp xt_conntrack xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm slhc nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.386338]  nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conntrack macvlan iptable_raw iptable_mangle iptable_filter ipt_ah ip_tables crc_ccitt compat fuse tcp_bbr cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 xfrm6_mode_tunnel xfrm6_mode_transport
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.457805]  xfrm6_mode_beet ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet ipcomp esp4 ah4 tunnel6 tunnel4 tun xfrm_user xfrm_ipcomp af_key xfrm_algo vfat fat nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic sha1_arm_neon sha1_arm drbg md5 echainiv ecb des_generic cbc authenc uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 exfat crc32c_generic
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.505723] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.180 #0
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.527908] Hardware name: Generic DT based system
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.534006] [<c030e45c>] (unwind_backtrace) from [<c030a96c>] (show_stack+0x10/0x14)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.538773] [<c030a96c>] (show_stack) from [<c0773cb4>] (dump_stack+0x94/0xa8)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.546761] [<c0773cb4>] (dump_stack) from [<c031bd04>] (__warn+0xe4/0x100)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.553874] [<c031bd04>] (__warn) from [<c031bd58>] (warn_slowpath_fmt+0x38/0x48)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.561270] [<c031bd58>] (warn_slowpath_fmt) from [<bf6df938>] (ieee80211_sta_update_pending_airtime+0x104/0x1ec [mac80211])
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.568734] [<bf6df938>] (ieee80211_sta_update_pending_airtime [mac80211]) from [<bf6dbf90>] (ieee80211_tx_monitor+0xd38/0xf5c [mac80211])
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.580021] [<bf6dbf90>] (ieee80211_tx_monitor [mac80211]) from [<bf6dc200>] (ieee80211_tx_status+0x4c/0x54 [mac80211])
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.592387] [<bf6dc200>] (ieee80211_tx_status [mac80211]) from [<bf7c45c0>] (ath10k_txrx_tx_unref+0x258/0x2d4 [ath10k_core])
Sat May 16 18:34:11 2020 kern.err kernel: [167885.600683] mmc0: Reset 0x1 never completed.
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.603502] [<bf7c45c0>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf7bfa94>] (ath10k_htt_txrx_compl_task+0xb2c/0xff4 [ath10k_core])
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.618950] [<bf7bfa94>] (ath10k_htt_txrx_compl_task [ath10k_core]) from [<bf809740>] (ath10k_pci_napi_poll+0x74/0x104 [ath10k_pci])
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.630953] [<bf809740>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c0672918>] (net_rx_action+0x138/0x2fc)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.643029] [<c0672918>] (net_rx_action) from [<c0301520>] (__do_softirq+0xe0/0x240)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.652572] [<c0301520>] (__do_softirq) from [<c0320014>] (irq_exit+0xd4/0x138)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.660556] [<c0320014>] (irq_exit) from [<c035cce4>] (__handle_domain_irq+0x9c/0xac)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.668022] [<c035cce4>] (__handle_domain_irq) from [<c030140c>] (gic_handle_irq+0x5c/0x90)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.675747] [<c030140c>] (gic_handle_irq) from [<c030b4cc>] (__irq_svc+0x6c/0x90)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.684334] Exception stack(0xdf46df80 to 0xdf46dfc8)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.691715] df80: 00000001 00000000 00000000 c0313b00 ffffe000 c0a03c78 c0a03c28 00000000
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.696838] dfa0: 00000000 410fc075 00000000 00000000 df46dfc8 df46dfd0 c0308048 c030804c
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.705083] dfc0: 60000013 ffffffff
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.713328] [<c030b4cc>] (__irq_svc) from [<c030804c>] (arch_cpu_idle+0x34/0x38)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.717070] [<c030804c>] (arch_cpu_idle) from [<c0352b04>] (do_idle+0xdc/0x1a0)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.724446] [<c0352b04>] (do_idle) from [<c0352e24>] (cpu_startup_entry+0x18/0x1c)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.731909] [<c0352e24>] (cpu_startup_entry) from [<8030170c>] (0x8030170c)
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.739327] ---[ end trace 1c95b6fc4a24b33a ]---
Sat May 16 18:34:11 2020 kern.warn kernel: [167885.747731] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon

garygary1, 2020-05-16:
默认设置编译的。

yinzhefuyun, 2020-05-16:
重启试一试，我最近发现如果固件刚刚升级后然后用就会断流，重启后就不会了，原因不明

garygary1, 2020-05-17:
好的，我试试看。不过还算稳定的。

shenzt68, 2020-05-21:
自从驱动改变后，无线中继一直不行！一直重启，都没时间导出日志。

AmadeusGhost, 2020-05-25:
```
wget https://github.com/AmadeusGhost/lede/commit/c1a7c226099fb7a244cd634752317b8272fb2182.patch
### rename to 1.patch
git apply 1.patch
```
要不试下mac80211升到5.4.36会不会好一点

Andylive5518, 2020-05-30:
我的直接假死，后来查了一下日志应该是netdata这个插件对闪存溢出引起的。

closed 2020-07-04 by coolsnowwolf