coolsnowwolf/lede #4617: 新版可道云适配 kodbox.1.09.zip

suojiang, 2020-05-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
可道云发布了新版的安装包，安装提示sqlite不可用
wget http://static.kodcloud.com/update/download/kodbox.1.09.zip
unzip kodbox.1.09.zip && chmod -Rf 777 ./*
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
可道云发布了新版的安装包，安装提示sqlite不可用
### (1) 具体问题
[image]

A：

### (2) 路由器型号和固件版本
我家云 openwrt 
Powered by LuCI Master (git-20.117.60969-420c61a) / OpenWrt R20.5.9
A：

### (3) 详细日志

A：

EnnawYang, 2020-05-17:
那就等luci更新咯 或者使用旧版

hxlls, 2020-05-17:
可能需要SQLITE ，可以编译进去试试，如果还不行还是等大雕出手把，新版的好像加入了新的机制，看起来还是不错的

EnnawYang, 2020-05-17:
> 可能需要SQLITE ，可以编译进去试试，如果还不行还是等大雕出手把，新版的好像加入了新的机制，看起来还是不错的

SQLITE我另外装了没用 应该是要改配置文件，我还是等等吧（ ，新版看起来是不错

suojiang, 2020-07-03:
OpenWrt R20.7.1 实测已经可以使用新版可道云

EnnawYang, 2020-07-03:
早就可以了 不过部分功能更新Lienol大没有推过来

suojiang, 2020-07-03:
> 早就可以了 不过部分功能更新Lienol大没有推过
还有什么特别功能吗？

EnnawYang, 2020-07-03:
> > 早就可以了 不过部分功能更新Lienol大没有推过
> > 还有什么特别功能吗？

修复了wabdav，增加了一些设置项

suojiang, 2020-07-03:
> > > 早就可以了 不过部分功能更新Lienol大没有推过
> > > 还有什么特别功能吗？
> 
> 修复了webdav，增加了一些设置项

嗯 正在看官方解决方案 http://doc.kodcloud.com/v2/#/help/pathInfo

closed 2020-07-04 by coolsnowwolf