coolsnowwolf/lede #4620: 修改5GHz无线参数后，2.4GHz无线会无法启用

zhouym233, 2020-05-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

调整5GHz的发射功率后，2.4GHz无线会掉线。重启2.4GHz无线的话似乎两个无线都会掉线。断电重启后无线会恢复。
[image]

### (2) 路由器型号和固件版本

A：友华WR1200JS，OpenWrt R20.5.9 / LuCI Master (git-20.117.60969-420c61a)

### (3) 详细日志

A：
#### 系统日志：
```
Mon May 18 15:31:37 2020 daemon.info hostapd: wlan1: STA 30:f7:72:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon May 18 15:31:43 2020 daemon.notice hostapd: Remove interface 'wlan0'
Mon May 18 15:31:43 2020 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Mon May 18 15:31:43 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 98:6c:f5:xx:xx:xx
Mon May 18 15:31:43 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 74:23:44:xx:xx:xx
Mon May 18 15:31:43 2020 daemon.notice hostapd: wlan0: AP-DISABLED
Mon May 18 15:31:43 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Mon May 18 15:31:43 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mon May 18 15:31:43 2020 kern.info kernel: [37876.276180] device wlan0 left promiscuous mode
Mon May 18 15:31:43 2020 kern.info kernel: [37876.280959] br-lan: port 3(wlan0) entered disabled state
Mon May 18 15:31:43 2020 daemon.notice netifd: Network device 'wlan0' link is down
Mon May 18 15:31:44 2020 daemon.notice hostapd: Remove interface 'wlan1'
Mon May 18 15:31:44 2020 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Mon May 18 15:31:44 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 4c:49:e3:xx:xx:xx
Mon May 18 15:31:44 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED f4:96:34:xx:xx:xx
Mon May 18 15:31:44 2020 daemon.notice hostapd: wlan1: AP-DISABLED
Mon May 18 15:31:44 2020 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Mon May 18 15:31:44 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Mon May 18 15:31:44 2020 kern.info kernel: [37876.846303] device wlan1 left promiscuous mode
Mon May 18 15:31:44 2020 kern.info kernel: [37876.851498] br-lan: port 2(wlan1) entered disabled state
Mon May 18 15:31:44 2020 daemon.notice netifd: Network device 'wlan1' link is down
Mon May 18 15:31:44 2020 user.err wsdd2[3008]: error: wsdd-mcast-v4: wsd_send_soap_msg: send
Mon May 18 15:31:45 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Mon May 18 15:31:46 2020 kern.info kernel: [37878.703992] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Mon May 18 15:31:46 2020 kern.info kernel: [37878.716433] br-lan: port 2(wlan1) entered blocking state
Mon May 18 15:31:46 2020 kern.info kernel: [37878.721921] br-lan: port 2(wlan1) entered disabled state
Mon May 18 15:31:46 2020 kern.info kernel: [37878.727797] device wlan1 entered promiscuous mode
Mon May 18 15:31:46 2020 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon May 18 15:31:46 2020 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
Mon May 18 15:31:46 2020 daemon.notice netifd: radio1 (9222): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Mon May 18 15:31:46 2020 daemon.notice netifd: radio1 (9222): Command failed: Invalid argument
Mon May 18 15:31:46 2020 daemon.err hostapd: Using interface wlan1 with hwaddr d4:5f:25:xx:xx:xx and ssid "HiTV_d4xxxx"
Mon May 18 15:31:47 2020 kern.info kernel: [37879.753539] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Mon May 18 15:31:47 2020 kern.info kernel: [37879.760506] br-lan: port 2(wlan1) entered blocking state
Mon May 18 15:31:47 2020 kern.info kernel: [37879.765938] br-lan: port 2(wlan1) entered forwarding state
Mon May 18 15:31:47 2020 daemon.notice hostapd: wlan1: interface state HT_SCAN->ENABLED
Mon May 18 15:31:47 2020 daemon.notice hostapd: wlan1: AP-ENABLED
Mon May 18 15:31:47 2020 daemon.notice netifd: Network device 'wlan1' link is up
Mon May 18 15:31:48 2020 daemon.info hostapd: wlan1: STA f4:96:34:ea:xx:xx IEEE 802.11: authenticated
Mon May 18 15:31:48 2020 daemon.info hostapd: wlan1: STA f4:96:34:ea:xx:xx IEEE 802.11: associated (aid 1)
Mon May 18 15:31:48 2020 cron.info crond[9632]: crond (busybox 1.30.0) started, log level 5
```
#### 内核日志：
```
[   30.673902] br-lan: port 3(wlan0) entered forwarding state
[   30.743519] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[37876.276180] device wlan0 left promiscuous mode
[37876.280959] br-lan: port 3(wlan0) entered disabled state
[37876.846303] device wlan1 left promiscuous mode
[37876.851498] br-lan: port 2(wlan1) entered disabled state
[37878.703992] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[37878.716433] br-lan: port 2(wlan1) entered blocking state
[37878.721921] br-lan: port 2(wlan1) entered disabled state
[37878.727797] device wlan1 entered promiscuous mode
[37879.753539] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[37879.760506] br-lan: port 2(wlan1) entered blocking state
[37879.765938] br-lan: port 2(wlan1) entered forwarding state
```

hcym, 2020-05-18:
ssh改，luci容易引起不适

AmadeusGhost, 2020-05-18:
是最新源码编译的嘛？

zhouym233, 2020-05-18:
> 是最新源码编译的嘛？

不是，
是在5月9日的`mt76: bump to lastest version`这条提交之后编译的。

AmadeusGhost, 2020-05-18:
> 是在5月9日的`mt76: bump to lastest version`这条提交之后编译的。

请用最新代码编译，问题 commit 我已在 #4590 中回退

zhouym233, 2020-05-18:
> 请用最新代码编译，问题 commit 我已在 #4590 中回退

好的，非常感谢解答。

closed 2020-05-18 by zhouym233