coolsnowwolf/lede #4622: 请问软路由怎么安装flexget，配合transmission自动下载种子

m2662m, 2020-05-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
1037u软路由一台，装了一块2.5的笔记本硬盘，用tr下载种子，适当下载电影观看。由于tr没有rss功能，所以找到flexget，官网显示支持linux并且也支持openwrt，但是通过步骤都不能成功安装
这是官网的openwrt地址：https://flexget.com/InstallWizard/OpenWrt
貌似python版本不对应，还有几个依赖也不能成功安装

### (1) 具体问题

A：怎么才能成功在openwrt上安装成功flexget？

### (2) 路由器型号和固件版本

A：1037u物理安装 openwrt 4.19.115

### (3) 详细日志

A：root@OpenWrt:~# opkg install python python-sqlite3 pyyaml
Package python (2.7.17-2) installed in root is up to date.
Package python-sqlite3 (2.7.17-2) installed in root is up to date.
Unknown package 'pyyaml'.
Collected errors:
 * opkg_install_cmd: Cannot install package pyyaml.
root@OpenWrt:~# opkg install python-sqlite
Unknown package 'python-sqlite'.
Collected errors:
 * opkg_install_cmd: Cannot install package python-sqlite.
root@OpenWrt:~# opkg install python-expat
Unknown package 'python-expat'.
Collected errors:
 * opkg_install_cmd: Cannot install package python-expat.
root@OpenWrt:~# opkg install python-openssl
Package python-openssl (2.7.17-2) installed in root is up to date.
root@OpenWrt:~# opkg install python-bzip2
Unknown package 'python-bzip2'.
Collected errors:
 * opkg_install_cmd: Cannot install package python-bzip2.
root@OpenWrt:~# opkg install distribute
Unknown package 'distribute'.
Collected errors:
 * opkg_install_cmd: Cannot install package distribute.
root@OpenWrt:~# easy_install pip
-ash: easy_install: not found
root@OpenWrt:~# pip install flexget
-ash: pip: not found
root@OpenWrt:~#

leungyauming, 2020-05-19:
PyYAML, FlexGet等等是pip的東西，不是OpenWrt opkg所管理的。

mjyhj, 2020-05-19:
直接docker一个

closed 2020-05-27 by m2662m