coolsnowwolf/lede #4625: ssr plus 拨号掉线 不能工作

sk007, 2020-05-18:
软路由 PPPOE拨号
如果中途 掉线重拨（1次或多次后） 会导致ssr plus 一直不能连接上
需要手动 重启 下ssr plus 才可以使用

希望添加个 
自动检查状态 和 掉线重启插件选项

cattypapa, 2020-05-19:
n1也有这个问题...重新拨号以后，需要手动保存一下ssr界面，ssr才能再次运行。

sk007, 2020-05-19:
是的 只要多次重新拨号 或者wan口掉线重新连间隔比较久 就需要重新保存ssr 界面 才可以使用

cattypapa, 2020-05-19:
希望大雕能加哥自动重启...

dolphinpaopao, 2020-05-19:
此项目已经删除ssr+有问题也不会在这里修复了。我用4月17号编译的版本，宽带48h自动断线重播时没有出现这个问题。另外请检查以下高级设置，有类似功能。

a1392429292, 2020-05-19:
我正准备发 issue，看到你的就在这里描述一下吧，我的情况是，没有pppoe重新拨号，也没有wan口掉线，就是好好的，ssr plus就出不去了，需要到ssr plus里先停用，然后再启用刚刚选择的线路，就又可以用了，可以确定的是绝对不是线路的问题，仅仅是ssr plus重启一下，就好了。。。

a1392429292, 2020-05-19:
> 此项目已经删除ssr+有问题也不会在这里修复了。我用4月17号编译的版本，宽带48h自动断线重播时没有出现这个问题。另外请检查以下高级设置，有类似功能。

那现在在哪里可以获得技术支持，请给条明路

himho, 2020-05-19:
> 此项目已经删除ssr+有问题也不会在这里修复了。我用4月17号编译的版本，宽带48h自动断线重播时没有出现这个问题。另外请检查以下高级设置，有类似功能。

只是隐藏：删掉注释还是一样可以编译！就是不知道以后还会不会维护

dolphinpaopao, 2020-05-19:
> > 此项目已经删除ssr+有问题也不会在这里修复了。我用4月17号编译的版本，宽带48h自动断线重播时没有出现这个问题。另外请检查以下高级设置，有类似功能。
> 
> 那现在在哪里可以获得技术支持，请给条明路

https://github.com/fw876/helloworld
现在是独立的

a1392429292, 2020-05-20:
> > > 此项目已经删除ssr+有问题也不会在这里修复了。我用4月17号编译的版本，宽带48h自动断线重播时没有出现这个问题。另外请检查以下高级设置，有类似功能。
> > 
> > 
> > 那现在在哪里可以获得技术支持，请给条明路
> 
> https://github.com/fw876/helloworld
> 现在是独立的

怎么食用这个独立的ssr plus啊？放到哪？或者说有没有一个加入到lede的教程呢？谢谢了。。。

coolsnowwolf, 2020-05-20:
这个有可能是docker 独有的问题

sk007, 2020-05-20:
> 这个有可能是docker独有的问题

希望可以有空 改进下

a1392429292, 2020-05-21:
可我并没有使用docker啊，整个lede里，就加入了 ddns，firewall，flowoffload，ssr-plus，没其他的了，同样出现ssr plus，好好的就无法出去，只需要停用在启动，就可以出去了;
看日志也没有异常；全部是
2020-05-21 10:24:10 ShadowsocksR No Problem.
2020-05-21 10:13:02 ShadowsocksR No Problem.
2020-05-21 10:01:54 ShadowsocksR No Problem.
2020-05-21 09:50:46 ShadowsocksR No Problem.
但就是出不去，需要restart一下才行，
而不行到行，线路都没换过。。。。。

closed 2020-07-04 by coolsnowwolf