coolsnowwolf/lede #4632: 编译qbittorrent报错，有日志大神请进！

chenxudong2020, 2020-05-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错！
github Action全自动编译  不知道是什么原因 具体配置见项目
https://github.com/chenxudong2020/Openwrt-HC5962_Lean_firmware

### (2) 路由器型号和固件版本

A：HC5962或者B70  20200519 今天的源码

### (3) 详细日志

A： make[3] -C package/lean/rblibtorrent compile
 make[3] -C package/system/opkg compile
 make[3] -C feeds/luci/collections/luci compile
 make[3] -C feeds/packages/net/ariang compile
 make[3] -C feeds/luci/applications/luci-app-ddns compile
 make[3] -C package/kernel/mt76 compile
 make[3] -C package/lean/luci-app-autoreboot compile
 make[3] -C package/lean/luci-lib-fs compile
 make[3] -C package/lean/qBittorrent compile
 make[3] -C feeds/luci/applications/luci-app-aria2 compile
 make[3] -C package/lean/luci-app-filetransfer compile
 make[3] -C package/lean/luci-app-qbittorrent compile
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/runner/work/Openwrt-HC5962_Lean_firmware/Openwrt-HC5962_Lean_firmware/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..684c112b40f8704128fe27fba1102c7eaf00190b
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..684c112b40f8704128fe27fba1102c7eaf00190b
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..684c112b40f8704128fe27fba1102c7eaf00190b
awk: cmd. line:14: (FILENAME=- FNR=623) fatal: print to "standard output" failed (No space left on device)
/home/runner/work/Openwrt-HC5962_Lean_firmware/Openwrt-HC5962_Lean_firmware/openwrt/tmp/info/.files-packageinfo.mk:623: *** unterminated call to function 'eval': missing ')'.  Stop.
make[1]: *** [prepare-tmpinfo] Error 2
/home/runner/work/Openwrt-HC5962_Lean_firmware/Openwrt-HC5962_Lean_firmware/openwrt/include/toplevel.mk:83: recipe for target 'prepare-tmpinfo' failed
make: *** [world] Error 2
/home/runner/work/Openwrt-HC5962_Lean_firmware/Openwrt-HC5962_Lean_firmware/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
##[error]Process completed with exit code 2.

zheshifandian, 2020-05-19:
re-run make -j1 V=s 
然后把报错发上来

chenxudong2020, 2020-05-19:
上回那个wifi计划那个bug是不是你修复的，我发现编译hc5962以后也有那个问题，你pr给lean吧，现在这个日志明天才有。

chenxudong2020, 2020-05-19:
详细日志我刚看了下 开启单线程编译之后  日志未打出 github提示空间不足 
详细请看https://github.com/chenxudong2020/Openwrt-HC5962_Lean_firmware/actions/runs/109434742

coolsnowwolf, 2020-05-20:
No space left on device 没有空间了

zheshifandian, 2020-05-20:
gitaction给的空间用完了
用实体机编译啊

chenxudong2020, 2020-05-20:
实体机没有科学账号 怎么办啊

zheshifandian, 2020-05-20:
用gitpod（有轻度的gui界面）make download下载依赖，然后把dl文件夹下载下来就好了

chenxudong2020, 2020-05-20:
我都中转到gitee了

closed 2020-07-04 by coolsnowwolf