coolsnowwolf/lede #4634: 如何在github actions中添加编译多拨与负载均衡的配置,使编译的固件中有这两个功能

xiaoweb, 2020-05-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
如何在github actions中添加编译多拨与负载均衡的配置,使编译的固件中有这两个功能

### (1) 具体问题
如何在github actions中添加编译多拨与负载均衡的配置,使编译的固件中有这两个功能
A：

### (2) 路由器型号和固件版本
j1900 最新版固件
A：

### (3) 详细日志

A：

xiaoweb, 2020-05-20:
没有人关注吗

9l, 2020-05-24:
这不是默认功能么

chenyangcyx, 2020-05-24:
make menuconfig里面，LUCI-Applications里面，mwan3是负载均衡，syncdial是多拨。
详细可以看这篇文章：https://www.right.com.cn/forum/thread-344825-1-1.html

xiaoweb, 2020-05-27:
> make menuconfig里面，LuCI ---> Applications，mwan3是负载均衡，syncdial是多拨，选了syncdial会自动选上mwan3
> 
> 详细可以看这篇文章：https://www.right.com.cn/forum/thread-3682029-1-1.html
> LUCI里面的各个软件都有详细的说明

嗯 编译我知道, 我是说在github actions的构建文件里如何配置

chenyangcyx, 2020-05-27:
哦  这个意思啊。那你可以在make menuconfig之后停下，把当前目录生成的.config文件下载下来，之后用这个.config文件内容编译就行了
我自己的repository中写了一个小娱XY-C5的自动编译，你可以看看：
https://github.com/chenyangcyx/lede/blob/master/.github/workflows/cyX-OpenWRT%20compile.yml

xiaoweb, 2020-05-28:
> 哦  这个意思啊。那你可以在make menuconfig之后停下，把当前目录生成的.config文件下载下来，之后用这个.config文件内容编译就行了
> 
> 我自己的repository中写了一个小娱XY-C5的自动编译，你可以看看：
> 
> https://github.com/chenyangcyx/lede/blob/master/.github/workflows/cyX-OpenWRT%20compile.yml

嗯，非常感谢我试试

closed 2020-06-04 by xiaoweb