coolsnowwolf/lede #4635: netdata, ttyd 好像不支持https的访问

doumeki, 2020-05-19:
试过uhttpd 和 nginx ，都不能访问https下的netdata, ttyd,页面直空白，手动贴入http的地址可以访问。
同时发现有些软件比如可道云也不行（但网上搜索似乎可以）。
syncthing 和verysync又可以，地址除了端口不同没有其它区别。

不知道是ａｐｐ限制还是web服务器的问题? 是否能path?

coolsnowwolf, 2020-05-20:
Web 浏览器的限制，iframe 的权限问题

garypang13, 2020-05-20:
可以的,我用的nginx

[image]
[image]

doumeki, 2020-05-20:
> 可以的,我用的nginx
> 
> [image]
> [image]

有什么特别设置吗。我单独贴https的netdata,ttyd连接进去也不行。
你的内部访问真实的netdata是用的http还是https?
估计还是nginx 配置问题
下面是我的配置，麻烦看下呢

user  root;
worker_processes  1;

#error_log  logs/error.log;
#error_log  logs/error.log  notice;
#error_log  logs/error.log  info;

pid        /var/run/nginx.pid;

events {
    worker_connections  1024;
}

http {
    include       mime.types;
    default_type  application/octet-stream;

    sendfile on;
    keepalive_timeout 0;

    client_body_buffer_size 10K;
    client_header_buffer_size 1k;
    client_max_body_size 1G;
    large_client_header_buffers 2 1k;

    gzip on;
    gzip_http_version 1.1;
    gzip_vary on;
    gzip_comp_level 1;
    gzip_proxied any;
    gzip_types text/plain text/css application/json application/x-javascript text/xml application/xml application/xml+rss text/javascript application/javascript image/svg+xml;

    root /www;

    server {
        listen 80 default_server;
        listen [::]:80 default_server;
        server_name _;
        return 301 https://$host$request_uri;
    }

    server {
        listen 443 ssl default_server;
        listen [::]:443 ssl default_server;
        server_name  localhost;

        ssl_protocols TLSv1 TLSv1.1 TLSv1.2;
        ssl_prefer_server_ciphers on;
        ssl_ciphers "EECDH+ECDSA+AESGCM:EECDH+aRSA+AESGCM:EECDH+ECDSA+SHA384:EECDH+ECDSA+SHA256:EECDH+aRSA+SHA384:EECDH+aRSA+SHA256:EECDH:DHE+AESGCM:DHE:!RSA!aNULL:!eNULL:!LOW:!RC4:!3DES:!MD5:!EXP:!PSK:!SRP:!DSS:!CAMELLIA:!SEED";
        ssl_session_tickets off;

        ssl_certificate /etc/nginx/nginx.cer;
        ssl_certificate_key /etc/nginx/nginx.key;

        location ~* .(jpg|jpeg|png|gif|ico|css|js)$ {
            expires 365d;
        }

        include luci_uwsgi.conf;

    }

    include /etc/nginx/conf.d/*.conf;
}

garypang13, 2020-05-20:
真实地址都是https
netdata编译的时候要把 --disable-https 去掉
然后在netdata配置文件中开启ssl
另外要把luci中netdata的真实地址改为https
ttyd在后台勾选启用 SSL,并设置好SSL证书路径即可.

doumeki, 2020-05-20:
> 真实地址都是https
> netdata编译的时候要把 --disable-https 去掉
> 然后在netdata配置文件中开启ssl
> 另外要把luci中netdata的真实地址改为https
> ttyd在后台勾选启用 SSL,并设置好SSL证书路径即可.

去掉disable-https后好像编译说找不到netdata, 不知道是不是patch的问题．
另外ttyd好像并没有后台界面能选，强行在命令行加入-S参数结果报错．ttyd也要单独设置证书位置吗？

garypang13, 2020-05-20:
netdata参考这个补丁 https://github.com/garypang13/Actions-OpenWrt-Nginx/blob/master/patches/002-netdata-with-dbengine.patch

ttyd我用的是官方19.07的luci,有直接设置SSL的地方.

closed 2020-07-04 by coolsnowwolf

iKira, 2022-10-06:
> netdata参考这个补丁 https://github.com/garypang13/Actions-OpenWrt-Nginx/blob/master/patches/002-netdata-with-dbengine.patch
> 
> ttyd我用的是官方19.07的luci,有直接设置SSL的地方.

请问官方ttyd你是从这里 (https://github.com/tsl0922/ttyd) git clone 到 /home/ubuntu/lede/package/lean 目录进行编译的吗？

sjkhsl, 2022-11-25:
你解决了吗？我也是同样的问题！