coolsnowwolf/lede #4639: 树莓派4 32位添加curl支持

qq635938204, 2020-05-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
openwrt做为旁路由使用，在公司网络环境下需要认证，因此需要使用curl发送post请求，如何添加curl支持？
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
硬件设备使用的是树莓派4，自己编译的最新版本固件，添加curl支持应该怎么配置？
### (1) 具体问题
ssh登陆后运行curl命令提示curl not found
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

AmadeusGhost, 2020-05-20:
编译的时候选上curl这个包啊🙄

qq635938204, 2020-05-22:
> 编译的时候选上curl这个包啊🙄

menuconfig已经选了curl支持，opkg手动安装libcurl也提示已经安装

9l, 2020-05-24:
```
# File Transfer
CONFIG_PACKAGE_curl=y

#
# Select Boost libraries
#
#
# Libraries
CONFIG_PACKAGE_libcurl=y
```

看看配置文件里，这两项都是y么

qq635938204, 2020-05-25:
> ```
> # File Transfer
> CONFIG_PACKAGE_curl=y
> 
> #
> # Select Boost libraries
> #
> #
> # Libraries
> CONFIG_PACKAGE_libcurl=y
> ```
> 
> 看看配置文件里，这两项都是y么
> 
> PS: 树莓派4不是64位的CPU么，为啥要编译的32位的？

在menuconfig界面把curl下所有的都选上重新编译可以了，64位版本的我编出来ssr plus无法正常运行

closed 2020-05-25 by qq635938204