coolsnowwolf/lede #4640: 对GCC10有兼容性问题

lwzhenglittle, 2020-05-20:
1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用带有gcc10的Linux发行版进行编译，无法找到gcc与g++

在```./scripts/feeds update -a``` 时，
```
Checking 'gcc'... failed.
Checking 'working-gcc'... ok.
Checking 'g++'... failed.
Checking 'working-g++'... ok.
Build dependency: Please install the GNU C Compiler (gcc) 4.8 or later
Build dependency: Please install the GNU C++ Compiler (g++) 4.8 or later
Prerequisite check failed. Use FORCE=1 to override.
```

检查本机gcc（使用gcc -v）:

```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++,d --with-isl --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-install-libiberty --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-libunwind-exceptions --disable-werror gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.1.0 (GCC)
```

g++:

```
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++,d --with-isl --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-install-libiberty --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-libunwind-exceptions --disable-werror gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.1.0 (GCC)
```

无法进行下一步编译

lwzhenglittle, 2020-05-20:
继续执行```./scripts/feeds install -a```

出现
```
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... failed.
Checking 'working-gcc'... ok.
Checking 'g++'... ^Cmake[1]: *** [/home/lwzheng/lede/include/toplevel.mk:177: staging_dir/host/.prereq-build] Interrupt
make: *** [/home/lwzheng/lede/include/toplevel.mk:83: prepare-tmpinfo] Interrupt
Cannot open './tmp/.packageinfo': No such file or directory
Can't open file './tmp/.targetinfo': No such file or directory
Installing all packages from feed packages.
Installing package 'acl' from packages
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'libssp' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'libssp' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
此后重复以上WARNING
```

coolsnowwolf, 2020-05-22:
自行安装 gcc8

xinxijishuwyq, 2020-05-23:
> 1.关于你要提交的问题
>
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [ x ] 没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：使用带有gcc10的Linux发行版进行编译，无法找到gcc与g++
>
> 在`./scripts/feeds update -a` 时，
>
> ```
> Checking 'gcc'... failed.
> Checking [...]

这个问题我之前遇到了，替换include/prereq-build.mk为官方最新版本即可

Rapiz1, 2020-05-27:
如果使用 `squashfs`, `squashfskit4` 同样需要替换为上游最新版本
都是 gcc10 的兼容性问题。希望尽早和上游同步。

closed 2020-06-02 by lwzhenglittle