coolsnowwolf/lede #4643: ssh 公钥失效，仅能在 luci 填写有用。

DbArl, 2020-05-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：20.5.20 版本，已使用 ssh-copy-id 上传公钥，但是仍然需要验证密码。同时尝试过手机上的 ssh 客户端，也是一样的问题。

### (2) 路由器型号和固件版本

A：k3 20.5.20

### (3) 详细日志

A：以下是 ssh -v 的 debug。
Last login: Thu May 21 01:36:12 on ttys000
(base) Al@MacPro ~ % ssh -v root@10.0.0.1
OpenSSH_8.1p1, LibreSSL 2.7.3
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 47: Applying options for *
debug1: Connecting to 10.0.0.1 [10.0.0.1] port 22.
debug1: Connection established.
debug1: identity file /Users/Al/.ssh/id_rsa type 0
debug1: identity file /Users/Al/.ssh/id_rsa-cert type -1
debug1: identity file /Users/Al/.ssh/id_dsa type -1
debug1: identity file /Users/Al/.ssh/id_dsa-cert type -1
debug1: identity file /Users/Al/.ssh/id_ecdsa type -1
debug1: identity file /Users/Al/.ssh/id_ecdsa-cert type -1
debug1: identity file /Users/Al/.ssh/id_ed25519 type -1
debug1: identity file /Users/Al/.ssh/id_ed25519-cert type -1
debug1: identity file /Users/Al/.ssh/id_xmss type -1
debug1: identity file /Users/Al/.ssh/id_xmss-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_8.1
debug1: Remote protocol version 2.0, remote software version dropbear
debug1: no match: dropbear
debug1: Authenticating to 10.0.0.1:22 as 'root'
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: ssh-rsa
debug1: kex: server->client cipher: aes128-ctr MAC: hmac-sha2-256 compression: none
debug1: kex: client->server cipher: aes128-ctr MAC: hmac-sha2-256 compression: none
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: Server host key: ssh-rsa SHA256:OcMmtTqlBnCIyYT94D08/LzdxzdEnlBTnXrtsCU8TME
debug1: Host '10.0.0.1' is known and matches the RSA host key.
debug1: Found key in /Users/Al/.ssh/known_hosts:1
debug1: rekey out after 4294967296 blocks
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: rekey in after 4294967296 blocks
debug1: Will attempt key: /Users/Al/.ssh/id_rsa RSA SHA256:Ds4GSeDFMoGyuJwq5WSb4EGI0Gu7AqHD9h5gl4D53xA
debug1: Will attempt key: /Users/Al/.ssh/id_dsa 
debug1: Will attempt key: /Users/Al/.ssh/id_ecdsa 
debug1: Will attempt key: /Users/Al/.ssh/id_ed25519 
debug1: Will attempt key: /Users/Al/.ssh/id_xmss 
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey,password
debug1: Next authentication method: publickey
debug1: Offering public key: /Users/Al/.ssh/id_rsa RSA SHA256:Ds4GSeDFMoGyuJwq5WSb4EGI0Gu7AqHD9h5gl4D53xA
debug1: Authentications that can continue: publickey,password
debug1: Trying private key: /Users/Al/.ssh/id_dsa
debug1: Trying private key: /Users/Al/.ssh/id_ecdsa
debug1: Trying private key: /Users/Al/.ssh/id_ed25519
debug1: Trying private key: /Users/Al/.ssh/id_xmss
debug1: Next authentication method: password
root@10.0.0.1's password:

coolsnowwolf, 2020-05-22:
待测试

9l, 2020-05-24:
你删掉 rsa 密钥，生成一个 ed25519 密钥试试

9l, 2020-05-24:
翻了一圈系统里的配置文件找到了为什么会这样，openwrt 用的是 dropbear 不是 openssh，ssh-copy-id 上传的公钥是在用户目录下的 ~/.ssh/authorized_keys 文件里，但是这里的 dropbear 读取的是 /etc/dropbear/authorized_keys 这个文件，luci 界面写入的也是这个文件，所以不能用 ssh-copy-id 上传公钥，只能从 luci 写入或者手动更改 /etc/dropbear/authorized_keys 这个文件

DbArl, 2020-05-24:
> 翻了一圈系统里的配置文件找到了为什么会这样，openwrt 用的是 dropbear 不是 openssh，ssh-copy-id 上传的公钥是在用户目录下的 ~/.ssh/authorized_keys 文件里，但是这里的 dropbear 读取的是 /etc/dropbear/authorized_keys 这个文件，luci 界面写入的也是这个文件，所以不能用 ssh-copy-id 上传公钥，只能从 luci 写入或者手动更改 /etc/dropbear/authorized_keys 这个文件

好的。我尝试了传到 dropbear 就没问题了。之前应该是很久以前的配置遗留导致的 ss-copy-id 有效。

closed 2020-05-24 by DbArl