coolsnowwolf/lede #4647: R7800 5g信号经常消失

madcat0513, 2020-05-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
5G信号不够稳定
### (1) 具体问题

A：r7800的5g信号经常消失，重启能恢复，但运行一段时间后又会消失

### (2) 路由器型号和固件版本

A：R7800 自己编译的OpenWrt R20.4.8 / LuCI Master (git-20.117.60969-420c61a)

### (3) 详细日志
内核日志
[ 4573.741866] ------------[ cut here ]------------
[ 4573.741987] WARNING: CPU: 0 PID: 26407 at target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/ath10k-ct-regular/ath10k-ct-2020-03-25-3d173a47/ath10k-5.4/txrx.c:134 ath10k_txrx_tx_unref+0x548/0x724 [ath10k_core]
[ 4573.745573] Invalid VHT rate, nss: 3  hw_rate: 15 ratecode: 255
[ 4573.764483] Modules linked in: xt_FULLCONENAT pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ts_fsm ts_bm slhc nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table
[ 4573.818997]  nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda macvlan iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_cake nf_conntrack tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip
[ 4573.889741]  ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb nls_utf8 nls_cp936 sha1_generic ecb authenc leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple ohci_platform ohci_hcd phy_qcom_dwc3 ahci ehci_platform sd_mod ahci_platform libahci_platform libahci libata scsi_mod ehci_hcd gpio_button_hotplug
[ 4573.923824] CPU: 0 PID: 26407 Comm: kworker/0:1 Not tainted 4.14.176 #0
[ 4573.945984] Hardware name: Generic DT based system
[ 4573.952593] Workqueue: events swconfig_led_work_func
[ 4573.957467] [<c030f2b8>] (unwind_backtrace) from [<c030b494>] (show_stack+0x14/0x20)
[ 4573.962581] [<c030b494>] (show_stack) from [<c07cd0c4>] (dump_stack+0x94/0xa8)
[ 4573.970306] [<c07cd0c4>] (dump_stack) from [<c0319078>] (__warn+0xf0/0x11c)
[ 4573.977328] [<c0319078>] (__warn) from [<c03190d8>] (warn_slowpath_fmt+0x34/0x4c)
[ 4573.984224] [<c03190d8>] (warn_slowpath_fmt) from [<bf6f2b50>] (ath10k_txrx_tx_unref+0x548/0x724 [ath10k_core])
[ 4573.991968] [<bf6f2b50>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf6edd5c>] (ath10k_htt_t2h_msg_handler+0x72c/0x11b4 [ath10k_core])
[ 4574.001771] [<bf6edd5c>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf744e1c>] (ath10k_pci_htt_rx_cb+0x1dc/0x23c [ath10k_pci])
[ 4574.013948] [<bf744e1c>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf70d08c>] (ath10k_ce_per_engine_service+0xa0/0x114 [ath10k_core])
[ 4574.026099] [<bf70d08c>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf70d178>] (ath10k_ce_per_engine_service_any+0x78/0xb4 [ath10k_core])
[ 4574.038051] [<bf70d178>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf745114>] (ath10k_pci_napi_poll+0xac/0x160 [ath10k_pci])
[ 4574.051127] [<bf745114>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c06c66e4>] (net_rx_action+0x144/0x31c)
[ 4574.063532] [<c06c66e4>] (net_rx_action) from [<c03015c8>] (__do_softirq+0xf0/0x264)
[ 4574.073171] [<c03015c8>] (__do_softirq) from [<c031d834>] (irq_exit+0xdc/0x148)
[ 4574.081068] [<c031d834>] (irq_exit) from [<c03602a8>] (__handle_domain_irq+0xa8/0xc8)
[ 4574.088094] [<c03602a8>] (__handle_domain_irq) from [<c0301488>] (gic_handle_irq+0x6c/0xb8)
[ 4574.096078] [<c0301488>] (gic_handle_irq) from [<c030c08c>] (__irq_svc+0x6c/0x90)
[ 4574.104235] Exception stack(0xd745be00 to 0xd745be48)
[ 4574.111891] be00: dd5106d4 40000013 00000002 0000e824 c08fe67c dd7687c0 c08177ac 000346dc
[ 4574.116931] be20: 00000002 00000020 dd7a0024 00000000 00000000 d745be50 c062bb78 c07e6f88
[ 4574.125075] be40: 20000013 ffffffff
[ 4574.133232] [<c030c08c>] (__irq_svc) from [<c07e6f88>] (_raw_spin_unlock_irqrestore+0x1c/0x20)
[ 4574.136540] [<c07e6f88>] (_raw_spin_unlock_irqrestore) from [<c062bb78>] (mdiobb_send_bit+0x4c/0x7c)
[ 4574.145216] [<c062bb78>] (mdiobb_send_bit) from [<c062bcac>] (mdiobb_cmd+0x84/0xd4)
[ 4574.154497] [<c062bcac>] (mdiobb_cmd) from [<c062bdf4>] (mdiobb_read+0x5c/0xf4)
[ 4574.161877] [<c062bdf4>] (mdiobb_read) from [<c062879c>] (ar8xxx_mii_read32+0x28/0x5c)
[ 4574.169169] [<c062879c>] (ar8xxx_mii_read32) from [<c0628ad8>] (ar8xxx_read+0x70/0x94)
[ 4574.177153] [<c0628ad8>] (ar8xxx_read) from [<c062aba4>] (ar8327_read_port_status+0x14/0x40)
[ 4574.185054] [<c062aba4>] (ar8327_read_port_status) from [<c0627ce0>] (ar8216_read_port_link+0x3c/0x160)
[ 4574.193648] [<c0627ce0>] (ar8216_read_port_link) from [<c0627e10>] (ar8xxx_sw_get_port_link+0xc/0x1c)
[ 4574.202763] [<c0627e10>] (ar8xxx_sw_get_port_link) from [<c062541c>] (swconfig_led_work_func+0x7c/0x344)
[ 4574.212147] [<c062541c>] (swconfig_led_work_func) from [<c0330ad0>] (process_one_work+0x24c/0x404)
[ 4574.221690] [<c0330ad0>] (process_one_work) from [<c0331b88>] (worker_thread+0x344/0x58c)
[ 4574.230460] [<c0331b88>] (worker_thread) from [<c0336b24>] (kthread+0x15c/0x168)
[ 4574.238705] [<c0336b24>] (kthread) from [<c0307e88>] (ret_from_fork+0x14/0x2c)
[ 4574.246263] ---[ end trace a2fdf856e9dd47e2 ]---

系统日志
Wed May 20 20:49:21 2020 kern.warn kernel: [ 4573.741866] ------------[ cut here ]------------
Wed May 20 20:49:21 2020 kern.warn kernel: [ 4573.741987] WARNING: CPU: 0 PID: 26407 at target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/ath10k-ct-regular/ath10k-ct-2020-03-25-3d173a47/ath10k-5.4/txrx.c:134 ath10k_txrx_tx_unref+0x548/0x724 [ath10k_core]
Wed May 20 20:49:21 2020 kern.warn kernel: [ 4573.745573] Invalid VHT rate, nss: 3  hw_rate: 15 ratecode: 255
Wed May 20 20:49:21 2020 kern.warn kernel: [ 4573.764483] Modules linked in: xt_FULLCONENAT pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ts_fsm ts_bm slhc nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table
Wed May 20 20:49:21 2020 kern.warn kernel: [ 4573.818997]  nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda macvlan iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_cake nf_conntrack tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip
[... 14 lines cut ...]
[... cut ...]
Wed May 20 20:49:22 2020 kern.warn kernel: [ 4574.212147] [<c062541c>] (swconfig_led_work_func) from [<c0330ad0>] (process_one_work+0x24c/0x404)
Wed May 20 20:49:22 2020 kern.warn kernel: [ 4574.221690] [<c0330ad0>] (process_one_work) from [<c0331b88>] (worker_thread+0x344/0x58c)
Wed May 20 20:49:22 2020 kern.warn kernel: [ 4574.230460] [<c0331b88>] (worker_thread) from [<c0336b24>] (kthread+0x15c/0x168)
Wed May 20 20:49:22 2020 kern.warn kernel: [ 4574.238705] [<c0336b24>] (kthread) from [<c0307e88>] (ret_from_fork+0x14/0x2c)
Wed May 20 20:49:22 2020 kern.warn kernel: [ 4574.246263] ---[ end trace a2fdf856e9dd47e2 ]---

AmadeusGhost, 2020-05-21:
`R20.4.8`都一个月前了，用最新源码编译看看

Aes64X, 2020-05-22:
> `R20.4.8`都一个月前了，用最新源码编译看看

换ath10k-ct试试？
我在竞斗云9984上截到大量的：
ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Wed May 20 23:59:58 2020 kern.err kernel: [  254.425381] mmc0: Reset 0x1 never completed.
Thu May 21 00:00:02 2020 kern.warn kernel: [  257.956348] ath10k_warn: 8 callbacks suppressed
Thu May 21 00:00:02 2020 kern.warn kernel: [  257.956363] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:03 2020 kern.err kernel: [  259.225379] mmc0: Reset 0x1 never completed.
Thu May 21 00:00:03 2020 kern.warn kernel: [  259.494690] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:03 2020 kern.warn kernel: [  259.494800] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:03 2020 kern.warn kernel: [  259.500896] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:03 2020 kern.warn kernel: [  259.508238] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:04 2020 kern.warn kernel: [  259.515539] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:04 2020 kern.warn kernel: [  259.522762] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:04 2020 kern.warn kernel: [  259.530107] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:04 2020 kern.warn kernel: [  259.537372] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:04 2020 kern.warn kernel: [  259.544659] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:05 2020 kern.err kernel: [  260.575355] mmc0: Reset 0x1 never completed.
Thu May 21 00:00:07 2020 kern.warn kernel: [  263.028366] ath10k_warn: 12 callbacks suppressed
Thu May 21 00:00:07 2020 kern.warn kernel: [  263.028382] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:07 2020 kern.err kernel: [  263.465846] mmc0: Reset 0x1 never completed.
Thu May 21 00:00:07 2020 kern.warn kernel: [  263.502654] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:07 2020 kern.warn kernel: [  263.502760] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  264.585660] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  264.585798] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  264.591915] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  264.599230] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  264.606510] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  265.440702] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  265.440829] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:11 2020 kern.err kernel: [  266.905424] mmc0: Reset 0x1 never completed.

madcat0513, 2020-05-25:
编译最新的版本，貌似好了

closed 2020-05-25 by madcat0513