coolsnowwolf/lede #4648: libcryptopp编译报错

zheshifandian, 2020-05-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：libcryptopp编译报错

### (2) 路由器型号和固件版本

A：r619ac

### (3) 详细日志

A：CMake Error: The source directory "/home/zheshifandian/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libcryptopp/libcryptopp-5_6_4" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
Makefile:66: recipe for target '/home/zheshifandian/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libcryptopp/libcryptopp-5_6_4/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/zheshifandian/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libcryptopp/libcryptopp-5_6_4/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/zheshifandian/lede/package/lean/libcryptopp'
time: package/lean/libcryptopp/compile#0.47#0.31#6.45
package/Makefile:111: recipe for target 'package/lean/libcryptopp/compile' failed
make[2]: *** [package/lean/libcryptopp/compile] Error 2

AmadeusGhost, 2020-05-21:
[image]
实测正常编译

zheshifandian, 2020-05-22:
一直提示找不到CMakeLists.txt但是有这个文件

coolsnowwolf, 2020-05-22:
make clean

closed 2020-07-04 by coolsnowwolf