coolsnowwolf/lede #4651: 红米AC2100固件有不明主机连接

wefrtghsq, 2020-05-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：刷入固件连网后概况的DHCP分配下有一台mac为54:ee:75:34:ca:7d的非本地主机， 主机名为？， 在实时信息的活动连接里面可以看到该主机建立了新加坡等地的连接。在无线的已连接站点下面看不到该主机。路由器除连接上一级路由器外，没有其他有线连接。上一级路由器只连接了光猫，没有其他有线连接。

### (2) 路由器型号和固件版本

A：红米AC2100  OpenWrt R20.5.20 / LuCI Master (git-20.117.60969-420c61a)

### (3) 详细日志

A：

coolsnowwolf, 2020-05-22:
看看是不是 zerotier 进来的

wefrtghsq, 2020-05-22:
编译时没有选zerotier

Promix953, 2020-05-22:
MAC地址 | 54:ee:75:34:ca:7d
组织名称 | Wistron InfoComm(Kunshan)Co.,Ltd.
国家/地区 | CN
省份(州) | Jiangsu
城市 | Kunshan
街道 | 168# First Avenue,Kunshan Integrated Free Trade Zone,Kunshan,Jiangsu,China
邮编 | 215300

hxlls, 2020-05-23:
纬创力的MAC，大概率是笔记本

wefrtghsq, 2020-05-24:
原因已经找到，是我初始化设置时用有线连接的网卡。

closed 2020-05-24 by wefrtghsq