coolsnowwolf/lede #4652: lan口数据包计数数量溢出，导致路由器网络奇慢

msn1983aa, 2020-05-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：因为使用了京东云、挂PT等数据量比较大，每隔半个月左右，就会出现lan口统计数据的数据包计数溢出变为负数，此时网络里的设备就会上网超级慢，或者直接断网。

### (2) 路由器型号和固件版本

A：软路由，配置为：J1900+2G+16Gmsata

### (3) 详细日志

A：
[image]

hxlls, 2020-05-22:
走错门了把，你这种魔改固件不应该找作者么？

msn1983aa, 2020-05-22:
> 走错门了把，你这种魔改固件不应该找作者么？

esir这个固件没有改，就是加了些软件包直接编译出来的

hxlls, 2020-05-22:
[image]
不改？版本号都改了。。明摆着啊其他还做了什么改就更不用说了

coolsnowwolf, 2020-05-22:
居然有这个现象？

msn1983aa, 2020-05-22:
> 居然有这个现象？

因为1月份挂上京东云之后一直上传比较猛（每天100G以上），所以遇到几次了，玩pt或者挂京东云的人应该会遇到，之前19.x版本的时候也有这样的现象。

hxlls, 2020-05-22:
[image]
[image]

我的一天就过T流量，也没这个问题，这问题真不是这个源码问题

msn1983aa, 2020-05-22:
> [image]
> [image]
> 
> 我的一天就过T流量，也没这个问题，这问题真不是这个源码问题

厉害了，您这是数据中心么，这么大流量

doumeki, 2020-05-22:
> 走错门了把，你这种魔改固件不应该找作者么？
魔改作者和lean可是有PY交易的哦 :)))))))))

mjyhj, 2020-05-23:
我也出现过一次负数，但是网络好像没问题

ElectronicElephant, 2020-05-31:
> 
> 
> 居然有这个现象？

可以肯定确实存在统计变负数的现象，我也遇到过，但是网络应该没有问题的。

Her0R0cket, 2020-06-04:
OpenWrt R20.5.9
[image]
下挂两个限速玩客云,虽然有负数,但网速没问题.
看看内核日志有没有什么报错

msn1983aa, 2020-06-11:
> OpenWrt R20.5.9
> [image]
> 下挂两个限速玩客云,虽然有负数,但网速没问题.
> 看看内核日志有没有什么报错

您还在挂玩客云啊？项目都死掉了，我都只做下载机用了

Her0R0cket, 2020-06-11:
> 
> 
> > OpenWrt R20.5.9
> > [image]
> > 下挂两个限速玩客云,虽然有负数,但网速没问题.
> > 看看内核日志有没有什么报错
> 
> 您还在挂玩客云啊？项目都死掉了，我都只做下载机用了

我好几个月没看过了,看到有上传就没管了,没想到都废了

closed 2020-07-04 by coolsnowwolf