coolsnowwolf/lede #4654: 近2天的更新导致自动编译全部空间不足报错

Cathgao, 2020-05-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

自从这两天更新之后，只要是用到了lean这套源码的action自动编译全部提示no space left，包括lean这里的action

badcrazy, 2020-05-22:
一样，更新以后自动编译提示 No space left on device，很费解

THPHOENIX, 2020-05-22:
我来补充点细节把。我忍不住自己找了两天自己问题，设备是orangepi_zero，物理机，虚拟机，git的action都编译失败，还一度重装系统，删仓库，还注册过另外一个账号来action。。。。
补充总结下详细的报错具体内容：
```
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [world] Error 1
/home/runner/work/ff/ff/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..5ae70efb04f2cbe564276603b91b6022d7386672
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..5ae70efb04f2cbe564276603b91b6022d7386672
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..5ae70efb04f2cbe564276603b91b6022d7386672
awk: cmd. line:14: (FILENAME=- FNR=177) fatal: print to "standard output" failed (No space left on device)
/home/runner/work/ff/ff/openwrt/tmp/info/.files-packageinfo.mk:176: *** unterminated call to function 'eval': missing ')'.  Stop.
make[1]: *** [prepare-tmpinfo] Error 2
/home/runner/work/ff/ff/openwrt/include/toplevel.mk:83: recipe for target 'prepare-tmpinfo' failed
/home/runner/work/ff/ff/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..5ae70efb04f2cbe564276603b91b6022d7386672
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..5ae70efb04f2cbe564276603b91b6022d7386672
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..5ae70efb04f2cbe564276603b91b6022d7386672
/home/runner/work/ff/ff/openwrt/tmp/info/.files-packageinfo.mk:176: *** unterminated call to function 'eval': missing ')'.  Stop.
/home/runner/work/ff/ff/openwrt/include/toplevel.mk:83: recipe for target 'prepare-tmpinfo' failed
make[1]: *** [prepare-tmpinfo] Error 2
/home/runner/work/ff/ff/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```
表现在github的提示结果就是
`Process completed with exit code 2.`
`Failed to remove 'http.https://github.com/.extraheader' from the git config`
`No space left on device`
然后我去物理机和虚拟机，也是报错
## recipe for target 'world' failed
要不就是这个*world*关键词换成其他的。
突然我想到还保留了5月20号凌晨的配置文件和源码，action编译是成功的。然后对比把这个配置文件push到现在的源码，编译失败！！！我才觉得可能是源码更新出了问题，我就来翻issue了！！！果然有人提了！！！！！

maodia, 2020-05-22:
我也报同样的错误“make: *** [world] Error 2”，无论是github的action还是我本地的ubuntu虚拟机构建，结果都一样

badcrazy, 2020-05-23:
> 我也报同样的错误“make: *** [world] Error 2”，无论是github的action还是我本地的ubuntu虚拟机构建，结果都一样

一样，/home/runner/work/action/action/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
这个问题引起空间不足的错误

garypang13, 2020-05-23:
https://github.com/actions/virtual-environments/issues/709#issuecomment-630093068 单纯的github actions的问题,应该已经修复了.

Cathgao, 2020-05-23:
github小动作造成的编译环境bug，现action已经编译成功

closed 2020-05-23 by Cathgao

THPHOENIX, 2020-05-23:
卧槽，刚编译成功了。心态崩了，问题解决了。orangepi_zero可以用了

Cathgao, 2020-05-23:
> 卧槽，刚编译成功了。心态崩了，问题解决了。orangepi_zero可以用了

所以你物理机和虚拟机不行应该是你自己问题。我这边因为是移动网所以依赖action提供固件就没去本地尝试。

zsbai, 2021-06-06:
如果是GitHub action修复了，今天我又遇到了相同的问题....
action编译出现
toplevel.mk:225: recipe for target 'world' failed