coolsnowwolf/lede #4655: 编译组件包引用冲突

bigbugcc, 2020-05-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：kmod-ath10k和kmod-ath10k-ct组件包冲突

### (2) 路由器型号和固件版本

A：树莓派4B 

### (3) 详细日志

A：Collected errors:
 * check_data_file_clashes: Package kmod-ath10k-ct-smallbuffers wants to install file /home/bigbug/WorkSpace/lede/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/modules/4.19.122/ath10k_core.ko
	But that file is already provided by package  * kmod-ath10k-ct
 * check_data_file_clashes: Package kmod-ath10k-ct-smallbuffers wants to install file /home/bigbug/WorkSpace/lede/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/modules/4.19.122/ath10k_pci.ko
	But that file is already provided by package  * kmod-ath10k-ct
 * opkg_install_cmd: Cannot install package kmod-ath10k-ct-smallbuffers.
 * check_data_file_clashes: Package kmod-ath10k wants to install file /home/bigbug/WorkSpace/lede/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/modules/4.19.122/ath10k_core.ko
	But that file is already provided by package  * kmod-ath10k-ct
 * check_data_file_clashes: Package kmod-ath10k wants to install file /home/bigbug/WorkSpace/lede/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/modules/4.19.122/ath10k_pci.ko
	But that file is already provided by package  * kmod-ath10k-ct
 * opkg_install_cmd: Cannot install package kmod-ath10k.
make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/home/bigbug/WorkSpace/lede'
make[1]: *** [package/Makefile:108: /home/bigbug/WorkSpace/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/bigbug/WorkSpace/lede'
make: *** [/home/bigbug/WorkSpace/lede/include/toplevel.mk:227: world] Error 2

Cathgao, 2020-05-22:
自己选择一个，把另一个取消掉

bigbugcc, 2020-05-23:
> 自己选择一个，把另一个取消掉

是的，目前也只能这样了

cattypapa, 2020-06-01:
老哥，借楼问你一下，树莓派4b编译的时候，需要选择哪个内核才能有wifi啊？

bigbugcc, 2020-06-04:
> 老哥，借楼问你一下，树莓派4b编译的时候，需要选择哪个内核才能有wifi啊？

这个要在无线中选添加

closed 2020-07-04 by coolsnowwolf