coolsnowwolf/lede #4658: 红米ac2100没有5g wifi

cattypapa, 2020-05-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ “x”] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：红米ac2100编译完之后，没有5gwifi，已经选择了mt7615 wifi驱动，hostapd等，编译完，只有一个2.4Gwifi，设置一番之后，才有13个信道，希望得到解答。

### (2) 路由器型号和固件版本

A：redmi ac2100

### (3) 详细日志

A：无

wefrtghsq, 2020-05-24:
我也遇到过同样的问题，红米AC2100 5GWIFI只能设置前面几个信道， 并且不能设置160M， 如果设置后5G不能启动，只能重新刷固件，刷固件时不要保留设置。

cattypapa, 2020-05-24:
> 我也遇到过同样的问题，红米AC2100 5GWIFI只能设置前面几个信道， 并且不能设置160M， 如果设置后5G不能启动，只能重新刷固件，刷固件时不要保留设置。

不知道这个问题到底怎么解决，唉

neighbads, 2020-05-27:
同样遇到这个问题。

kl9800, 2020-05-27:
请问上面三位大佬，
我也想自己编译红米ac2100的固件，
但是
Target System ， Subtarget ， Target Profile 这三个分别选什么？我找了半天只找到了7621，没找到redmi呀 。。。。

neighbads, 2020-05-27:
@kl9800 

Target System (MediaTek Ralink MIPS)  ---> 
Subtarget (MT7621 based boards)  --->
Target Profile (Xiaomi Redmi Router AC2100)  --->

kl9800, 2020-05-27:
> @kl9800
> 
> Target System (MediaTek Ralink MIPS) --->
> Subtarget (MT7621 based boards) --->
> Target Profile (Xiaomi Redmi Router AC2100) --->

谢谢大佬

kl9800, 2020-05-27:
> Target Profile

可是我找了半天，我的Target Profile下面没有Xiaomi Redmi Router AC2100 呀

kl9800, 2020-05-27:
额没事了，我强制拉取一下就好了，之前改了点东西所以拉取失败了。。

kl9800, 2020-05-27:
然后我还想问一下，红米这个，编译出的固件体积最大不能超过多少M？25M吗？

neighbads, 2020-05-27:
@kl9800 https://github.com/coolsnowwolf/lede/blob/1be41d9899aa11c5d864ee19649d8e2c83f740c5/target/linux/ramips/image/mt7621.mk#L916

kl9800, 2020-05-27:
但实际真的能用这么多么？

> @kl9800
> 
> https://github.com/coolsnowwolf/lede/blob/1be41d9899aa11c5d864ee19649d8e2c83f740c5/target/linux/ramips/image/mt7621.mk#L916

但实际真的能用这么多么？是不是如果超过30M会刷不进去？

kl9800, 2020-05-27:
> 但实际真的能用这么多么？
> 
> > @kl9800
> > https://github.com/coolsnowwolf/lede/blob/1be41d9899aa11c5d864ee19649d8e2c83f740c5/target/linux/ramips/image/mt7621.mk#L916
> 
> 但实际真的能用这么多么？是不是如果超过30M会刷不进去？

比如overlay之类的问题，实际上我最大可以刷进去多大体积的固件？

closed 2020-07-04 by coolsnowwolf